springfall2008/batpred #1084: Hass.get history issue with latest update

stefanwest1, 2024-05-11:
**Describe the bug**
Since the latest update, I’m getting the following error.
ERROR: Exception raised Hass.get_history() takes 1 positional argument but 2 were given

**Expected behavior**
A clear and concise description of what you expected to happen.

**Predbat version**

7.18.6

**Environment details**

- Appdaemon Docker

**Screenshots**

I can’t copy the log file right now as I’m on my phone so the best I can do is upload a screenshot

[image]

stefanwest1, 2024-05-11:
Dropping back to 7.18.2 fixes the issue

PianSom, 2024-05-12:
I had the same issue (using Appdeamon Docker) and the same solution

sammort, 2024-05-13:
Just wanted to +1 this issue for me, also running AppDaemon as a Docker container

springfall2008, 2024-05-14:
Now fixed on main for testing https://github.com/springfall2008/batpred/pull/1093

closed 2024-05-15 by springfall2008