springfall2008/batpred #1085: PauseCharge was not switched off when a charge window was planned/enabled, so battery would not charge

mpartington, 2024-05-12:
**Describe the bug**
Spotted that my battery was not charging, despite Predbat planning to do so.  Looking at the logs I can see Predbat switches :
2024-05-12 13:01:36.183835 INFO pred_bat: Inverter 0 Wrote inverter_mode to PauseCharge successfully

In later runs, it changes the status to 'charging', however does not switch charge pause off, so the battery remains in limbo

**Expected behavior**
Set the pause feature to disabled and allow battery to charge

**Predbat version**

v7.18.6

**Environment details**

- Standard HAOS installer 

**Screenshots**
[image]
**Log file**
[appdaemon.log](https://github.com/springfall2008/batpred/files/15286436/appdaemon.log)

gcoan, 2024-05-12:
I had exactly the same at 11am this morning 

The battery was paused charging at 8:07, ignoring the accidental change I made then reverted at 9:34, predbat started charging at 11:00 but the pause charge wasn't removed - I manually removed it at 11:03

Predbat 7.18.1

[image]

[image]

Givtcp logs showing no attempt to change the pause 
[image]
[image]
[image]

gcoan, 2024-05-13:
@springfall2008 I see you are working on a fix for this.  Do you need my logfile from the above?

springfall2008, 2024-05-14:
The fix is already on main

mpartington, 2024-05-15:
Will reopen if not fixed

closed 2024-05-15 by mpartington