springfall2008/batpred #1087: Charge window not updated to reflect changes in plan

nbullus, 2024-05-12:
**Describe the bug**
When the charge window schedule changes from what was previously planned, it is not always being updated to reflect the current plan. 

**Expected behaviour**
The next scheduled charge window should be updated in line with the most recent charge plan.  The current plan has following defined:

22:30 - 01:00 - Freeze Charge
01:00 - 01:30 - Charge
01:30 - 02:00 - Mixed Charge / Discharge

None of these times match with a charge schedule of 00:00 - 01:20 as shown in the screenshot. This may have been correct at 16:00, but has obviously changed since. Either the charge schedule is wrong or the plan is wrong.

**Predbat version**

7.18.6

**Environment details**

- GivEnergy AIO
- Standard HAOS installer

**Screenshots**

The following shows what is currently scheduled vs what is now planned. The charge schedule and the plan are not in sync.

[image]

The charge schedule 00:00 - 01:20 was calculated at 16:00 and has not been updated since then, despite multiple changes to the plan and the next charge slot.

**Log file**

**GivTCP Log:**

**2024-05-12 16:00:09,468 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 00:00 - 01:20
2024-05-12 16:00:12,372 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success**
2024-05-12 16:00:18,641 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-05-12 16:00:23,807 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-05-12 16:00:29,854 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: PauseCharge
2024-05-12 16:00:31,396 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to PauseCharge was a success
2024-05-12 16:00:39,891 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 100
2024-05-12 16:00:45,187 - Inv1 - write       -  [INFO    ] - Setting Charge Target 100 was a success
2024-05-12 16:01:45,240 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 6000 (50)
2024-05-12 16:01:47,762 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-05-12 16:01:54,079 - Inv1 - write       -  [INFO    ] - Disabling Charge Schedule
2024-05-12 16:01:55,820 - Inv1 - write       -  [INFO    ] - Disabling Charge was a success
2024-05-12 16:02:02,907 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: PauseDischarge
2024-05-12 16:02:08,781 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to PauseDischarge was a success
2024-05-12 16:02:12,987 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 49
2024-05-12 16:02:14,540 - Inv1 - write       -  [INFO    ] - Setting shallow charge 49 was a success
2024-05-12 16:06:04,045 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-05-12 16:06:05,601 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-05-12 16:06:11,681 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: PauseCharge
2024-05-12 16:06:13,230 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to PauseCharge was a success
2024-05-12 16:06:21,704 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-05-12 16:06:23,250 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-05-12 16:06:29,059 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-05-12 16:06:29,625 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response'), <traceback object at 0x7f933166c980>)
2024-05-12 16:26:05,227 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 16:31
2024-05-12 16:26:08,107 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-05-12 16:26:14,168 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Timed Export
2024-05-12 16:26:14,706 - Inv1 - write       -  [INFO    ] - Setting export mode was a success
2024-05-12 16:26:16,261 - Inv1 - write       -  [INFO    ] - Enabling Discharge was a success
2024-05-12 16:26:22,491 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 37
2024-05-12 16:26:27,661 - Inv1 - write       -  [INFO    ] - Setting shallow charge 37 was a success
2024-05-12 16:30:18,849 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Eco
2024-05-12 16:30:19,666 - Inv1 - write       -  [INFO    ] - Setting dynamic mode was a success
2024-05-12 16:30:21,221 - Inv1 - write       -  [INFO    ] - Setting shallow charge 37 was a success
2024-05-12 16:30:37,315 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 00:00
2024-05-12 16:30:41,309 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-05-12 16:30:47,401 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 6000 (50)
2024-05-12 16:30:48,944 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-05-12 16:30:55,220 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: Disabled
2024-05-12 16:31:00,982 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2024-05-12 16:31:05,273 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-05-12 16:31:06,823 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success

Predbat Log:

[Predbat_log_12-5-2024.txt](https://github.com/springfall2008/batpred/files/15287416/Predbat_log_12-5-2024.txt)

springfall2008, 2024-05-14:
I'm not sure I understand the issue as scheduled charge is disabled

"024-05-12 16:01:54,079 - Inv1 - write - [INFO ] - Disabling Charge Schedule"

The charge window is normally set around 30 minutes before the charge time, do you not see this?

nbullus, 2024-05-14:
I forgot to check what happened at 00:00 to see if it simply enabled the existing schedule, or updated the schedule and enabled at 01:00 next morning. I've seen it where it sets the schedule and enables more than 30 mins before charge time. I will wait till it happens again and update the issue.

nbullus, 2024-05-15:
Trefor. Although not directly the cause of this issue, I do want to point out the fact that charge schedules are not being defined/enabled 30 mins before they are due and that the enabling of charge schedules is remaining on.

In the following screenshot t and Predbat/GivTCP logs, the current charge schedule (12:00 - 04:00) was set way more than 30 mins before its needed. In fact the logs indicate that a previous 03:00 - 07:00 charge schedule was defined and Enabled at 03:00 (which is correct), but then the charge schedule was not disabled when the schedule ended and the next charge window (12:00 - 04:00) was scheduled. Therefore we have an active charge window schedule/enabled 8 hrs before it is planned to start.

[image]

GivTCP Log:

2024-05-15 03:00:05,325 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: Disabled
2024-05-15 03:00:08,782 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2024-05-15 03:00:15,333 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 03:00 - 07:30
2024-05-15 03:00:18,189 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
2024-05-15 03:00:24,159 - Inv1 - write       -  [INFO    ] - Enabling Charge Schedule
2024-05-15 03:00:29,633 - Inv1 - write       -  [INFO    ] - Setting Charge Enable was a success
2024-05-15 03:00:35,645 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-05-15 03:00:37,154 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-05-15 05:00:11,211 - Inv1 - read        -  [INFO    ] - Saving current energy stats at start of day rate tariff 2024-05-15 07:30:05,347 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 12:00 - 16:00
2024-05-15 07:30:08,191 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success

[Predbat_log_15-5-2024.txt](https://github.com/springfall2008/batpred/files/15318049/Predbat_log_15-5-2024.txt)

nbullus, 2024-05-20:
I think the question has now been answered. The charge schedule is being enabled more than 30 mins before its due to start as shown in the previous screenshot.

nbullus, 2024-05-24:
Here is a more recent example on a much simpler Flux tariff. The charge schedule is both wrong and still Enabled.

[image]

GivTCP confirms a charge schedule was enabled at 01:30 which is correct for the standard Flux  02:00-05:00 off-peak charge, but has remained Enabled after the charge period ended:

2024-05-24 01:30:05,634 - Inv1 - write       -  [INFO    ] - Enabling Charge Schedule
2024-05-24 01:30:07,160 - Inv1 - write       -  [INFO    ] - Setting Charge Enable was a success
2024-05-24 02:00:03,105 - Inv1 - read        -  [INFO    ] - Saving current energy stats at start of night rate tariff (Dynamic)
2024-05-24 02:00:05,513 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 75
2024-05-24 02:00:07,177 - Inv1 - write       -  [INFO    ] - Setting Charge Target 75 was a success
2024-05-24 02:00:29,689 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 53
2024-05-24 02:00:31,337 - Inv1 - write       -  [INFO    ] - Setting Charge Target 53 was a success
2024-05-24 02:05:22,602 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 94
2024-05-24 02:05:24,246 - Inv1 - write       -  [INFO    ] - Setting Charge Target 94 was a success
2024-05-24 02:10:22,327 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 93
2024-05-24 02:10:23,975 - Inv1 - write       -  [INFO    ] - Setting Charge Target 93 was a success
2024-05-24 02:15:21,896 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 91
2024-05-24 02:15:23,544 - Inv1 - write       -  [INFO    ] - Setting Charge Target 91 was a success
2024-05-24 02:20:22,076 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 90
2024-05-24 02:20:23,744 - Inv1 - write       -  [INFO    ] - Setting Charge Target 90 was a success
2024-05-24 02:25:22,156 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 89
2024-05-24 02:25:23,802 - Inv1 - write       -  [INFO    ] - Setting Charge Target 89 was a success
2024-05-24 02:30:21,988 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 88
2024-05-24 02:30:23,431 - Inv1 - write       -  [INFO    ] - Setting Charge Target 88 was a success
2024-05-24 02:40:22,359 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 87
2024-05-24 02:40:23,820 - Inv1 - write       -  [INFO    ] - Setting Charge Target 87 was a success
2024-05-24 02:45:22,176 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 86
2024-05-24 02:45:23,831 - Inv1 - write       -  [INFO    ] - Setting Charge Target 86 was a success
2024-05-24 02:50:22,031 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 87
2024-05-24 02:50:23,680 - Inv1 - write       -  [INFO    ] - Setting Charge Target 87 was a success
2024-05-24 02:55:22,167 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 86
2024-05-24 02:55:28,389 - Inv1 - write       -  [INFO    ] - Setting Charge Target 86 was a success
2024-05-24 03:05:21,897 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 100
2024-05-24 03:05:23,542 - Inv1 - write       -  [INFO    ] - Setting Charge Target 100 was a success
2024-05-24 05:00:05,632 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 19:30 - 20:00
2024-05-24 05:00:08,482 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
2024-05-24 05:00:13,297 - Inv1 - read        -  [INFO    ] - Saving current energy stats at start of day rate tariff (Dynamic)

The attached Predbat log confirms the original charge window was 02:00-05:00 and enabled at 01:30, but this was changed to 19:30-20:00 at 05:00 and left Enabled. It should have been Disabled the moment that the plan changed and the 19:30-20:00 charge removed:

[Predbat_Log_2024-05-24.txt](https://github.com/springfall2008/batpred/files/15429420/Predbat_Log_2024-05-24.txt)

springfall2008, 2024-05-24:
This is normal, the charge schedule is only changed 30 minutes before a charge slot or before the current settings if they are wrong. 

The charge enable should be for AC charging only, are you actually seeing a problem with PV charging?

mpartington, 2024-05-24:
Interesting, this seems a hardware dependant issue.  From an AC3 perspective,  when I was on a predictable tariff such as GO, it was my preference to maintain the charge schedule 'on' all the time, as it reduced (in my eyes) the risk of the batteries not charging if there was a router/dongle/GivTCP connectivity issue as it was all .  

Are you definitely on the latest firmware, as i thought this had been fixed on AIO (i.e. an ac charge target preventing solar exceeding that value)?

Alternatively, have you switched on the 'Inverter SOC Reset' option, I think this was to get around this, by resetting to 100%

nbullus, 2024-05-24:
No issue with PV charging the battery with surplus solar.

I agree its fine to set the charge schedule times in advance, but you shouldn't be leaving the charge schedule enabled after its finished when the next charge schedule is another 14 hours away. The charge schedule should be Enabled when it gets within 30 mins of the start window and Disabled at the end of that window (unless there is another charge window set and its due to start within next 30 mins). Predbat is sometimes leaving the charge schedule enabled after its finished (as in this example) which means the battery can no longer charge for another 14 hours if the plan was to change and another charge slot was dropped in suddenly. This is because you have a future charge window Enabled and it wont then charge until this time is reached. By Disabling the charge schedule after its finished, it prevents any last minute charge slots from being ignored.

I had a temporary automation running over the top of Predbat which disabled the charge schedule if it was either more than 30 mins away from the start, or the end of the schedule had been reached. I then had no issues with plan changes on Agile and last minute charge slots being dropped in worked as expected. The moment I turned this automation off I had issues again. When I moved over from Agile to Flux I also turned this automation off an  I started getting charge schedules left enabled after they had finished. Last minute charge slots being dropped into the plan on Flux is unlikely, but it can happen and often does on Agile and other tariffs.

When you first start Predbat up, it will often set the charge times up, but won't Enable the schedule until it gets within 30 mins of it. Soon as that first schedule has passed, it can set the next charge window up ready, but leave the charge schedule Enabled instead of Disabling it and then re-Enabling when it gets to within 30 mins of the next window.

nbullus, 2024-05-24:
> Interesting, this seems a hardware dependant issue. From an AC3 perspective, when I was on a predictable tariff such as GO, it was my preference to maintain the charge schedule 'on' all the time, as it reduced (in my eyes) the risk of the batteries not charging if there was a router/dongle/GivTCP connectivity issue.
> 
> Are you definitely on the latest firmware, as i thought this had been fixed on AIO (i.e. an ac charge target preventing solar exceeding that value)?
> 
> Alternatively, have you switched on the 'Inverter SOC Reset' option, I think this was to get around this, by resetting to 100%

The  'Inverter SOC Reset' option is always on for me.

I'm also on the latest inverter firmware and battery firmware BMS8. There is a BMS 9, but nothing in there that should effect this. 

You cannot have the 'Enable Charge Schedule' permanently left 'on' for Agile as it causes last minute charge slots added to be plan to be ignored if you don't also update the start/end times (which it often doesn't). The Predbat log confirms that Predbat is leaving 'Enable Charge Schedule' turned on after one schedule has ended and the next schedule setup some 14+ hours in the future. Predbat made no attempt to disable the schedule once it was complete and simply updated the next start/end time. 

I've lost track of how many times its sets up a future schedule many hours away, there is then a last minute charge slot added to the plan and its just ignored as the future charge schedule now has the wrong start time. If an earlier charge slot is added to the plan, then the start time of the next charge window should be updated to reflect it.

nbullus, 2024-05-24:
Just to avoid confusion, I made a mistake when I stated "The battery can now no longer charge with surplus solar later on as planned before 19:30". What i mean to say was "The battery can now no longer charge if another charge slot is added before the  19:30".

closed 2025-01-01 by springfall2008