springfall2008/batpred #1100: 7.19.0 errors

PianSom, 2024-05-14:
Using Appdaemon docker

Updated from 7.18.2 without issue. All appeared good. Restarted HA Container (for another reason). Got an error thrown by Predat. error file.txt showed as below. Restart Appdaemon cleared error and started normal working

<details><summary>Details</summary>
<p>

2024-05-14 20:25:03.091335 WARNING pred_bat: ------------------------------------------------------------
2024-05-14 20:25:03.091540 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-05-14 20:25:03.091747 WARNING pred_bat: Worker Ags: {'id': '115ddb3c0888472583eafd3630bc028c', 'name': 'pred_bat', 'objectid': '8ed9ed8194f1489c8e70901895c094af', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7fa55a7e53c0>>, 'pin_app': True, 'pin_thread': 8, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-8'}}
2024-05-14 20:25:03.091925 WARNING pred_bat: ------------------------------------------------------------
2024-05-14 20:25:03.092245 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/local/lib/python3.10/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
  File "/conf/apps/predbat/predbat.py", line 14769, in run_time_loop
    raise e
  File "/conf/apps/predbat/predbat.py", line 14765, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/local/lib/python3.10/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
  File "/conf/apps/predbat/predbat.py", line 13656, in update_pred
    self.fetch_sensor_data()
  File "/conf/apps/predbat/predbat.py", line 12767, in fetch_sensor_data
    self.import_today = self.minute_data_import_export(self.now_utc, "import_today", scale=self.import_export_scaling, required_unit="kWh")
  File "/conf/apps/predbat/predbat.py", line 5110, in minute_data_import_export
    raise ValueError
ValueError

2024-05-14 20:25:03.092430 WARNING pred_bat: ------------------------------------------------------------
2024-05-14 20:26:54.914950 WARNING pred_bat: ------------------------------------------------------------
2024-05-14 20:26:54.915175 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-05-14 20:26:54.916252 WARNING pred_bat: Worker Ags: {'id': 'ae8d18e6e83b4629a55665aadcdc31b4', 'name': 'pred_bat', 'objectid': '252c5a9c36b7432fa2573dd140222c45', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7fa55a7d4b80>>, 'pin_app': True, 'pin_thread': 8, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-8'}}
2024-05-14 20:26:54.916551 WARNING pred_bat: ------------------------------------------------------------
2024-05-14 20:26:54.918242 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/local/lib/python3.10/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
  File "/conf/apps/predbat/predbat.py", line 14747, in update_time_loop
    raise e
  File "/conf/apps/predbat/predbat.py", line 14742, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/local/lib/python3.10/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
  File "/conf/apps/predbat/predbat.py", line 13657, in update_pred
    self.fetch_inverter_data()
  File "/conf/apps/predbat/predbat.py", line 13183, in fetch_inverter_data
    curve = inverter.find_charge_curve(discharge=False)
  File "/conf/apps/predbat/predbat.py", line 2567, in find_charge_curve
    battery_power_data = self.base.get_history_wrapper(entity_id=battery_power_sensor, days=self.base.max_days_previous)
  File "/conf/apps/predbat/predbat.py", line 5070, in get_history_wrapper
    raise ValueError
ValueError

2024-05-14 20:26:54.927958 WARNING pred_bat: ------------------------------------------------------------

</p>
</details>

SwiftRR, 2024-05-14:
Mine is a HAOS install on Raspberry Pi 4 SSD.

I have felt that I have been letting down Trefor by opting out of automatic updates and testing since v7.17.1.
Having seen your update report on 7.19.0, I have felt guilty and I have updated.

Finger-crossed, the installation of v7.19.0 has gone smoothly with no issues.

I have checked predbat.log, appdaemon-predbat log, givtcp log and HA log.
No errors.

I will investigate the update with its new features tomorrow after leaving it overnight to settle down (albeit with little activity with battery staying at 100% most of the night (on Agile and Octopus Outgoing 15p)

Rob

PianSom, 2024-05-15:
Not sure there’s any need to feel guilty! Sometimes it’s best just to pick your moment to do the update dance. ATM I’m holding off the move to givtcp beta and bottlecapdave’s Octopus 11 until I feel I have the time to recover if something breaks. 

You may want to restart HA and see if you - like me - have predbat throw an error.

SwiftRR, 2024-05-15:
As I am in 'testing mode', I have restarted HA and I think I got the same issue as you experienced in your Docker virtualisation.
I then restarted appdaemon-predbat which cleared the issue as shown by log below:

 Predbat pre-install in AppDaemon
-----------------------------------------------------------
 Add-on version: 1.0.10
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.5  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.3.3
 Home Assistant Supervisor: 2024.04.4
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[09:17:59] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
Update complete
2024-05-15 08:00:03.255063 WARNING pred_bat: ------------------------------------------------------------
2024-05-15 08:00:03.256215 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-05-15 08:00:03.256857 WARNING pred_bat: Worker Ags: {'id': 'fea72e2b8f36445989415ff06ccdf720', 'name': 'pred_bat', 'objectid': '7d8b94cb91d3410597c67aed74e444f1', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f922e7110>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-05-15 08:00:03.257489 WARNING pred_bat: ------------------------------------------------------------
2024-05-15 08:00:03.260367 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 14769, in run_time_loop
    raise e
  File "/config/apps/predbat.py", line 14765, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 13656, in update_pred
    self.fetch_sensor_data()
  File "/config/apps/predbat.py", line 12752, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 5157, in minute_data_load
    raise ValueError
ValueError

2024-05-15 08:00:03.260931 WARNING pred_bat: ------------------------------------------------------------
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service appdaemon: stopping
[08:04:27] INFO: Service AppDaemon exited with code 0 (by signal 0)
s6-rc: info: service appdaemon successfully stopped
s6-rc: info: service init-appdaemon: stopping
s6-rc: info: service init-appdaemon successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: appdaemon-predbat
 Predbat pre-install in AppDaemon
-----------------------------------------------------------
 Add-on version: 1.0.10
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.5  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.3.3
 Home Assistant Supervisor: 2024.05.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[08:04:33] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started

springfall2008, 2024-05-15:
I'm pretty sure this is just as the fetch history command was killed during the restart, hopefully it has no functional impact beyond the first run after an update

gcoan, 2024-05-15:
To add my experience : upgraded from 7.18.1 to 7.19.0 main using HACS.
Restarted app daemon 

No errors, predbat just working fine

SJB1235, 2024-05-15:
I tried this update last night and got the same error as PianSom, and Predbat reverted to the previously installed version (7.18.6). Tried it a couple of times with the same result.  
I've just tried it  again and it worked ok.

springfall2008, 2024-05-20:
Cause not determined, closing for now

closed 2024-05-20 by springfall2008