springfall2008/batpred #1101: Unusually high overnight SOC

Tah08, 2024-05-14:
On two recent occasions (over the weekend) and for tomorrow, I've noticed an usually high overnight SOC of 80-90% which is resulting in a 100% SOC by around 11am.

**Expected behaviour**
I would expect a much lower overnight SOC to allow for tomorrow's solar.

**Predbat version**
17.9

**Environment details**

- GIV-AC3.0 Inverter, 9.5kwh battery
- Standard HAOS installer
- Intelligent Octopus
- Zappi charger
- old FIT

**Screenshots**
[Predbat settings.pdf](https://github.com/springfall2008/batpred/files/15314191/Predbat.settings.pdf)
[Predbat plan.pdf](https://github.com/springfall2008/batpred/files/15314188/Predbat.plan.pdf)
**[image]**
[image]

**Log file**
[predbat.log](https://github.com/springfall2008/batpred/files/15314209/predbat.log)

springfall2008, 2024-05-15:
Can you try turning off combine charge?

springfall2008, 2024-05-15:
And then if that doesn't change things re-run the plan with debug enabled and post the log?

springfall2008, 2024-05-15:
Just an update I tried on my side with 0 export and it stopped charging, so something is odd...

springfall2008, 2024-05-15:
One thing that is odd is your charge curve, I think it makes little sense on its current setting:

2024-05-14 22:32:36.303129 INFO pred_bat: Curve before adjustment is: {88: 0.52, 87: 0.99, 86: 0.52}

This seems slightly more logical, but 0.53 flat to 85% doesn't seem quite right, are you sure everything is set up correctly?

2024-05-14 22:32:36.306394 INFO pred_bat: Curve curve can be entered into apps.yaml or set to auto:
  battery_charge_power_curve:
    100 : 0.53
    99 : 0.53
    98 : 0.53
    97 : 0.53
    96 : 0.53
    95 : 0.53
    94 : 0.53
    93 : 0.53
    92 : 0.53
    91 : 0.53
    90 : 0.53
    89 : 0.53
    88 : 0.53
    86 : 0.53
    85 : 0.53

Tah08, 2024-05-16:
@springfall2008 as my curve is set to auto, I wasn't sure if you think I should change anything in the short term?

Tah08, 2024-05-16:
As mentioned, this is a recent occurrence so i have rolled back versions and I think i may have helped to isolate the issue:

Here is my plan for 7.18.3 which is kinda what I'd expect - [predbat plan 7.18.3.pdf](https://github.com/springfall2008/batpred/files/15340873/predbat.7.18.3.pdf)

Here is my plan for 7.18.4 which looks a bit off - [predbat plan 7.18.4.pdf](https://github.com/springfall2008/batpred/files/15340878/predbat.7.18.4.pdf)

Tah08, 2024-06-22:
As requested, debug log file attached - showing some overnight charging and freeze charging which is resulting in 100% SOC by 11am tomorrow morning.
[predbat_debug_14_30_00.zip](https://github.com/user-attachments/files/15936747/predbat_debug_14_30_00.zip)

closed 2024-06-28 by Tah08