springfall2008/batpred #1104: Query about the accuracy of suggested charge and discharge rate from the 'auto generate charge curve' routines

PianSom, 2024-05-16:
I have an AIO, and GivTCP returns `number.givtcp_SERIALNO_battery_charge_rate` and `...discharge_rate` as both being 6kW. Looking at actual experience reported by GivTCP my battery is actually discharging at around 6.27kW and is charging at around 5.8kW

However the `input_number.predbat_battery_rate_max_scaling_discharge` and charge settings suggested by Predbat seem to bear little resemblance to these values. I am using the Predbat suggested values, but Geoffrey thought it may be worth checking that the Predbat suggestions are correct?

[image]

PianSom, 2024-05-16:
Not least because the actual discharge rate reported is 5% MORE than the rated value! Hence the question.

(Wasn't really querying the curves.)

springfall2008, 2024-11-16:
Probably down to losses, maybe you set them too high?

closed 2024-11-16 by springfall2008