springfall2008/batpred #1107: Predbat not working after latest update

alanvalente75, 2024-05-16:
Hi

since the last update Predbat doesn’t work on HA - The logs state the following

homeassistant/appdaemon/apps/batpred/predbat.py", line 14212, in async_restore_settings_yaml
    if (item["value"] != item.get("default", None)) and item.get("restore", True):

SwiftRR, 2024-05-16:
Have you tried restarting the appdaemon-predbat addon?
From posts, this often seems to get predbat working again following an update.

Rob

alanvalente75, 2024-05-16:
Thank you Rob that worked 👍

thewookiewon, 2024-05-17:
`-----------------------------------------------------------
 Add-on: appdaemon-predbat
 Predbat pre-install in AppDaemon
-----------------------------------------------------------
 Add-on version: 1.0.10
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.3  (aarch64 / raspberrypi5-64)
 Home Assistant Core: 2024.5.4
 Home Assistant Supervisor: 2024.05.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[15:57:53] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-05-17 15:58:34.283779 WARNING pred_bat: ------------------------------------------------------------
2024-05-17 15:58:34.284093 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-05-17 15:58:34.284254 WARNING pred_bat: Worker Ags: {'id': 'abf4bde2aa1e4980ba2e3e498edbffb1', 'name': 'pred_bat', 'objectid': 'f7de7dfe834943fa843d65306a2842a1', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7f80a3fe10>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-05-17 15:58:34.284369 WARNING pred_bat: ------------------------------------------------------------
2024-05-17 15:58:34.286933 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 14780, in update_time_loop
    raise e
  File "/config/apps/predbat.py", line 14775, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 13685, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 11938, in calculate_plan
    self.calculate_yesterday()
  File "/config/apps/predbat.py", line 11744, in calculate_yesterday
    ) = self.run_prediction(charge_limit_best, charge_window_best, [], [], False, end_record=(24 * 60), save="yesterday")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 6194, in run_prediction
    ) = pred.run_prediction(charge_limit, charge_window, discharge_window, discharge_limits, pv10, end_record, save, step)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 1721, in run_prediction
    pv_now += pv_forecast_minute_step[minute + offset]
              ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
KeyError: 1440

2024-05-17 15:58:34.287083 WARNING pred_bat: ------------------------------------------------------------
2024-05-17 16:00:03.895244 WARNING pred_bat: ------------------------------------------------------------
2024-05-17 16:00:03.895499 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-05-17 16:00:03.895690 WARNING pred_bat: Worker Ags: {'id': '5ee92bd886e440e5a44369831fe97b86', 'name': 'pred_bat', 'objectid': 'f7de7dfe834943fa843d65306a2842a1', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f80a3fe10>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-05-17 16:00:03.896010 WARNING pred_bat: ------------------------------------------------------------
2024-05-17 16:00:03.898660 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 14802, in run_time_loop
    raise e
  File "/config/apps/predbat.py", line 14798, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 13685, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
[... cut ...]
    ) = pred.run_prediction(charge_limit, charge_window, discharge_window, discharge_limits, pv10, end_record, save, step)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 1721, in run_prediction
    pv_now += pv_forecast_minute_step[minute + offset]
              ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
KeyError: 1440

2024-05-17 16:00:03.898821 WARNING pred_bat: ------------------------------------------------------------`

thewookiewon, 2024-05-17:
```
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: appdaemon-predbat
 Predbat pre-install in AppDaemon
-----------------------------------------------------------
 Add-on version: 1.0.10
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.3  (aarch64 / raspberrypi5-64)
 Home Assistant Core: 2024.5.4
 Home Assistant Supervisor: 2024.05.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[15:57:53] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-05-17 15:58:34.283779 WARNING pred_bat: ------------------------------------------------------------
2024-05-17 15:58:34.284093 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-05-17 15:58:34.284254 WARNING pred_bat: Worker Ags: {'id': 'abf4bde2aa1e4980ba2e3e498edbffb1', 'name': 'pred_bat', 'objectid': 'f7de7dfe834943fa843d65306a2842a1', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7f80a3fe10>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-05-17 15:58:34.284369 WARNING pred_bat: ------------------------------------------------------------
2024-05-17 15:58:34.286933 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 14780, in update_time_loop
    raise e
  File "/config/apps/predbat.py", line 14775, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 13685, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 11938, in calculate_plan
    self.calculate_yesterday()
  File "/config/apps/predbat.py", line 11744, in calculate_yesterday
    ) = self.run_prediction(charge_limit_best, charge_window_best, [], [], False, end_record=(24 * 60), save="yesterday")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 6194, in run_prediction
    ) = pred.run_prediction(charge_limit, charge_window, discharge_window, discharge_limits, pv10, end_record, save, step)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 1721, in run_prediction
    pv_now += pv_forecast_minute_step[minute + offset]
              ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
KeyError: 1440

2024-05-17 15:58:34.287083 WARNING pred_bat: ------------------------------------------------------------
2024-05-17 16:00:03.895244 WARNING pred_bat: ------------------------------------------------------------
2024-05-17 16:00:03.895499 WARNING pred_bat: Unexpected error in worker for App pred_bat:
[... 8 lines cut ...]
  File "/config/apps/predbat.py", line 14802, in run_time_loop
    raise e
  File "/config/apps/predbat.py", line 14798, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 13685, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 11938, in calculate_plan
    self.calculate_yesterday()
  File "/config/apps/predbat.py", line 11744, in calculate_yesterday
    ) = self.run_prediction(charge_limit_best, charge_window_best, [], [], False, end_record=(24 * 60), save="yesterday")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 6194, in run_prediction
    ) = pred.run_prediction(charge_limit, charge_window, discharge_window, discharge_limits, pv10, end_record, save, step)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 1721, in run_prediction
    pv_now += pv_forecast_minute_step[minute + offset]
              ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
KeyError: 1440

2024-05-17 16:00:03.898821 WARNING pred_bat: ------------------------------------------------------------
```

RobinCu, 2024-05-17:
+1 with the Key error 1440.

Tried a few AD restarts but still present. I've rolled back to 7.19.0 and it's working again.

thewookiewon, 2024-05-17:
Hmm tried rolling back to 7.19.0, restarted HA and still getting Key error 1440.

thewookiewon, 2024-05-17:
Yep, confirmed rolling back to 7.19.0 fixes the problem for me

SwiftRR, 2024-05-18:
I upgraded to 7.19.1 yesterday with no issues.

rob

Neomancer86, 2024-05-18:
+1 with errors,

7.19.1 Changed to ERROR: Exception raised 1440
7.19.2  Changed to ERROR: Exception raised 1470

RobinCu, 2024-05-18:
7.19.3 AD logs:
```2024-05-18 15:23:57.384137 WARNING pred_bat: ------------------------------------------------------------
2024-05-18 15:23:57.385162 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-05-18 15:23:57.385373 WARNING pred_bat: Worker Ags: {'id': 'dab4788f4e894b61a1f9d4991c0e618d', 'name': 'pred_bat', 'objectid': 'a56da057f3424f1a85dec2af70cc311d', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7f4592596590>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-05-18 15:23:57.385580 WARNING pred_bat: ------------------------------------------------------------
2024-05-18 15:23:57.389137 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/batpred/predbat.py", line 14746, in update_time_loop
    raise e
  File "/config/apps/batpred/predbat.py", line 14741, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/batpred/predbat.py", line 13649, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/batpred/predbat.py", line 11908, in calculate_plan
    self.calculate_yesterday()
  File "/config/apps/batpred/predbat.py", line 11706, in calculate_yesterday
    ) = self.run_prediction(charge_limit_best, charge_window_best, [], [], False, end_record=(24 * 60), save="yesterday")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/batpred/predbat.py", line 6187, in run_prediction
    ) = pred.run_prediction(charge_limit, charge_window, discharge_window, discharge_limits, pv10, end_record, save, step)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/batpred/predbat.py", line 1714, in run_prediction
    pv_now += pv_forecast_minute_step[minute + offset]
              ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
KeyError: 1470

2024-05-18 15:23:57.389404 WARNING pred_bat: ------------------------------------------------------------
2024-05-18 15:25:08.928140 WARNING pred_bat: ------------------------------------------------------------
2024-05-18 15:25:08.928401 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-05-18 15:25:08.928616 WARNING pred_bat: Worker Ags: {'id': '0980e503eeb242d3af6fe2922a73cc20', 'name': 'pred_bat', 'objectid': '5cd1083d38be422b92645f681ed72240', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f4592596590>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-05-18 15:25:08.928802 WARNING pred_bat: ------------------------------------------------------------
2024-05-18 15:25:08.932529 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/batpred/predbat.py", line 14768, in run_time_loop
    raise e
  File "/config/apps/batpred/predbat.py", line 14764, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/batpred/predbat.py", line 13649, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/batpred/predbat.py", line 11908, in calculate_plan
    self.calculate_yesterday()
  File "/config/apps/batpred/predbat.py", line 11706, in calculate_yesterday
    ) = self.run_prediction(charge_limit_best, charge_window_best, [], [], False, end_record=(24 * 60), save="yesterday")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/batpred/predbat.py", line 6187, in run_prediction
    ) = pred.run_prediction(charge_limit, charge_window, discharge_window, discharge_limits, pv10, end_record, save, step)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/batpred/config/predbat.py", line 1721, in run_prediction
    pv_now += pv_forecast_minute_step[minute + offset]
              ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
KeyError: 1470

2024-05-18 15:25:08.932726 WARNING pred_bat: ------------------------------------------------------------```

rholligan, 2024-05-18:
+1 with errors,

7.19.1 Changed to ERROR: Exception raised 1440
7.19.2 Changed to ERROR: Exception raised 1470

Restarting appdaemon did not help

Neomancer86, 2024-05-20:
Adding some further logs:

Update complete
2024-05-20 13:16:50.276060 WARNING pred_bat: ------------------------------------------------------------
2024-05-20 13:16:50.276359 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-05-20 13:16:50.276621 WARNING pred_bat: Worker Ags: {'id': '9448d5d3ccb1441aab8099b46afa0bcd', 'name': 'pred_bat', 'objectid': '4bd645716c5044a69c7ffecb95ba4992', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7feb993f0510>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-05-20 13:16:50.276687 WARNING pred_bat: ------------------------------------------------------------
2024-05-20 13:16:50.277999 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 15052, in update_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 15047, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 13937, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 12196, in calculate_plan
    self.calculate_yesterday()
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11993, in calculate_yesterday
    ) = self.run_prediction(charge_limit_best, charge_window_best, [], [], False, end_record=(24 * 60), save="yesterday")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 6203, in run_prediction
    ) = pred.run_prediction(charge_limit, charge_window, discharge_window, discharge_limits, pv10, end_record, save, step)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 2090, in run_prediction
    predict_pv_power[stamp] = round((pv_forecast_minute_step[minute] + pv_forecast_minute_step.get(minute + step, 0)) * (30 / step), 3)
                                     ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 1470

2024-05-20 13:16:50.278105 WARNING pred_bat: ------------------------------------------------------------

springfall2008, 2024-05-20:
I believe the issue if you changed forecast_hours in apps.yaml to something <48 it seems to trigger a bug, perhaps try putting it back to 48. I will try to fix it soon.

RobinCu, 2024-05-20:
That worked for me. I have no idea why I changed forecast_hours; I must have been messing around.

Neomancer86, 2024-05-20:
Confirmed! fixed for me too!

SwiftRR, 2024-05-20:
For the record, I have upgraded successfully through 7.19.0 up in steps to 7.19.5 without anything untoward.
I have checked my apps.yaml file and my forecast hours are 48.
Your comment in apps.yaml is apt, but it would be good to fix it to prevent predbats going awol.
#_Number of hours forward to forecast, best left as-is unless you have specific reason_

Rob

springfall2008, 2024-05-20:
Fixed in v7.19.6

closed 2024-05-20 by springfall2008