springfall2008/batpred #1127: Predbat not working on all updates after v17.19.0 - Exception Raised 1440/1470

Neomancer86, 2024-05-20:
**Describe the bug**
Predbat does not start, no automation and the following errors in status
7.19.1 Changed to ERROR: Exception raised 1440
7.19.2 Changed to ERROR: Exception raised 1470

**Expected behavior**
Predbat to run as normal, i'm having to stay on an older version

**Predbat version**

17.19.1
17.19.2
17.19.3
17.19.4
17.19.5

**Environment details**

- GivEnergy + 9.5 Battery
- Standard HAOS installer 
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
-----------------------------------------------------------
 Add-on: AppDaemon
 Python Apps and Dashboard using AppDaemon 4.x for Home Assistant
-----------------------------------------------------------
 Add-on version: 0.16.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.3  (amd64 / qemux86-64)
 Home Assistant Core: 2024.5.4
 Home Assistant Supervisor: 2024.05.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[10:38:27] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
Update complete
2024-05-20 13:16:50.276060 WARNING pred_bat: ------------------------------------------------------------
2024-05-20 13:16:50.276359 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-05-20 13:16:50.276621 WARNING pred_bat: Worker Ags: {'id': '9448d5d3ccb1441aab8099b46afa0bcd', 'name': 'pred_bat', 'objectid': '4bd645716c5044a69c7ffecb95ba4992', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7feb993f0510>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-05-20 13:16:50.276687 WARNING pred_bat: ------------------------------------------------------------
2024-05-20 13:16:50.277999 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 15052, in update_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 15047, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
[... 90 lines cut ...]
  battery_charge_power_curve:
    100 : 0.14
    99 : 0.14
    98 : 0.24
    97 : 0.27
    96 : 0.37
    95 : 0.47
    94 : 0.44
    93 : 0.64
    92 : 0.69
    91 : 0.81

2024-05-20 13:16:47.085150 INFO pred_bat: Find discharge curve with sensors sensor.givtcp_fd2311g523_soc_kwh and number.givtcp_fd2311g523_battery_discharge_rate and predbat.status and sensor.givtcp_fd2311g523_battery_power
2024-05-20 13:16:50.101411 INFO pred_bat: Find discharge curve has 15.0 days of data, max days 15
2024-05-20 13:16:50.133677 INFO pred_bat: Note: Can not find battery charge curve (no final curve), one of the required settings for predbat_status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml
2024-05-20 13:16:50.134245 INFO pred_bat: Found 1 inverters totals: min reserve 0.48 current reserve 0.48 soc_max 9.52 soc 9.14 charge rate 0.38 kW discharge rate 3.5999999999999996 kW battery_rate_min 0.0 w ac limit 5.0 export limit 5.0 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
2024-05-20 13:16:50.183584 INFO pred_bat: Base charge    window [ 20-05 05:00:00 - 20-05 16:00:00 @ 0p 97%, 21-05 05:00:00 - 21-05 16:00:00 @ 0p 97% ]
2024-05-20 13:16:50.184154 INFO pred_bat: Base discharge window [ 20-05 00:00:00 - 20-05 00:00:00 @ 0p 100.0% ]
2024-05-20 13:16:50.184502 INFO pred_bat: Will recompute the plan as it is invalid
2024-05-20 13:16:50.185091 INFO pred_bat: Best charge    window [ 20-05 05:00:00 - 20-05 16:00:00 @ 24.21p 97.0%, 20-05 19:00:00 - 21-05 02:00:00 @ 24.21p 97.0%, 21-05 02:00:00 - 21-05 05:00:00 @ 14.52p 97.0%, 21-05 05:00:00 - 21-05 16:00:00 @ 24.21p 97.0% ]
2024-05-20 13:16:50.185991 INFO pred_bat: Best discharge window [ 20-05 13:00:00 - 20-05 13:30:00 @ 14.84p 100.0%, 20-05 13:30:00 - 20-05 14:00:00 @ 14.84p 100.0%, 20-05 14:00:00 - 20-05 14:30:00 @ 14.84p 100.0%, 20-05 14:30:00 - 20-05 15:00:00 @ 14.84p 100.0%, 20-05 15:00:00 - 20-05 15:30:00 @ 14.84p 100.0%, 20-05 15:30:00 - 20-05 16:00:00 @ 14.84p 100.0%, 20-05 16:00:00 - 20-05 16:30:00 @ 24.52p 100.0%, 20-05 16:30:00 - 20-05 17:00:00 @ 24.52p 100.0%, 20-05 17:00:00 - 20-05 17:30:00 @ 24.52p 100.0%, 20-05 17:30:00 - 20-05 18:00:00 @ 24.52p 100.0%, 20-05 18:00:00 - 20-05 18:30:00 @ 24.52p 100.0%, 20-05 18:30:00 - 20-05 19:00:00 @ 24.52p 100.0%, 20-05 19:00:00 - 20-05 19:30:00 @ 14.84p 100.0%, 20-05 19:30:00 - 20-05 20:00:00 @ 14.84p 100.0%, 20-05 20:00:00 - 20-05 20:30:00 @ 14.84p 100.0%, 20-05 20:30:00 - 20-05 21:00:00 @ 14.84p 100.0%, 20-05 21:00:00 - 20-05 21:30:00 @ 14.84p 100.0%, 20-05 21:30:00 - 20-05 22:00:00 @ 14.84p 100.0%, 20-05 22:00:00 - 20-05 22:30:00 @ 14.84p 100.0%, 20-05 22:30:00 - 20-05 23:00:00 @ 14.84p 100.0%, 20-05 23:00:00 - 20-05 23:30:00 @ 14.84p 100.0%, 20-05 23:30:00 - 21-05 00:00:00 @ 14.84p 100.0%, 21-05 00:00:00 - 21-05 00:30:00 @ 14.84p 100.0%, 21-05 00:30:00 - 21-05 01:00:00 @ 14.84p 100.0%, 21-05 01:00:00 - 21-05 01:30:00 @ 14.84p 100.0%, 21-05 01:30:00 - 21-05 02:00:00 @ 14.84p 100.0%, 21-05 05:00:00 - 21-05 05:30:00 @ 14.84p 100.0%, 21-05 05:30:00 - 21-05 06:00:00 @ 14.84p 100.0%, 21-05 06:00:00 - 21-05 06:30:00 @ 14.84p 100.0%, 21-05 06:30:00 - 21-05 07:00:00 @ 14.84p 100.0%, 21-05 07:00:00 - 21-05 07:30:00 @ 14.84p 100.0%, 21-05 07:30:00 - 21-05 08:00:00 @ 14.84p 100.0%, 21-05 08:00:00 - 21-05 08:30:00 @ 14.84p 100.0%, 21-05 08:30:00 - 21-05 09:00:00 @ 14.84p 100.0%, 21-05 09:00:00 - 21-05 09:30:00 @ 14.84p 100.0%, 21-05 09:30:00 - 21-05 10:00:00 @ 14.84p 100.0%, 21-05 10:00:00 - 21-05 10:30:00 @ 14.84p 100.0%, 21-05 10:30:00 - 21-05 11:00:00 @ 14.84p 100.0%, 21-05 11:00:00 - 21-05 11:30:00 @ 14.84p 100.0%, 21-05 11:30:00 - 21-05 12:00:00 @ 14.84p 100.0%, 21-05 12:00:00 - 21-05 12:30:00 @ 14.84p 100.0%, 21-05 12:30:00 - 21-05 13:00:00 @ 14.84p 100.0%, 21-05 13:00:00 - 21-05 13:30:00 @ 14.84p 100.0% ]
2024-05-20 13:16:50.186588 INFO pred_bat: PV Forecast 0.0 kWh and 10% Forecast 0.0 kWh pv cloud factor None
2024-05-20 13:16:50.187916 INFO pred_bat: Load divergence over 4.0 hours mean 697.36 W, min 250.2 W, max 1249.8 W, std dev 251.41 W, divergence 36.05%
2024-05-20 13:16:50.225492 INFO pred_bat: Yesterday basic charge window best: [{'start': 120, 'end': 300, 'average': 14.52}] charge limit best: [9.52]
2024-05-20 13:16:50.270207 INFO pred_bat: ERROR: Exception raised 1470

RobinCu, 2024-05-20:
Can you try the fix here: https://github.com/springfall2008/batpred/issues/1107

Neomancer86, 2024-05-20:
Confirmed, https://github.com/springfall2008/batpred/issues/1107#issuecomment-2120544185 fixed this for me.

I tried to update from the GUI but updated in apps.yaml resolved it.

springfall2008, 2024-05-20:
Fixed in latest release v7.19.6

closed 2024-05-27 by springfall2008