springfall2008/batpred #1131: Carbon Intensity no appearing in my plan - Something I've not configured correctly

Sycamore967, 2024-05-21:
**Describe the bug**
I've enabled Carbon intensity but don't get figures in the plan.  I'm sure it's something I've missed - can anyone point me to where I should be looking?

**Expected behavior**
Figures in the plan

**Predbat version**

7.19.5

**Environment details**

I've reviewed the latest GE Apps.yaml and can't see I've missed the carbon entry, but I've attached mine just in case.

[apps (3).zip](https://github.com/springfall2008/batpred/files/15388717/apps.3.zip)

**Screenshots**
This is my plan

[image]

Home Costs is empty and if I select an entity I get an error

[image]

The integration looks ok though, it's v 0.0.3

[image]

springfall2008, 2024-05-27:
Are you using the integration fork as mine is 0.2.0 - should be this one: https://github.com/jfparis/sensor.carbon_intensity_uk

Sycamore967, 2024-05-28:
Hi Treffor,

Well that's fixed part of it - I get a chart now
[image]
and the figures are coming through
[image]
But The plan is still missing it
[image]
Anywhere else I can look to see why the metric isn't being pulled in?

johnleeshore, 2024-06-01:
See Trefors video at https://www.youtube.com/watch?v=uVUnrESRZXs&t=575s @2 mins 25

springfall2008, 2024-06-02:
Did you remove the old version of carbon intensity too?

Sycamore967, 2024-06-03:
Hi just an update, yes I removed the old version of Carbon intensity, but it still didn't spring in to life.  However I then updated Prebat to a newer version and it started working.  I didn't change anything else and it hadn't been working since my last post, so I can't say for sure why it now works ok.  I can deffo say I had installed the wrong version of Carbon intensity first, but uninstalling and re-installing and several reboots of HAOS didn't fix it, but it's all good now.