springfall2008/batpred #1135: Urgent- Predbat constantly charging

Sycamore967, 2024-05-25:
**Describe the bug*
Everything was fine last night. Woke up to find Predbat has set a constant charge window and has been since the end of my intelligent window. A restart hasn’t fixed it. 

**Expected behavior**
Normal ops

**Predbat version**

7.19.7

**Environment details**

GE Hybrid 3.6
Two 9.5 batteries 
Octopus intelligent 

**Screenshots**
Attached

**Log file**
On its way
[image]
[image]
[image]

RobinCu, 2024-05-25:
There seems to be widespread problems with the Octopus API.

This fix posted in the Predbat Facebook group should work as a temporary solution:
disable the following two lines in your apps.yml file:

  #metric_octopus_import: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)'
  #metric_octopus_export: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate)'

and add the following (times / amounts need to be adjusted to your tarfiff)
example:
  rates_import:
    -  start: "02:00:00"
       end: "05:00:00"
       rate: 14.7987
    -  start: "05:00:00"
       end: "16:00:00"
       rate: 24.6645
    -  start: "16:00:00"
       end: "19:00:00"
       rate: 34.5303 
    -  start: "19:00:00"
       end: "02:00:00"
       rate: 24.6645       
  
  rates_export:
    -  start: "02:00:00"
       end: "05:00:00"
       rate: 5.2591
    -  start: "05:00:00"
       end: "16:00:00"
       rate: 14.8507
    -  start: "16:00:00"
       end: "19:00:00"
       rate: 24.9907
    -  start: "19:00:00"
       end: "02:00:00"
       rate: 14.8507

Or drop Predbat into read-only.

Sycamore967, 2024-05-25:
Oh man, thanks so much.  I don't do Facebook, so thank you for posting so quickly.

springfall2008, 2024-05-25:
The api seems to be back now

Sycamore967, 2024-05-26:
Thanks all the api has been ok since yesterday lunch time

closed 2024-05-26 by Sycamore967