springfall2008/batpred #1139: Issue with AppDaemon decouple update

nickgee31, 2024-05-26:
**Describe the bug**
At 11:30 this morning, my Predbat status changed to Changed to Warn - Inverter 0 write to inverter_mode failed. I have reverted back to V7.19.7 and all is ok now. I've attached the log and it appears to be this that is the issue: "Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/services/select_option"

**Predbat version**

V7.20.0

**Environment details**

- GivEnergy AC3 + 8.2 Battery
- Standard HAOS installer 

[predbat.log](https://github.com/springfall2008/batpred/files/15447256/predbat.log)

jacekowski, 2024-05-26:
It seems to be calling incorrect endpoint,  it should be calling http://supervisor/core/api/services/select/select_option

springfall2008, 2024-05-27:
This is fixed now in the latest release 7.20.4

closed 2024-05-28 by nickgee31