springfall2008/batpred #1149: 'Inverter' object has no attribute 'get_state_wrapper' on 7.20.x for Solis

marcoaddario, 2024-05-28:
**Describe the bug**
Starting from the new 7.20.x version the execution stops and raises and exception

**Predbat version**

7.20.x (until 7.20.4)

**Environment details**

- Solis Inverter
- Standard HAOS installer

**Screenshots**
These are the relevant lines of the log:
```
...
2024-05-28 08:59:37.757990 INFO pred_bat: Inverter 0: Type GS Ginlong Solis
2024-05-28 08:59:37.759007 INFO pred_bat: WARN: Inverter 0: Using inverter type GS - not all features are available
...
2024-05-28 08:59:42.970747 INFO pred_bat: Found 1 inverters totals: min reserve 0.28 current reserve 0.28 soc_max 7.1 soc 7.03 charge rate 3.5999999999999996 kW discharge rate 3.5999999999999996 kW battery_rate_min 100.0 w ac limit 4.6 export limit 4.6 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
2024-05-28 08:59:43.025985 INFO pred_bat: Base charge    window [ 29-05 00:00:00 - 29-05 05:30:00 @ 0p 100%, 30-05 00:00:00 - 30-05 05:30:00 @ 0p 100% ]
2024-05-28 08:59:43.027088 INFO pred_bat: Base discharge window [ 28-05 00:00:00 - 28-05 00:00:00 @ 0p 100.0%, 29-05 00:00:00 - 29-05 00:00:00 @ 0p 100.0% ]
2024-05-28 08:59:43.027938 INFO pred_bat: Reset inverter settings to safe mode (set_charge_window=True set_discharge_window=True force=)
2024-05-28 08:59:43.028861 INFO pred_bat: Adjust idle time, charge 00:00:00-00:00:00 discharge 00:00:00-00:00:00
2024-05-28 08:59:43.029901 INFO pred_bat: Adjust idle time computed idle is 08:55:00-23:59:00
2024-05-28 08:59:45.053075 INFO pred_bat: Switch sensor.predbat_GS_0_scheduled_charge_enable is now off
2024-05-28 08:59:45.054125 INFO pred_bat: Inverter 0 Wrote scheduled_charge_enable to False successfully and got off
2024-05-28 08:59:47.085386 INFO pred_bat: Inverter 0 Wrote 0 to charge_start_hour, successfully now 0
2024-05-28 08:59:49.128397 INFO pred_bat: Inverter 0 Wrote 0 to charge_start_minute, successfully now 0
2024-05-28 08:59:51.174659 INFO pred_bat: Inverter 0 Wrote 0 to charge_end_hour, successfully now 0
2024-05-28 08:59:53.218544 INFO pred_bat: Inverter 0 Wrote 0 to charge_end_minute, successfully now 0
2024-05-28 08:59:53.221613 INFO pred_bat: Inverter 0 Turning off scheduled charge
2024-05-28 08:59:53.224763 INFO pred_bat: Inverter 0 Current Target SOC is 100 already at target
2024-05-28 08:59:53.226859 INFO pred_bat: ERROR: Exception raised 'Inverter' object has no attribute 'get_state_wrapper'
```

**Log file**
full log attached
[predbat.log](https://github.com/springfall2008/batpred/files/15466019/predbat.log)

springfall2008, 2024-05-28:
I've found the issue, will be in the next release

springfall2008, 2024-05-28:
Fix release in 7.20.5

marcoaddario, 2024-05-29:
That's great! thanks a lot Trefor. perfectly fixed.

closed 2024-05-29 by marcoaddario