springfall2008/batpred #1155: ERROR: Exception raised float division by zero

dandwhelan, 2024-05-31:
**Describe the bug**
ERROR: Exception raised float division by zero.

I accidentally updated to 20.6 from 19.7 and predbat no longer works. 
[apps.txt](https://github.com/user-attachments/files/15518795/apps.txt)
[appdeamontwo.txt](https://github.com/user-attachments/files/15518796/appdeamontwo.txt)
[appdaemonone.txt](https://github.com/user-attachments/files/15518797/appdaemonone.txt)

This appears all the time and predbat can't proceed. 

**Expected behavior**
The plan to update 

**Predbat version**

7.20.6

**Environment details**

- Inverter 5kw giv gen2 and 8.2kw battery
- Standard HAOS installer 
- Anything else?

I updated to 20.6 and after that it stopped working. 
**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

dandwhelan, 2024-05-31:
I tried downgrading but I still get the same error. I've no idea what's going on. Prob something wrong with my apps.yaml?

gcoan, 2024-05-31:
Have you tried restarting appdeamon and/or home assistant, can sometimes cure these kind of issues

dandwhelan, 2024-05-31:
> Have you tried restarting appdeamon and/or home assistant, can sometimes cure these kind of issues

Yeah I have that ago a few times. No joy

dandwhelan, 2024-05-31:
I fixed this by copying the latest apps.yaml from github and pasting it into my environment. Weird as I had made no changes...

closed 2024-05-31 by dandwhelan