springfall2008/batpred #1158: Further extensions to solcast integration

gcoan, 2024-06-01:
**Is your feature request related to a problem? Please describe.**
Have watched the launch video for the new solcast direct integration option https://www.youtube.com/watch?v=K0KOMqoN-WI and there's a couple of things I'd like to see added 

**Describe the solution you'd like**
The today and tomorrow sensors are useful, especially as they contain the 10% estimates that are a bit painful to get out of the solcast data structures.
Also though I'd like to see d3 and d4 pv sensors as I often look at the longer range solar forecast 

The other thing I'd really like to do is continue with the solar forecast in the energy dashboard. It's straight forward to add for solcast so don't know if it's a matter of some extra template sensors or similar?

**Describe alternatives you've considered**
Continue as is with one of the forks of the solcast integration 

Adopt an open source or other provider of solar forecasts

SwiftRR, 2024-06-02:
I agree with Geoffrey's suggestions.

For additional sensors, I would like to see additional sensors from d3 up to d7 which would reflect those that were available in the old Solcast integration. I like to display these as entities in a simple card which gives a useful perspective on longer range solar forecast.

I have upgraded to v7.21.3 and the Solar chart is coming along well.

Rob

gcoan, 2024-06-02:
Been trying to find out how to feed forecast data into the ha energy dashboard and not getting anywhere so may have to study the code for the solcast integration or forecast.solar to work out how they do it.

A big advantage for me of feeding the ha energy dashboard (other than it follows the HA standard rather than having a custom apex chart) is that you can scroll back through history and see historical forecast vs actuals

Thanks for the extra day sensors in the latest release

gcoan, 2024-08-16:
Think I’ll close this as these are all options in the Solcast integration which is being maintained under BJreplay and can be used with Predbat.  Predbat does the basic solar forecasting, if you need more complex things, use the Solcast integration

closed 2024-08-16 by gcoan