springfall2008/batpred #1167: v7.21.5 on Appdaemon in Docker - Solcast cloud issue

PianSom, 2024-06-02:
Running as per title.

apps.yaml reads - 
[image]
(with string copied from Solcast "Resource Id" field for my site)

Log file shows- 
```
2024-06-02 16:08:51.252399 INFO pred_bat: Using solcast cloud
2024-06-02 16:08:51.253290 INFO pred_bat: Return cached data for https://api.solcast.com.au//rooftop_sites age 151.2 minutes
2024-06-02 16:08:51.253963 INFO pred_bat: Warn: No solar data has been configured.
```

springfall2008, 2024-06-02:
The API Key is not the resource ID, its something different you get from your account:

[image]

PianSom, 2024-06-02:
Apologies for being dim!

closed 2024-06-02 by PianSom