springfall2008/batpred #1178: Afternoon discharge on IOG

nickgee31, 2024-06-04:
**Describe the bug**
Quite often there is a short discharge in the afternoon when on the IOG tariff. Solar will then charge the battery back up to 100%. There doesn't seem to be any change in behaviour with having the 'Inverter Hybrid' option toggled either off or on.

**Expected behavior**
That the discharge is left until very late in the day in case it is required. Also is inefficient to force discharge and then charge battery back up with solar.

**Predbat version**

V7.22.2

**Environment details**

- GivEnergy Gen2 5.0kW + 9.5kW battery
- Predbat Addon

**Screenshots**

[image]
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/15550392/predbat.log)

springfall2008, 2024-06-04:
Can run you with debug enabled and provide that logfile please?

nickgee31, 2024-06-04:
No problem, just run that for you. There's a force discharge scheduled for tomorrow at 15:30 too. Let me know if you need anything else

[image]

[predbat-2.log](https://github.com/user-attachments/files/15569382/predbat-2.log)

springfall2008, 2024-06-05:
Not sure I get it, the calculation claims the export at 15:30 improves the cost slightly but it really can't do.

Do you have the power chart, can you show it for that period too please?

springfall2008, 2024-06-05:
Okay I think I sort of understand the logic now, you could discharge more than the planned nightly discharge, but in the 10% scenario doing so would run out of battery.

However some extra discharge earlier in the case where the battery won't get back to 100% again will have the same sort of impact on the normal scenario but less so on the 10% case as the load will be higher and that comes off the same export amount.

I'm thinking that ignoring metric keep in the 10% scenario maybe a good idea, will have a tinker when I get time

springfall2008, 2024-06-05:
Actually metric keep is already ignored in the 10% case...

Can you try something and adjust 'metric_battery_value_scaling' down to say 0.8 and see what plan you get?

nickgee31, 2024-06-05:
Sorry just got round to trying it, slot had moved to 16:00.

Before: 
[image]

After: 
[image]

nickgee31, 2024-06-05:
Think I spoke too soon, it came back

[image]

springfall2008, 2024-06-05:
Interesting, I have some ideas on how to fix it, but its not a bug as such it relates to how the 10% scenario works (I think).

nickgee31, 2024-06-05:
No worries, it doesn't cause too much of a problem really but just thought it was worth mentioning. Let me know if I can be of any help at all

springfall2008, 2024-06-06:
Would you mind testing the version on 'main' (just select main from the update drop down menu) and see if it changes at all?

nickgee31, 2024-06-07:
Apologies only got round to testing this today, unfortunately still an afternoon discharge today

springfall2008, 2024-06-23:
There is a new release that's worth trying

closed 2024-11-17 by nickgee31