springfall2008/batpred #1180: Solar power not showing on chart

Emmoore103, 2024-06-05:
**Describe the bug**
I have added the new pv daily prediction chart and it works except for the PV output line (orange) that is just a flat line.

I made sure the sensor was correct (format - sensor.givtcp_xxxxxxxxxx_pv_power). This I copied from Entities in GivTCP. I checked there was an output to the entity which there is. 

**Expected behavior**
I expect the graph to show actual PV input.

**Predbat version**

7.22.2

**Environment details**

- Givenergy 
- Standard HAOS installer
- GivTCP is in MQTT - not sure if that is relevent.

**Screenshots**
[image]

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

Emmoore103, 2024-06-05:
Thank you.
That has done the job.
Interesting that it is understandably jagged due to the passing cloud etc.
I wonder if it would be possible to make an entity that smooths out the graph by some sort of averaging. Probably difficult and not that important. I can at least see generally what is happening.

Emmoore103, 2024-06-05:
Just to show what I am now getting.

[image]

I think this is now correct, but any comments greatly appreciated.

springfall2008, 2024-06-05:
Looks right, you can try a longer average period e.g

    group_by:
      func: avg
      duration: 30m

Emmoore103, 2024-06-05:
Hi Trefor.

Thank you for confirming and for the code to try.

Brilliant work as always.

Kind regards
Eric

gcoan, 2024-06-05:
Here's my attempt to get this chart working.

[image]

The bottom is the 'out of the box' chart showing one PV array which works fine.

But I have 3 arrays so really would like to see all 3 of them on the same chart, the top is the adapted chart showing 2 of my arrays. Once I get this working I can then add the 3rd.
I can include 3 arrays as lines by just duplicating the entity definition in the chart but I really want the array solar generations to be stacked one above each other, but when I do this I end up with the 2 solar, the PV and PV10% values all being stacked which is clearly not right!

The best I have got to is to change the group_by for the predbat_pv_today to 'diff', but this only works for one or other of the predbat_ areas, can't get both to work.

Any ideas @springfall2008 ?

```yaml
type: custom:apexcharts-card
header:
  title: Solar forecast
  show: true
  show_states: true
  colorize_states: true
apex_config:
  chart:
    height: 300px
  tooltip:
    enabled: true
    shared: true
    followCursor: true
all_series_config:
  type: column
graph_span: 24h
span:
  start: day
stacked: true
update_interval: 5m
yaxis:
  - id: capacity
    show: true
    opposite: true
    decimals: 0
    max: 100
    min: 0
    apex_config:
      tickAmount: 10
  - id: kWh
    show: true
    min: 0
    apex_config:
      tickAmount: 10
  - id: header_only
    show: false
series:
  - entity: sensor.g_xxx_pv_power
    name: G Solar Power
    stroke_width: 2
    float_precision: 2
    color: Orange
    yaxis_id: kWh
    unit: kW
    extend_to: now
    show:
      legend_value: true
      in_header: false
  - entity: sensor.h_yyy_pv_power
    name: H Solar Power
    stroke_width: 2
    float_precision: 2
    color: Red
    yaxis_id: kWh
    unit: kW
    extend_to: now
    show:
      legend_value: true
      in_header: false
  - entity: sensor.predbat_pv_today
    name: Forecast
    color: grey
    opacity: 0.3
    stroke_width: 1
    type: area
    group_by:
      func: raw
      duration: 5m
    extend_to: false
    yaxis_id: kWh
    show:
      legend_value: false
      in_header: false
    data_generator: |
      return entity.attributes.detailedForecast.map((entry) => {
            return [new Date(entry.period_start), entry.pv_estimate];
          });
  - entity: sensor.predbat_pv_today
    name: Forecast 10%
    color: grey
    opacity: 0.3
    stroke_width: 1
    type: area
    group_by:
      func: raw
      duration: 5m
    extend_to: false
    yaxis_id: kWh
    show:
      legend_value: false
      in_header: false
    data_generator: |
      return entity.attributes.detailedForecast.map((entry) => {
            return [new Date(entry.period_start), entry.pv_estimate10];
          });
  - entity: sensor.sum_solar_energy_today
    yaxis_id: header_only
    name: Today Actual
    stroke_width: 2
    color: Orange
    show:
      legend_value: true
      in_header: true
      in_chart: false
  - entity: sensor.predbat_pv_today
    yaxis_id: header_only
    name: Today Forecast
    color: Grey
    show:
      legend_value: true
      in_header: true
      in_chart: false
  - entity: sensor.predbat_pv_today
    yaxis_id: header_only
    attribute: remaining
    name: Remaining
    color: Grey
    show:
      legend_value: true
      in_header: true
      in_chart: false
```

springfall2008, 2024-06-06:
There is a stacked option but I think its global (the entire chart).

Maybe create chart string 1 and then the total as that's the same thing?

closed 2024-08-06 by springfall2008

gcoan, 2024-08-06:
I gave up with trying to show all the different inverter power outputs stacked together and just created a template sensor for all 3 array power outputs and charted that