springfall2008/batpred #1181: Suspected typo in code

rossetf, 2024-06-05:
I suspect line 4048 of predbat.py (v7.22.2) should target discharge_stop_service its the last line from the snippet below:-

```  if self.inv_has_service_api:
            if target_soc > 0:
                service_data = {
                    "device_id": self.base.get_arg("device_id", index=self.id, default=""),
                    "target_soc": target_soc,
                    "power": int(self.battery_rate_max_charge * MINUTE_WATT),
                }
                self.call_service_template("charge_start_service", service_data)
            else:
                service_data = {"device_id": self.base.get_arg("device_id", index=self.id, default="")}
                self.call_service_template("charge_stop_service", service_data)

    def adjust_discharge_immediate(self, target_soc):
        """
        Adjust from discharging or not discharging based on passed target soc
        """
        if self.inv_has_service_api:
            if target_soc > 0:
                service_data = {
                    "device_id": self.base.get_arg("device_id", index=self.id, default=""),
                    "target_soc": target_soc,
                    "power": int(self.battery_rate_max_discharge * MINUTE_WATT),
                }
                self.call_service_template("discharge_start_service", service_data)
            else:
                service_data = {"device_id": self.base.get_arg("device_id", index=self.id, default="")}
               self.call_service_template("charge_stop_service", service_data)

```

springfall2008, 2024-06-05:
Actually its intensional as there is no discharge_stop_service, the assumption being that stop is ECO mode

rossetf, 2024-06-05:
Ah that makes sense.  Thanks sorry for the false bug

closed 2024-06-05 by rossetf