springfall2008/batpred #1190: Agile import w/loss calculation incorrect with plung pricing

Surb109, 2024-06-07:
**Describe the bug**
With Octopuse during price plunging the Import rate with loss caluclulation seem to be incorrect.

e.g. when Import rate is -2.03 with a 5% battery loss charge, the value is showing as -2.14

**Expected behavior**
With Import rate -2.03 with a 5% loss expect Import rate to be -1.93.
The import rate after losses should be higher not lower.

**Predbat version**

v7.22.3

**Environment details**

- Growatt Inverter and Givenergy All-in-one
- Home Assistant Container (Docker)

**Screenshots**
[image]

**Log file**

gcoan, 2024-06-07:
I raised the same problem a couple of months ago, see https://github.com/springfall2008/batpred/issues/881

In short, the calculation is correct

Surb109, 2024-06-07:
Thanks for the pointer, I did try and search for similar issues before posting, but didn't find anything.

I've taken alook at #881 and I'm still trying to understand the logic. 
If the maths has been proven to be correct, I'll leave it at that.

closed 2024-06-07 by Surb109