springfall2008/batpred #1199: Limit grid import (capacity tariff)

djfanatix, 2024-06-09:
**Is your feature request related to a problem? Please describe.**
In Belgium (and Norway too) we have to pay extra when you have a high grid import. (Capacity tariff)
This is measured by the highest peak per month during a 15 minutes period.

**Describe the solution you'd like**

Is there a possibility to have a setting to limit grid import? 
(For example when charging battery from grid max 2500w, but no limit when charging from solar?)
A discussed, a HA entity (input number), which sets the grid import limit, and can be changed with automations.

gcoan, 2024-06-09:
From a source discussion item https://github.com/springfall2008/batpred/discussions/1196

chalfontchubby, 2026-08-29:
Some design notes from a discussion, not a commitment to build yet - recording so the thinking isn't lost.

**Not as simple as it first looks.** Predbat already has `export_limit` (a fixed watt cap on grid export, enforced as a physical clamp inside the per-minute prediction loop) which looks like a ready-made symmetric case for a grid import cap. But the original ask here specifically wants "limit charge from grid, no limit when charging from solar" - and Predbat's charge windows routinely blend PV and grid within the same window (e.g. overnight top-up plus leftover morning PV), so there's no clean per-minute PV/grid split the way `export_limit`'s clamp has. A *blanket* total-import cap (no PV/grid distinction) would be roughly as easy as `export_limit`; the PV-exempt version as originally spec'd is genuinely harder.

**Billing is an average over a window, not an instantaneous limit.** The capacity tariff is "highest 15-minute average import in the month" - so a short high-power spike (a hob, say) is fine as long as the rest of that 15-minute window pulls the average back down. That means the right signal to react to is "projected average if the current trend holds for the rest of this window," not "is instantaneous load high right now" - which sidesteps the much harder problem of predicting how long a load spike will last.

**It's also a ratchet, which simplifies things further.** Once a peak is set for the month, reapproaching it costs nothing - only setting a *new, higher* peak matters. So the system doesn't need a static conservative cap at all; it only needs to intervene at the moment it's about to exceed whatever the running peak-so-far is.

**Where this lands, scope-wise:** rather than Predbat trying to track billing-cycle boundaries, percentage headroom, or when to reset the peak (all genuinely awkward - billing windows and reset dates vary by tariff/utility, and getting them wrong risks the whole feature being wrong in a way that isn't visible until the bill arrives), the cleaner design pushes that policy out to the user's own automation, matching how Predbat generally prefers to expose state rather than own external policy decisions:

- Predbat tracks and publishes a sensor: the peak 15-min average import observed since last reset.
- Predbat accepts an external target number (e.g. an `input_number`) and throttles charge rate to try to stay under it.
- A reset mechanism (button/service) lets the user's own automation clear the tracked peak once they've noted it for billing purposes, and set a new target however they like (fixed value, % of last peak, whatever their tariff logic requires) - Predbat itself never needs to know about billing months, quarter-hour alignment, or percentages.
- The peak needs to survive a Predbat restart (Storage component, per project convention).

This also naturally handles a real gotcha: if a load (e.g. an EV charger) isn't visible on whatever sensor feeds the "total load" figure, the reactive throttle can't see it and won't react to it - that's a measurement gap that has to be solved by getting that load onto the same total-import reading (e.g. combining `car_charging_energy` with house load), not something more clamp logic can fix.

**Suggested minimum first step**, if anyone wants to pick this up: a flat grid-vs-solar-blind `import_limit` (mirroring `export_limit`'s shape) as a first cut, with the reactive average-tracking/external-target design as a follow-up rather than trying to do the PV split from the start.

gcoan, 2026-08-29:
Thanks for the thoughts Rik, pushing all of the complexity of billing and measurement cycles out to external automations makes sense.  

One caveat on Predbat calculating the peak power draw though, it can only ever be as accurate as the inverter poll frequency. My inverters for example poll every 30 seconds so it won't catch every single spike as accurately as the home energy meter.

Could do with some more "I want this" feedback from the community to confirm that this is worth progressing.  No feedback has been received since this was raised 2 years ago.

Maybe an interim solution could be a simple import_limit that mirrors export_limit as being a hard grid import limit that you want predbat to model that can't be exceeded, e.g. due to DNO fuse size

https://springfall2008.github.io/batpred/apps-yaml/#inverter-control-configurations

Again, would need community input to see if worth building