springfall2008/batpred #1203: Possibility to use a sensor as forecasted load

djfanatix, 2024-06-10:
**Is your feature request related to a problem? Please describe.**
No

**Use Case**
Not every 'Tuesday' has the same home load profile (shift work), depending on presence. But we have some continuity in the load.
Proposal: make a home assistant sensor which defines the predicted load of the day.

**Improvement**
- Expert feature: Use a Home Assistant template sensor as a forecast load of the home. The user can make an automation depending on calendar, presence or else to define the template.
For Example:

sensor has template: `[350, 350, 350, 350, 350, 450, 450, 700, 700, 350, 350, 350, 350, 350, 350, 350, 350, 500, 1200, 1200, 1200, 550, 450, 350] `

Each value is the projected power consumption for that hour of the day.

But can also be the remaining power consumption for that day 

`      - name: "Load average"
        unique_id: load_average
        state: >-
              {% set current_hour = now().strftime('%H') | int %}
              {% set data = states('sensor.load_average_raw', 'data') | from_json %}
              {% set values = data.data %}
              {% set values_to_remove = current_hour - 1 %}
              {% if values_to_remove > 0 %}
                {% set values = values[values_to_remove:] %}
              {% endif %}
              {{ values }}`

springfall2008, 2024-06-11:
Interesting idea, but the load isn't just one figure, would scaling the entire day really represent the profile?

Have you looked at PredAI yet?

chalfontchubby, 2026-10-19:
_Posted by Claude on behalf of @chalfontchubby._

Closing as already available: `load_forecast` in `apps.yaml` takes your own sensor/attribute as the load forecast, and `load_forecast_only` uses it instead of history. See [apps.yaml](https://springfall2008.github.io/batpred/apps-yaml/).

closed 2026-10-19 by chalfontchubby