springfall2008/batpred #1204: 'load actual' in predbat is much higher than the GivTCP sensor value.

philkendall, 2024-06-10:
**Describe the bug**
'load actual' in predbat is much higher than the GivTCP sensor value.
History of predbat.load_energy_h0
[image]
(Ignore the little spike at 20:17; this was caused by me trying to debug and changing a value incorrectly)

History of GivTCP Load Energy Today KwH
[image]

Load scaling is set to 1.0
[image]

This is causing the in-day adjustment graph to show a massive increase in load, which is not reflecting reality.
[image]

**Expected behavior**
With the load scaling at 1.0, I expected the 'load actual' to match that current value provided by GivTCP.

**Predbat version**

7.22.4

**Environment details**

- GIV-AC3.0 and GivEnergy 9.5kWh battery
- Standard HAOS installer
- Anything else?

**Screenshots**
Settings:
[image]
[image]
[image]
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/15778263/predbat.log)

springfall2008, 2024-06-11:
Did you get the edit at all as I don't recognise this string?

2024-06-10 20:19:45.574344 INFO pred_bat: Mins Now: 1215 Latest Entry: 81.4431 Current Entry: 72.5141

springfall2008, 2024-06-11:
This is odd as mine seem identical.

What does apps.yaml set your load_today to?

[image]

springfall2008, 2024-06-11:
Actually I take that back, actually mine are higher its just not as much as yours.

I think I know why, as the GivTCP entities sometimes go down which Predbat doesn't allow...

springfall2008, 2024-06-11:
I'll have a go at fixing this soon, still think its working around a bug in GE kit..

springfall2008, 2024-06-11:
Fix on 'main' for testing

philkendall, 2024-06-11:
Thanks for looking into it so quickly.

Yes, that extra log line was added by me; I was interested in the data that went into the calculation.

I'm not at my PC right now, so I'll give the fix a test tomorrow and report back

philkendallsimba, 2024-06-12:
I've put the fix in, and so far so good. As of 09:45 this morning, the load actual is reading the correct value.

[image]

[image]

I shall continue to monitor.
Thanks.

closed 2024-06-23 by springfall2008