springfall2008/batpred #1206: Clipping Buffer (hybrid inverter)

rholligan, 2024-06-11:
**Is your feature request related to a problem? Please describe.**
I currently experience clipping when solar generation exceeds inverter AC capacity, and there is no or limited battery capacity to make use of inverter DC capacity.

**Describe the solution you'd like**
It would be great to see additional features designed around preventing clipping. Predbat does already attempt to accommodate clipping within the plan, however is closely optimised according to solar forecast and low / negative import rates (as intended!), which can come at the cost of reduced flexibility when e.g. battery is charged, and solar generation exceeds forecast rates. This is particularly evident in summer since the solar forecast looks at average generation across each hour, but in reality there could be peak generation for 10 minutes while there is a break in clouds, then 10 minutes of cloud cover etc.

The feature could work by allocating some overhead / buffer in the battery to accommodate clipping, especially during summer / peak generation months. The buffer should be active during period where generation is set to exceed inverter capacity. Once risk of clipping is over, the battery should revert to regular target SOC (e.g. 100%). The start / end time and amount of spare capacity could be determined based on e.g.:
1)	Manual user setting
2)	Ratio of import to export rate (e.g. allocate a higher buffer when export rate is significantly greater than import)
3)	Sunset / sunrise times, plus an offset

Additionally, battery charge power should be set to maximum throughout this period.

**Describe alternatives you've considered**
I currently run an automation which triggers manual discharge earlier in the day to create this buffer capacity within the battery.

**Additional context**
Add any other context or screenshots about the feature request here.

gcoan, 2024-06-11:
As an alternative solution to reduce clipping, could either make the battery charge rate an externally controllable entity (so could manage via an automation), or (preferable), predbat plans the charging at a rate such that solar charging is spread out over a much longer period rather than always charging at max rate.

Tomorrow being a good example (ignoring the zero rate import period mid day), there is loads of solar forecast and so the batteries (drained down overnight) will be fully charged by 11:00.  Thereafter solar will be exported, but risks clipping.

If the battery was charged lower and slower over a longer period of the day it would reduce the risk of clipping as its charging and exporting for longer
[image]

rholligan, 2024-06-12:
> As an alternative solution to reduce clipping, could either make the battery charge rate an externally controllable entity (so could manage via an automation), or (preferable), predbat plans the [...]

I would actually avoid using charge / power limits during periods where clipping is likely, since these can inadvertently result in clipping also. However, the same effect could be achieved by setting lower SOC targets (solving for clipping buffer), and prioritising all generation to export (battery is only charged at the minimum rate which avoids clipping).

PatrickJanssens, 2024-06-14:
I think the key is to have the battery SOC as low as possible prior to the time clipping is expected (I set it to 10%). As soon as clipping is happening I set an automation to switch off the Eco mode and set the system to Timed Discharge, still with a low battery discharge SOC (this can be done either based on the Solcast forecast, or based on the actual PV production). This way the Excess PV is sent to the battery, but when PV production drops below the inverter capacity the battery is discharging again to increase the buffer for more excess PV. I don't think there is a need to adjust the battery charge rate, as long as it can be configured to only send the excess PV to the battery (seems to work fine on my GivEnergy hybrid inverter using timed discharge). The limiting factor is really the battery capacity, so key is to start with an as low as possible battery SOC, and ensure it can discharge to the grid when no clipping is occuring to make space for more excess PV.
In order to ensure I have a full battery when the sun sets I set my system to revert back to eco mode and disable the discharge schedule once the remaining PV foreceast drops below a certain value (I presently set it to 12 to 15  kWh). This way I'm sure to have a full battery when the sun sets.

springfall2008, 2024-06-22:
I've been trying to fix this by fixing the model for clipping, a new version is being tested on main

rholligan, 2024-06-22:
Sounds good to me!

rholligan, 2024-08-20:
Hi Trefor, clipping prevention is working well in the background now based on forecast PV and inverter limits - thanks for the efforts on that.

Is there any appetite to add further user control around this feature? In an ideal world I'd be able to adjust clipping prevention, especially during the Scottish summers where there can be rather unpredictable bouts of cloud and sun which may not prevent clipping when relying on forecast generation alone.

PatrickJanssens, 2024-08-20:
I don't feel clipping is captured well by PredBat. The solar forecast are not accurate enough to capture actual peak loads, especially on day with scattered clouds. I presume solcast averages out the peaks so have frequently experienced the battery to be full when "unpredicted" sunshine appears. For me personally I have implemented an automation that discharges the battery in the morning to ensure there is sufficient battery capacity available for clipping. My target is to discharge to close to 15% of my battery capacity, so I still have some battery charge available for sudden loss of PV or sudden high loads. This automation is triggered by the overal solar production forecast for the day, meaning if the solar forecast is low I keep the battery charged, Towards the end of the day I stop discharging the battery, triggered by the remaining solar production forecast for the day to ensure I have sufficient battery charge to cover the evening. This has worked quite well for me. I have been playing around with the settings for these two values. Right now I have set the values to 25 kWh for the full day forecast and 12 kWh for the remaining PV forecast. That works quite well for me and I have managed to capture most of the excess clipping since I implemented this. Some examples below:

A perfect sunny day:
[image]
(Note I recovered some 65% battery charge from excess solar)

Scattered clouds day:
[image]

Cloudy day:
[image]

Note that I am on the Octopus Intelligent Go tarrif, so I do not worry about the price paid for my solar production, and I want to maximise the use of the cheap overnight tarriff. But I'm sure there room to optimise this further, especially for those on other tarriffs. Key for me would be to base the required buffer capacity for clipping on more conservative (in this case high) peak solar predictions.

gcoan, 2024-08-20:
> It sounds like addition of a "clipping buffer" could help to simplify things for you here as well. It sounds like having some degree of control over this clipping prevention would be important to you, to help tackle the more complex dual inverter setup.

Yes it would be useful, in effect I am creating my own "clipping buffer" in the battery by only letting the batteries charge partially overnight.

I think to be a complete solution there's several additional components I'd like to see within Predbat:

1. A configurable "clipping buffer" per inverter with the size of the buffer being user defined either as a fixed number or as a helper entity that I could control via an automation depending on solar forecast

2. Ability to vary the charging rate dynamically per inverter. At the moment (apart from overnight slow charging when switch.predbat_set_charge_low_power is turned on) Predbat uses either the inverter max rate, or an over-ride figure in apps.yaml, but you can't set a helper entity name in apps.yaml. If you could then programmatically you could set whether Predbat is to charge at max rate or trickle charge during the day.

3. Linked to 2, when the charging rate is set to zero, the battery pause mode is set to PauseCharging.  This is needed to get around inverters/firmwares that continue to charge at a slow rate when the charge rate is set to zero.

What I would anticipate is something like:

Overnight, charge rate is set to maximum so normal grid charging & discharging.  Clipping buffer set to (say) 2kWh so Predbat leaves 2kWh spare capacity in the battery.  Normal predbat optimisation of charge and discharge periods so that the battery gets charged overnight at the cheapest rates.

Sunrise, my automation sets the charge rate helper entity to zero.  Predbat sets inverter charge rate to zero and enables PauseCharge mode on the battery.

From sunrise to say 10am, all solar is exported.

At 10am, my automation sets the charge rate helper entity to 900W.  Predbat sets the inverter charge rate to 900W and disables PauseCharge mode.  The battery starts charging at 900W, this slow charge reduces the chance of clipping occurring through the solar peak.

At 3pm, my automation sets the charge rate helper entity to the inverter max rate, Predbat mirrors that on the inverter.

I have two inverters one of which is East facing and one of which is East and West facing so I would follow the above timings for inverter 1, but for inverter 2 I wouldn't start the trickle charging until about 1pm as the peak for that array is later.

If you are using the Solcast integration, the latest versions includes the site (array) specific breakdown of peak solar times in the peak time Solcast sensor so these can be easily used in an automation.  Taking account of either the estimate or estimate10 times if you wish to:
[image]

rholligan, 2024-09-27:
Hi Trefor, any thoughts on adding capabilities to adjust handling of clipping?

It's a sunny day with clouds here today meaning I'm often generating higher peak power than forecast, and with a lot of negative slots early this morning, my plan has resulted in battery being full over mid day / peak generation period resulting in clipping. 

It would have been great to have some control over this, to avoid the clipping.

springfall2008, 2025-01-01:
The cloud model should handle this, it just depends on your skew between the 10% and average scenario.

As this ticket is old I'm going to close it, please re-open if the issue is there on latest.

rholligan, 2025-01-04:
Hi Trefor, appreciate you're likely trying to tidy up the queue a bit here. 

This request is not a current issue per se, rather it's a feature request (still relevant to latest version); the current cloud model does not allow for any manual weighting or adjustment of the skew 10% and average scenario (unless I missed something in the docs). 

This request is to add some additional control over how clipping is handled (over and above the existing functionality - which for me does work as intended).

johnwb87, 2025-01-07:
Just seen this following the recent comments. I agree now an enhancement idea for the coming summer. I think the part missing from the current clipping model is the charge low power aspect.

A single discharge ahead of the peak window and then a low power charge through it would be a more elegant, efficient and cost effective solution.

Limiting the charge rate is what I started doing with automations before going fully predbat. Having the decision on when that happens calculated taking into account all the variables that predbat uses would be maximally effective.

I think it is the charge low power logic but applied between demand/idle/eco and freeze export across multiple slots.

rholligan, 2025-01-07:
> Just seen this following the recent comments. I agree now an enhancement idea for the coming summer. I think the part missing from the current clipping model is the charge low power aspect.
>
> A [...]

Glad to hear there are others in a similar position.

I actually prefer to avoid limiting the charge rate, because in the case of higher than predicted generation, this can also cause clipping in itself (e.g. I have 6.4 kWp panels, and inverter with 5 kW AC + 3 kW DC capacity - if the DC capacity was limited to e.g. 1 kW, clipping could occur at peak generation).

johnwb87, 2025-01-07:
An uninformed limit of 1 kW would be bad but solar peak minus inverter limit could be used as a lowest bound for the charge rate.

predbat already knows the inverter limit but currently only takes max solar from the forecast. In this potential enhancement if we told predbat in apps.yaml the peak solar then when the clipping model comes in it would reference that in then setting a reduced charge rate.

PatrickJanssens, 2025-01-07:
I'm personally not in favour of limiting the charge rate. I don't think it solves the issue and adds another risk of clipping as rholligan pointed out. For me key is to discharge the battery sufficiently before clipping occurs. Any strategy should be focusing on that. I did explain my approach earlier in this forum, so won't repeat here.

johnwb87, 2025-01-07:
> Personally I'm against force discharging the battery to provide room for more solar charging. It adds extra battery cycles and wastes circa 15% power if you charge and then discharge the battery. But if that's a solution for some then let's include it in the set of options that predbat can use

I agree with this based on my observations of predbat actions during clipping events last summer. What I saw was multiple, short full power discharges followed by full power charge from solar. This is incurring avoidable battery losses and also makes our export to the grid more spikey then it needs to be, both sub-optimal things.

My view is that a reduced charge rate is indeed part of the solution and a useful potential addition to the current solution. A reduced charge rate only has no impact if max solar would saturate both the AC and DC limits. If that happens then the inverter has likely been mis-sized or there are some strict export limits imposed by the DNO

A reduced charge rate is anything less than 100%. In the thread we've had examples of 6.4kWp and what looks like around 7.5kWp. So with 5kW of AC + 3kW of DC capacity both systems would benefit from a reduced charge rate in addition to forced export.

cyberkryten, 2025-01-07:
I think a lot of people have very different requirements for the clipping solutions and it is going to be very difficult to satisfy everyone!.  

I have a Hybrid system so there's no clipping issue as such on the charging because the inverter is not involved.

However, I have a 3.68kW export limit but a split E/W system that is capable of peak generation of 8kW so any charge/discharge cycling or charge rate limits would be no use in my situation - my issue is that the battery ends up full early in the day and so the clipped power is lost because I am not allowed to export it.

So, me a clipping solution would be a case of Predbat deciding how much power during the day will be generated above the amount I can use (3.68kW export + House Usage) and then reduce the battery charge by that amount in the overnight off-peak period, so I start the day with a somewhat empty battery then then export + fill the battery during the daytime.

johnwb87, 2025-01-07:
Large array and low export limit is a thinker. Is 8kW your installed capacity so you don’t actually see that peak?

In that scenario you want to be getting as close to max export at all times outside the off-peak window using both the battery and solar and achieve that with no clipping.

So depending on export rates could still discharge as soon as off-peak finishes assuming that still gets you to an SOC that limits clipping as much as possible. Likely also benefits from a longer plan forecast period so predbat can see if another high solar day is forecast. 

Interesting problems to solve with this one.

gcoan, 2025-01-07:
My scenario is large array (actually 3 arrays) but they match my inverter sizing (so no inverter DC clipping) and no DNO export limit.

But practically I do have an export "limit" in that if I do nothing, the batteries are full by 10am and then I could be generating 8-12kW across the 3 arrays.  Together that power output pushes the grid voltage up above the allowable limits and one or two of the inverters shut down as a result of the high voltage.
Voltage drops, inverters start up again, whole cycle repeats.

So basically I get a lot of peak rate throttling of my inverter output.

If instead of charging the batteries up at full rate on solar in the morning I leave them empty/part empty (e.g. with a slow rate charge or export all solar generation), and then during the solar peak I stop limiting the battery charge rate, I can divert 4.8kW of solar power into the batteries instead of the grid and this stops pushing the grid voltage out of tolerance.

So a different set of circumstances than @cyberkryten but a similar set of requirements for Predbat

cyberkryten, 2025-01-08:
I had a lot of issues due to the batteries so also had G100 limit triggers and voltages over 255v causing the Active Power Limit to drop to zero, which complicated the peak summer period a lot on the data side.

I have 10.5kW of panels installed and saw a peak generation of around 8kW in the summer into the batteries - as it is DC coupled and I have three batteries (5kW charge into each) they can take the full charge without issue.  It's hard to tell the actual generation as the SE portal reported 15kW (clearly wrong).

On the sunny days, I pretty much need to not charge overnight at all IF it was going to be sunny.  I had many days where I clipped to 4.5kW production for 8-10 hours though the peak of summer so don't really know how much power I 'lost'.  

The good news is there's several months before this is going to be a problem again ;)

johnwb87, 2025-01-09:
I think what @gcoan and I are saying is that in the first hours of generation, when the batteries are presumably charging, a better solution would be to export upto your export limit even discharging the batteries in this window and only put the potential clipping into the battery once an excess of solar occurs.
Maximising the export at all times is the solution and so depending on inverter arrangements (I’m thinking in terms of my GE Hybrid) reducing the charge rate is part of achieving that. That reduction could scale as the solar forecast increases and have an additional headroom added to it to avoid passing clouds clipping.
The main constraint for all of us is Grid Export whether because of the inverter itself or DNO limits so we have to maximise getting the most kWh down whatever kW size export pipe we have.

cyberkryten, 2025-01-10:
I'm not sure I entirely understand how reducing the charge rate helps?  What matters is how much energy is stored in the battery and so reducing the rate of charge and adding it over a longer period doesn't change that.   My guess is that you actually mean the system should be starting to export early and by doing so this will reduce how fast the batteries fill up (because some power is being sent to the grid) - is that correct?  Again, we may be struggling for terminology/implementation between different inverter/battery types (I'm thinking of this in terms of SolarEdge) as maybe you can force-exporting by saying only allow X kW DC to be stored?

I guess there are a couple of values that will need to be set/calculated to allow a clipping calculation to be accurate and the requirements are probably also different for AC and DC coupled batteries - in a Hybrid system, the DC batteries can fill at the full rate of solar generation whilst there is capacity.  In a system with an AC coupled battery, you are limited by the Inverter for how fast the batteries can be charged.

Of course the absolute key for the calculations is the absolute maximum output.  This will again vary because of Inverter Size, DNO Limit and house consumption and I'm sure it will need a lot of tweaking, to avoid a mistake where we don't have enough battery and end up paying peak prices for electricity!

gcoan, 2025-01-10:
> I'm not sure I entirely understand how reducing the charge rate helps? What matters is how much energy is stored in the battery and so reducing the rate of charge and adding it over a longer period [...]

By reducing the charge rate (or even setting it to zero), you are storing less solar energy in the batteries earlier in the day, so leaving space for later in the day, and as you say, will start exporting solar energy earlier.

To try to give an example, simplifying my own setup.

Inverter max AC throughput 5kW
Battery max charge rate 2.6kW

Nice sunny summer's day, generation starts at 5am, peaks at 12 with 7kW of peak power

If I don't intervene Predbat will let the batteries charge from solar from 5am,  they will be full by say 10am, and then solar will converted to AC, exported, but only up to the 5kW inverter limit.

If instead I don't let the batteries charge at all until 10am, or charge them at a reduced rate, any solar above the battery charge rate will be exported, and at 12am when I am generating 7kW peak then 2.6 of that will be going into the battery and 7-2.6=4.4kW will be exported.

Overall by delaying the charge I will export more and not lose the solar power that would have been clipped during the peak

gcoan, 2025-01-10:
Different behaviours for different inverter types.

Comes back for me to there not being a single way Predbat can handle clipping for us all, not least of which it won't know enough about our different setups.  For your inverters it sounds like a periodic force discharge is a solution, for my hybrid inverters, that has less use because of the over-arching 5kW AC inverter limit.  So if generating 4kW of solar I can only discharge the battery at 1kWh, and above 5kW I am back to clipping the solar generation.

johnwb87, 2025-01-11:
My set up and preferred solution is very similar to @gcoan. I’m pretty sure we both have GE Hybrid inverters so that would be why we are coming at it with the same inverter capabilities in mind.

cyberkryten, 2025-01-11:
Well, it is sunny this morning and the panels were generating nearly 3kW so I decided to have a play around with the various settings on the Inverter.  I set the solaredgemodbus_i1_storage_charge_limit figure to 500 with the mode left as "Maximize Self Consumption".

As it is DC coupled batteries I assumed it was the Grid AC charge limit and I didn't think this would do anything but it does actually restrict the charge power in the same way you have described on the GE Hybrid system.

The batteries went from charging at 2000W to 497W and at the same time, the export rate went from nothing up to 1.4kW so it did exactly as would be needed if the clipping calculations were accurate in predicting charge/export to avoid being left with excess.

However, shortly after, Predbat for some reason changed the discharge rate to 500W to match the charge rate (I tried again with 2000 and it did the same).  That's NOT what I wanted as it will limit self-usage - I'm not sure why they are tied together so I'll have a look through the code and if I can't figure either out I'll open a separate ticket for that (I was actually expecting Predbat to notice the reduced charge_rate value and change it back to the normal figure!)

johnwb87, 2025-01-11:
That sounds like it could be to do with how predbat reads battery rates from the SE inverters. Have a poke around at the sensors for that sort of thing.

cyberkryten, 2025-01-11:
I've never tried using set_charge_low_power because it says it needs an accurate battery charging curve and that is not possible on my system because the three batteries get out of sync in terms of SOC.

I have three batteries all of which can be at different SOC and have a sensor using the calculated average of the three.  If they are all 95% then they will all charge in a similar way, but if one is 90%, another 95% and the third 100% then the charge curve/rate will be very different.

johnwb87, 2025-01-11:
I wondered if the SE inverters worked with low power charge as a shortcut to understanding some of the parameters.

I just had a look in the solar edge apps.yaml template and I wonder if using inverter_limit_charge there instead of through the SE integration would prevent the behaviour you’re seeing of discharge reducing also.
If that works then the automation @gcoan described should work for you and potentially the clipping solution we’ve been discussing as well.

rholligan, 2025-01-13:
For me there are two key elements to clipping strategy for Predbat:
1) Prioritise AC capacity of inverter (consumption or export), then DC capacity. 
2) Ensure there is enough battery capacity to maximise availability of inverter DC capacity, during periods where there is risk of clipping.

Adding some level of user control of these two elements

rholligan, 2025-01-13:
For me there are two key elements to an effective clipping strategy for Predbat (from my perspective, environmental conditions listed below):

1. Prioritise full utilising AC capacity of inverter (consumption or export), then DC capacity. As is clear in the discussion, the appropriate way to realise this might be different for each inverter / system depending on controls available.
2. Ensure there is enough battery capacity to maximise availability of inverter DC capacity.

For me, adding some level of user control across these two elements (actually for me, purely #2) would unlock the level of tuning that could totally eliminate clipping.

Currently, clipping prevention works most of the time, but does not account for e.g. higher generation that forecast, especially in summer when that generation could easily saturate inverter AC capacity.

**Environmental Conditions**
Panels: 6.4 kWp (in Scotland, can see peaks of ~7.5 kW generation if cold and clear in summer)
Hybrid inverter: 5 kW AC + 3.6 kW DC capacity
Battery: 9.5 kWh

rholligan, 2025-01-13:
@cyberkryten  I'm not familiar with the Solar Edge system (I am on GivEnergy), but I did set an automation that monitors battery_charge_rate, and keeps it set at the maximum value (which mitigates any change made by Predbat). Not a perfect solution, but might help if you can't find the appropriate controls otherwise.

johnwb87, 2025-03-05:
Fascinating behavior today. After the last few days of interesting behaviour I decided to turn combine charge slots on. I'm on IOG so I don't miss out on much.

Then today we have this.

[image]
[image]

At first disconcerting to see daytime charging. But looking at GivTCP log Predbat is setting a reduced charge rate, see below, as we had been discussing. So this is positive behaviour. It would potentially optimum if the inverter was left in Demand but with the reduced charge as I've yet to identify how any usage beyond the solar is served. But I may again be coming with a GivEnergy bias on that point.

So anyway, pretty impressive, but could be further optimised both in terms of inverter settings and how it is communicated in the plan.

2025-03-05 11:42:40,145 - GivTCP - write - [INFO ] - Setting battery charge rate 600 was a success
2025-03-05 11:42:50,954 - GivTCP - write - [INFO ] - Setting Charge Slot 1 to: 05:30 - 12:00 was a success
2025-03-05 11:42:51,426 - GivTCP - write - [INFO ] - Setting Charge Schedule to enable was a success
2025-03-05 11:42:51,895 - GivTCP - write - [INFO ] - Setting Battery Pause Mode to Disabled was a success
2025-03-05 11:42:52,361 - GivTCP - write - [INFO ] - Setting Charge Target 66 was a success
2025-03-05 11:45:29,015 - GivTCP - write - [INFO ] - Setting battery charge rate 1200 was a success
2025-03-05 11:50:32,026 - GivTCP - write - [INFO ] - Setting battery charge rate 3600 was a success
2025-03-05 11:50:42,797 - GivTCP - write - [INFO ] - Setting Charge Schedule to disable was a success
2025-03-05 11:50:43,276 - GivTCP - write - [INFO ] - Setting Battery Pause Mode to PauseDischarge was a success
2025-03-05 11:53:07,315 - GivTCP - write - [INFO ] - Setting Charge Schedule to enable was a success
2025-03-05 11:53:07,785 - GivTCP - write - [INFO ] - Setting Battery Pause Mode to Disabled was a success
2025-03-05 11:53:08,267 - GivTCP - write - [INFO ] - Setting Charge Target 100 was a success
2025-03-05 12:00:27,990 - GivTCP - write - [INFO ] - Setting Charge Schedule to disable was a success
2025-03-05 12:00:28,465 - GivTCP - write - [INFO ] - Setting Battery Pause Mode to PauseCharge was a success
2025-03-05 12:00:28,945 - GivTCP - write - [INFO ] - Setting Charge Target 4 was a success
2025-03-05 12:12:37,977 - GivTCP - write - [INFO ] - Setting battery charge rate 400 was a success
2025-03-05 12:12:48,753 - GivTCP - write - [INFO ] - Setting Charge Slot 1 to: 05:30 - 18:00 was a success
2025-03-05 12:12:49,230 - GivTCP - write - [INFO ] - Setting Charge Schedule to enable was a success
2025-03-05 12:12:49,703 - GivTCP - write - [INFO ] - Setting Battery Pause Mode to Disabled was a success
2025-03-05 12:12:50,169 - GivTCP - write - [INFO ] - Setting Charge Target 74 was a success

gcoan, 2025-03-06:
I'm seeing similar, charging during the day at high import rates, which is concerning, but actually when I look, it was actually on a lower charge rate on one inverter and full charge rate on the other.

The dissimilar charge rates I was fine with, the larger battery needed charging up, so full rate is fine, but I didn't really like it being set to charging with pause discharge mode on as if the clouds cover the sun then I'll be grid importing.  Would have preferred to be in demand mode still

Also, does seem to be chopping and changing the plan a lot, swapping modes every few minutes which can't be good for inverter writes
[image]

rholligan, 2025-03-15:
More clipping today due to battery being charged to 100% during peak generation (I have a 5 kW hybrid inverter = 2.5 kW half hourly clipping threshold)

[image]

springfall2008, 2025-03-15:
Can you try the latest release made today?

rholligan, 2025-03-17:
> Can you try the latest release made today?

Certainly! I'm now on version: v8.16.5. Waiting for some more sun to test things out properly :-). 

It does look like tomorrow (18-Mar) will be a good day for generation, however in the current plan it looks like I will experience clipping from 11:30 to 14:00.

Let me know if there's anything else I can do to help triage 👍

[image]

rholligan, 2025-03-18:
Actual plan for today:

[image]

johnwb87, 2025-03-18:
I'm at risk of clipping today I suspect. With the 50% and 10% solcast forecast being basically identical the cloud model doesn't appear to be triggering the clipping model.

I have some head room today because of the amount of carbon metric I use led to some early export.

The main reason I'm replying is I have a similar, seemingly slightly random export shortly after the solar peak.

[image]

Tried to grab a debug from the web interface but it's just sitting there looking at me.

gcoan, 2025-03-18:
> identical the cloud model doesn't appear to be triggering the clipping model.

Is the clipping model only activated as part of the cloud model then @springfall2008 ?

I had some time ago turned off the cloud model because I didn't like the frequent chopping and changing of the plan it caused. And this was before inverter register writes was a "thing", I found it produced an unstable plan execution.

Does it have to be part of the cloud model (if it is)?

rholligan, 2025-03-19:
More clipping expected today, I've included predbat_debug.yaml too in case it's helpful

[image]

[predbat_debug.txt](https://github.com/user-attachments/files/19335999/predbat_debug.txt)

gcoan, 2025-03-19:
I've tried turning cloud model on and its making no difference to my plan, am still seeing the battery being held at 100% and in Demand mode from now until 5pm when load starts to exceed solar generation.

No forced discharge or anything.

Some of the slots I'm generating more than my inverter charge/discharge capacity (2.6kW per slot), but these are not more than my overall inverter DC capacity (5kW per slot)

[image]

johnwb87, 2025-03-19:
> Is the clipping model only activated as part of the cloud model then [@springfall2008](https://github.com/springfall2008) ?

I don't know that the cloud model is related to the clipping model. I actually don't think it is. My meaning was that I was near my export limit and often, because of the cloud model, that triggers the clipping model. Yesterday though the 50% and 90% Solcast where so close the cloud model would have had very minor impact on the predicted PV because of how it use the difference between the 50% and 90% to then apply variability to the PV prediction.

My broader observation from the three of us would be that it seems at some point in the last few releases, and since the previous sunny spell, the clipping model isn't coming into action even in the previously observed "choppy" way.

LeeFarm11, 2025-03-20:
3 phase GIV-3HY-11 with 20kwh batteries.

I also see similar problems in the plan (irrespective of problems with controlling the Inverter. That's a different issue). I do see the plan trying to allow the batteries to be in a relatively low state of charge at the end of overnight cheap rate. But the plan does not make any real attempt to prevent clipping.

On days like today I have to manually set my Inverter to Force Discharge with Discharge rate of 1% (minum allowed). 

Generally from around 9am (or once I am up to about 70-80%) to about 13:15. (I have storage heaters which come on shortly after 13:00 and cheap rate starts at 13:00 although I tell predbat that the cheap rate does not start until 14:00) 

This manual setting will send all excess Solar to the Grid, and discharge 110W from the battery. But once the Solar generation gets to 11kw, the battery will start charging instead of discharging. Currently I am generating 12.5kW with about 750W going to battery.

Here is the current plan. As you can see there is no attempt to prevent the battery getting to 100%.
ALso here is today progress so far with Foece Export active since 07:00. Today was not normal. Overnight the Inverter and Battery stopped talking to each other so predbat plan could not be achieved and instead the system just worked as if there was no battery. I reboot Inverter and Batteries at about 6:45 and all returned to normal.

[image]

[image]

[image]

[image]

rholligan, 2025-03-24:
Still looking like I'll experience clipping today on v8.16.7

[image]

gcoan, 2025-03-24:
others have been sharing their plans, and I have turned cloud model on myself,  but I too am not seeing any signs of Predbat trying to manage battery SoC to reduce clipping

one question I do have is how the clipping logic plays into cost optimisation?   I'm on Cosy with a current cheap import rate of 13.24p (14.58p after losses).  Export rate is 15p which is marginally higher so Predbat pretty much always charges the batteries overnight in the cheap periods and then lets all the solar export during the day.  Cost-wise this makes sense but if I can't export because of clipping it doesn't make sense

theholyfish, 2025-03-24:
Evening all.

Thanks Trefor for keeping predbat updated. 
Amazing work. 

Forget to add I'm on Agile tarrif and fixed 15p export.

To avoid clipping on a sunny days I forced export between 10am and 2pm via predbat.

On cloudy days I don't really mind a little clipping. 

I start to see my first clipping in the middle of Feb. 
9.24KWp 
15 panels south 
7 west. 

In the next week or so I'll start the hottub again. I set it to heat for 1½ in the morning and 1½ in the afternoon. This helps to keep my clipping under control in the summer. 

In solcast I've set my AC 99kw and DC to 11. 04kw
Thanks. 

Today's numbers. 😊

[image]
[image]

theholyfish, 2025-03-28:
Today's export plan looks great today only because unit prices are low today. 

[image]

Thanks.

gcoan, 2025-03-28:
I know @springfall2008 has said he is starting to look at this, but I am seeing quite a lot of clipping happening already and think I need to start intervening.

Has anyone got anywhere on managing predbat via the manual API to reduce clipping?

I'm on Octopus Cosy, at the moment import 13.23p in the cheap periods, 15p export, so its marginally profitable to charge in the cheap and export everything thereafter.

So a typical plan looks like this, charge fully in the cheap periods, export everything during the day:

[image]

And then today, my solar generation looked like this (top graph)

[image]

As you can see starting from about 9:30am to 3pm the generation causes grid voltage to rise (bottom graph) above limit and the inverters shut off, voltage drops, inverter starts back, and the cycle repeats.  Reducing my generated power considerably.

Today's forecast was 35kwh (achieved 27), tomorrow's is 56kWh so will be worse clipping than this.

The benefit between charging on Cosy at 13.23p (effective charge rate 14.58p) and exporting solar at 15p vs charging the batteries from solar is quite small after losses are included, so I plan to stop some of the predbat overnight charging and instead let the batteries solar charge during the peak generation period.

Last year my summer automation put Predbat into read only mode for most of the day and I manually changed battery pause mode and charge rate to stop any charging until 9:30am, then trickle feed charging during the day, then release for evening/night.  I could do the same, but would like to try to do more "inside" predbat if I can.

If I force freeze export for the 4-7am period Predbat is at the moment planning on not letting the battery charge at all from solar:

[image]

Similar if I freeze charge from 4-7am:

[image]

This might improve as the plan rolls forward.

I'm thinking I freeze export from 11pm-midnight and 4am-7am to keep the battery partly empty, then leave predbat freeze exporting until about 10am, then force demand through the peak solar generation AND set inverter_limit_charge to a low rate via the manual API

Anyone tried something like this?

CJDumbleton, 2025-03-28:
My workaround is similar to that of some others in this thread. I use Home Assistant automations to:
- At sunrise, predict if clipping is likely to occur
- If so, make room in the battery by exporting
- If at any point pv power becomes >= inverter limit, set `select.predbat_manual_export` for this half-hour

My approach is not perfect. I am often making room in the battery which is then not filled up. But, overall, I think the workaround makes me money by capturing solar energy that would otherwise be clipped. I have a 5kW hybrid inverter and peak generation of over 7kW.

My automations are here:
https://github.com/CJDumbleton/ha_predbat_automations/

gcoan, 2025-03-28:
I have been playing around with the predbat manual API to set the charge rates of my inverters.

One disadvantage of this approach is of course Predbat doesn't know about future (automation) plans for changing the charge rate.

Via the manual API have set the two inverter charge rates, to 500 and 0 respectively:
[image]

And the predbat plan shows a slower SoC rise (only 1 battery charging at 500W) in the 10-midnight Cosy period,  the forced freeze export in the 4-7 Cosy period and then slower SoC rise thereafter:

[image]

I'm going to add an automation to:
- stop all charging at sunrise
- set inverter 0 (which is the East facing array) charge rate to 1200 at 9:30am
- set inverter 1 (which is split East/West) charge rate to 1200 at 1pm
- set inverter 0 charge rate to 500 and inverter 1 to 0 at sunset ready for the 10-midnight Cosy period, PLUS this also reduces cross charging which is 99% of the time from inverter 0 to inverter 1

This is similar to how I handled clipping last year except I had predbat in read only mode all day and so was directly changing givtcp controls.  Only disadvantage I can see of doing it this way is I am not able to set battery pause mode so there is still a bit of trickle charging of the inverters that occurs when the charge rate is set to 0.

BuhJuhWuh, 2025-03-28:
Starting to see the clipping happening too. Like @gcoan I am on Cosy, so have been a bit surprised to see it merrily charging to 100% every morning regardless of the solar forecast.
[image]
I wondered if _best_soc_max_ could be used to set the battery to charge to say 50% (or whatever), but unfortunately this seems to limit the charging from solar too. So could a similar variable (say _best_soc_grid_max_ or something) be added which would add this functionality, i.e. set the target charge level for grid charging only, but allow charging from solar?

cyberkryten, 2025-03-31:
I tried using predbat_best_soc_max and found the same, that it stopped the battery charging from Solar.  

I switched to using Predbat because having lots of my own automations was a complete PITA so I'd rather work with everyone to try to figure out the best way to deal with clipping.

Obviously the problem is that with so many different configs, export limits etc that the prediction of clipping is hard and we all want to deal with it slightly differently.  

Having a 'charge to' value might work, but then you're losing out on the potential export you could have done early morning after the low rate period ends and before there's much solar - that window obviously narrows as the sun rises earlier.  Not easy!

johnwb87, 2025-03-31:
> My workaround is similar to that of some others in this thread. I use Home Assistant automations to:
>
> * At sunrise, predict if clipping is likely to occur
> * If so, make room in the battery by [...]

Thank you @CJDumbleton these are really helpful.

I think I am going to flip the logic of the make space automation so that it exports immediately, then use some freeze export as well. This is based on my understanding of the current logic that when there is clipping you first set an export for 4 hours after sunrise and then fill backwards towards sunrise.

johnwb87, 2025-03-31:
Thanks, I also prefer earlier in day as there is usually more overall demand on the grid on the morning, peaking at around 8am, then falling off into the midday solar peak. So where I can I also want to shift export away from the midday low demand. 

Hoping predbat then fills in with demand slots during that time and exports again later. Let's find out!

WyndStryke, 2025-03-31:
> Hoping predbat then fills in with demand slots during that time and exports again later. Let's find out!

You can just set it to demand for that 30m slot via select.predbat_manual_demand (passing in a time as the parameter, in the format "nn:nn:nn")

I do this to make sure that sunlight hours are in 'freeze export' mode like so:

```
  - action: select.select_option
    metadata: {}
    data:
      option: "09:00:00"
    target:
      entity_id: select.predbat_manual_freeze_export
```

johnwb87, 2025-03-31:
@WyndStryke Yes. I may well do that once I’ve had a few instances of the export automation working and can see how it’s working for me. 

In the first instance I am trying a less is more approach as I really want predbat to be making these choices as it’s undoubtedly better informed and more regularly updated than an automation can ever reasonably be.

WyndStryke, 2025-03-31:
I sometimes have a problem with freeze-export oscillating on & off with subsequent plans (not as bad as it was prior to the recent update), so I'm setting slots to freeze-export if it is in demand mode, the SoC is high, and there is PV, and predbat isn't currently active.

cyberkryten, 2025-03-31:
The scripts look useful, if only so we can each figure out 'what works' for our own systems without putting all that complexity into Predbat for testing.

I have an 8kW inverter but the DNO have given me a 3.6kW export limit - I assumed I could just change the value to 3.6 as I don't think this is doing anything on the AC side other than planning exports, but that doesn't show any clipping for tomorrow.  I don't see anything clipped until I drop it to 2.5 which seems odd as I'll definitely have issues with a peak of 5.2kW tomorrow

Is there a reason you're calculating from pv_energy, rather than using pv_power at each time point?   

I've had a play and don't quite understand what the tuning factor is doing either

CJDumbleton, 2025-03-31:
> I have an 8kW inverter but the DNO have given me a 3.6kW export limit - I assumed I could just change the value to 3.6 as I don't think this is doing anything on the AC side other than planning [...]

It looks like `pv_energy`'s attribute `results` has an entry for every five minutes now. I think it was every ten minutes when I set up the automation. So, I need to double the index at which to break out of the loop from 144 to 288 (24 hours of 5 minutes). And `inverter_limit` becomes the inverter rating divided by 12 instead of 6. I've updated the `pv_clipping_in_next_24_hours` [automation](https://github.com/CJDumbleton/ha_predbat_automations). Thanks for pointing out the bug @cyberkryten.

> Is there a reason you're calculating from pv_energy, rather than using pv_power at each time point?

I use `pv_energy` because I'm estimating excess kWh of forecast solar energy over and above the inverter ac capacity over the period.

> I've had a play and don't quite understand what the tuning factor is doing either

The automation tuning factor is designed to account for clouds and how aggressive you want to be with making room in the battery - the lower the value, the more room it will make. By accounting for clouds, I mean, your solar prediction can equal your inverter limit but if that solar turns out to be spikey due to intermittent clouds you can still get clipping.

[57 more replies]