springfall2008/batpred #1207: Multiple Solcast API Key support

campbellcopeland, 2024-06-13:
**Is your feature request related to a problem? Please describe.**
I have two Solcast API keys due to having a roof with three aspects - this is supported in the Solcast Addon but not PredBat

**Describe the solution you'd like**
comma separated list of API keys

**Describe alternatives you've considered**
Keep using the Solcast Addon which supports this

**Additional context**
Having more than one API key isn't supported by Solcast and likely has a limited subset of users - but I am one of them.

springfall2008, 2024-06-13:
This is already supported, just make a list of keys e.g.

api_key:
 - xxx
 - yyy

gcoan, 2024-06-13:
I'll add it to the documentation if its not clear

campbellcopeland, 2024-06-13:
Cheers - documentation update would be good. Thanks.

gcoan, 2024-06-13:
Done.

Updated install doc here: https://github.com/gcoan/batpred/blob/main/docs/install.md#predbat-direct-solcast-method

I moved most of the details of how to setup predbat to call Solcast directly into the apps.yaml documentation so its all documented in one place and not duplicated https://github.com/gcoan/batpred/blob/main/docs/apps-yaml.md#solcast-solar-forecast

closed 2024-06-18 by gcoan