springfall2008/batpred #1208: Energy rates are not pulling through - show as 0

originali, 2024-06-13:
Hello

Issue - energy rates not pulling through correctly despite showing correct values in the developer tools states (attached for your ref)
    event.octopus_energy_electricity_xxxxxxxx_previous_day_rates
    event.octopus_energy_electricity_xxxxxxxx_current_day_rates
    event.octopus_energy_electricity_xxxxxxxx_next_day_rates

    event.octopus_energy_electricity_xxxxxxxx_export_previous_day_rates
    event.octopus_energy_electricity_xxxxxxxx_export_current_day_rates
    event.octopus_energy_electricity_xxxxxxxx_export_next_day_rates

the rates should pull through and predbat determine best time for import/export

appdaemon-predbat version - 1.0.11
predbat shows - 7.22.5
currently set to idle and monitor mode as it kept charging ftom the grid and kept battery full.
also attached config and log

used to use the stand alone predbat but it stopped working so installed the one with appdeamon.

attachements
[entity atributes for rates octopus.txt](https://github.com/user-attachments/files/15827811/entity.atributes.for.rates.octopus.txt)
[predbat.log](https://github.com/user-attachments/files/15827828/predbat.log)
[apps.yaml.txt](https://github.com/user-attachments/files/15827832/apps.yaml.txt)

- 2nd gen 3.6kw inverter + 9.5kwh bat
- Standard HAOS on RPI
- Anything else?

**Screenshots**
[image]
[image]

Thanks for looking into this.

gcoan, 2024-06-13:
Looking at your apps.yaml, all of the energy rate lines are commented out.

As described in the energy rates documentation https://springfall2008.github.io/batpred/energy-rates/#configuring-predbat-to-use-the-octopus-energy-integration  Predbat should automatically use the Octopus integration but it looks like you have commented out the metric_octopus_import and metric_octopus_export regular expression lines

Either the sensor with your meter number in or the regular expression will do, e.g.

```yaml
  metric_octopus_import: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)'
  metric_octopus_export: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate)'
```

originali, 2024-06-15:
hello.

thanks you for coming back to me.
as you realsied I am no programmer neither can't script properly as you can
see.

i changed that as you suggested to:

  # Predbat will automatically find the event. entities from the link below
to the sensors
  metric_octopus_import:
sensor.octopus_energy_electricity_22j0047521_2311706855517_previous_rate

#'re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)'
  metric_octopus_export:
sensor.octopus_energy_electricity_22j0047521_2394300266802_export_current_rate

#'re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate)'

[image: image.png]

the log shows this

2024-06-15 15:05:02.079755 INFO pred_bat: Info: record_status Error:
Exception raised
2024-06-15 15:10:00.132406 INFO pred_bat: --------------- PredBat - update
at 2024-06-15 15:10:00+01:00 with clock skew 0 minutes, minutes now 910
2024-06-15 15:10:00.135641 INFO pred_bat: Saved current settings to
/config/predbat_config.json
2024-06-15 15:10:00.137888 INFO pred_bat: Using cached GITHub data for
https://api.github.com/repos/springfall2008/batpred/releases age 16.5
minutes
2024-06-15 15:10:00.140011 INFO pred_bat: Predbat /config/apps/predbat.py
version v8.0.0 currently running, latest version is v8.0.0 latest beta
v8.0.0
2024-06-15 15:10:00.142223 INFO pred_bat: Autoupdate: Currently up to date
2024-06-15 15:10:00.222745 INFO pred_bat: Configuration: forecast_hours 48
num_cars 0 debug enable is False calculate_plan_every 10
2024-06-15 15:10:00.226860 INFO pred_bat: Predbat mode is set to Monitor
2024-06-15 15:10:01.110610 INFO pred_bat: Found 11521 load_today datapoints
going back 8 days
2024-06-15 15:10:01.876397 INFO pred_bat: Car charging hold True threshold
6.0
2024-06-15 15:10:01.881331 INFO pred_bat: Current data so far today: load
8.5 kWh import 4.4 kWh export 0.1 kWh pv 4.8 kWh
2024-06-15 15:10:01.888547 INFO pred_bat: Fetching futurerate data from None
2024-06-15 15:10:01.891975 INFO pred_bat: Warn: No Octopus data in sensor
sensor.octopus_energy_electricity_22j0047521_2311706855517_previous_rate
attribute 'all_rates' / 'rates' / 'raw_today'
2024-06-15 15:10:01.894987 INFO pred_bat: Error: metric_octopus_import is
not set correctly or no energy rates can be read
2024-06-15 15:10:01.941928 INFO pred_bat: Info: record_status Error:
metric_octopus_import not set correctly or no energy rates can be read
2024-06-15 15:10:01.943844 INFO pred_bat: Error: Exception raised
2024-06-15 15:10:01.947156 INFO pred_bat: Error: Traceback (most recent
call last):
  File "/config/apps/predbat.py", line 11128, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/apps/predbat.py", line 9953, in update_pred
    self.fetch_sensor_data()
  File "/config/apps/predbat.py", line 9103, in fetch_sensor_data
    raise ValueError
ValueError

2024-06-15 15:10:01.993323 INFO pred_bat: Info: record_status Error:
Exception raised

[image: image.png]

not sure what's going on. Could you please advise? thanks in advance.

regards JK

gcoan, 2024-06-15:
I'm assuming that the sensor name sensor.octopus_energy_electricity_22j0047521_2311706855517_previous_rate

Is actually the name of your octopus integration sensor?  Normally you don't need to hard code the meter numbers into the sensor name, the regular expressions should find the sensor names correctly (but these are commented out)

Eg:

metric_octopus_import: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)' metric_octopus_export: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate)' 

But I think I've spotted another problem 

You have 

metric_octopus_import: sensor.octopus_energy_electricity_22j0047521_2311706855517_previous_rate 

It should be 

metric_octopus_import: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)'

CURRENT rate not PREVIOUS

originali, 2024-06-15:
I did revert to original (predetermined) values and enabled them. Predbat
behaved as it should have and picked the values correctly. I enabled it and
gave it access to charge/discharge rather than monitoring.

Sorry for bothering you with this . I am basic user trying to get a maximum
of the solar and battery.

Great piece of software and thank you for developing and support.

I'll keep an eye on it for now. Hopefully it is resolved.

Thanks!

sent from mobile phone

gcoan, 2024-06-15:
Glad you are sorted now.  I'll close this GitHub ticket as the original problem is resolved.  If you do get further issues or questions, please do reach out

closed 2024-06-15 by gcoan