springfall2008/batpred #1210: pedbat with ge_cloud and 3 phase inverter.

LeeFarm11, 2024-06-15:
**Is your feature request related to a problem? Please describe.**

Sorry if this is in wrong place. If it is wrong place, please can you point me to the correct place. 

I have Gen 3, 11kw 3 phase inverter. GivTCP is not yet working (I'm working with britkat1980). I'm trying to install Appdaemon-Predbat combined install, but with ge_cloud. I'm struggling with template file in https://github.com/springfall2008/batpred/tree/main/templates 
Based on instructions I think I need to replace the contents of /addon_configs/46f69597_appdaemon-predbat/apps/apps.yaml with the contents of givenery_cloud.yaml and then modify that version.

I see references to geseriale with e on the end, but is that a mistype? I think it should be without e on the end?

  '# Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  '# if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 'cexxxx'
  geserial2: 'ceyyyy'

  '# Sets the maximum period of zero load before the gap is filled, default 30 minutes
  '# To disable set it to 1440
  load_filter_threshold: 30

  '#
  '# Sensors, more than one can be specified and they will be summed up automatically
  '#
  '# For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  '# If you have three phase and one inverter per phase then you would need three load sensors
  '#
  '# For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  '#
  load_today:
    - sensor.ge_inverter_{geserial}_consumption_today
  import_today:
    - sensor.ge_inverter_{geserial}_grid_import_today
  export_today:
    - sensor.ge_inverter_{geserial}_grid_export_today
  pv_today:
    - sensor.ge_inverter_{geserial}_solar_today

  '# Load forecast can be used to add to the historical load data (heat-pump)
  '# To link to Predheat
  '# Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  '#load_forecast:
  '#  - predheat.heat_energy$external
  '#
  '# If you enable ge_cloud_data then the load/import and export data will be fetches from the GE cloud instead of from GivTCP sensors
  '# this is usually less efficient and of course prone to internet downtime, but could be useful if you lost your GivTCP data
  '# Set the serial to the inverter serial to pull the data from and the key to your API key
  '# When this is set load_today, import_today and export_today are not used
  '#
  ge_cloud_data: False
  ge_cloud_serial: '{geseriale}'
  ge_cloud_key: 'xxxx'

  '# Controls/status - must by 1 per inverter
  '#
  num_inverters: 1
  inverter_type:
   - 'GEC'
   - 'GEC'
  '#

I only have 1 inverter. Do i need to comment all the references to second inverter {geserial2} throughout the yaml?

And then I see reference to separate sensors for each of the 3 phases, but sorry I don't know where to get that information. Is it possible to add instructions for how to deal with 3 phase GE inverter?

Are there other things I need to change due to 3 phase? I was also trying to look at installing dashboard from predbat_dashboard.yaml but I can't find it. I guess this will also be complicated by 3 phases?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

gcoan, 2024-06-15:
Hello and welcome to Predbat

I'll try and help answer your questions...

> Based on instructions I think I need to replace the contents of /addon_configs/46f69597_appdaemon-predbat/apps/apps.yaml with the contents of givenery_cloud.yaml and then modify that version.

Yes that's correct.  My intent is to collapse all the different template files down into ideally just one 'master' template file as having lots of different templates might make it easier for the newcomer to have the right starting point, but it adds an additional 'copy the right template file in' step, and moreover what happens is that new features don't get added consistently to all the different apps.yaml files and this causes different problems.

> I see references to geseriale with e on the end, but is that a mistype? I think it should be without e on the end?

Yes that is a typo, I will correct that, it should be 'geserial' as per apps.yaml:

```
'# Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
'# if it doesn't work or if you have more than one inverter you will need to list both
geserial: 'cexxxx'
geserial2: 'ceyyyy'
```

> I only have 1 inverter. Do i need to comment all the references to second inverter {geserial2} throughout the yaml?

Yes you should do. 

> And then I see reference to separate sensors for each of the 3 phases, but sorry I don't know where to get that information. Is it possible to add instructions for how to deal with 3 phase GE inverter?

In the apps.yaml documentation https://springfall2008.github.io/batpred/apps-yaml/#data-from-home-assistant the 'add entries for each phase' is in the section for configuring Predbat to use PV, load, import and export data from Home Assistant.  It doesn't appear in the following section for GE cloud data so I *think* the assumption is that the GE cloud data when retrieved is for all 3 phases, not just for a single phase.  Indeed there's no configuration options to look at different phase data so if the GE cloud data doesn't cover all 3 phases then we have a different issue.

I will update the documentation to make this clearer.

> Are there other things I need to change due to 3 phase? I was also trying to look at installing dashboard from predbat_dashboard.yaml but I can't find it. I guess this will also be complicated by 3 phases?

I'm not sure that you being on 3 phases should affect the dashboard as predbat manages all phases together and there isn't separate sensors for the different phases.  When GivTCP is working for the 3 phases then yes you probably will want to have different dashboard views for the different phases and use eg the GivTCP battery card to show the phase activity.

But the predbat_dashboard appears to be broken.  In my own install the logfile says that the dashboard is being written to but the last write of it is April 2024.

I'll raise a separate predbat github issue for that to Trefor to look at

LeeFarm11, 2024-06-15:
Thanks very much for those hints and tips. 
The instructions suggest to sort out dashboards (in Predbat Output and Configuration Controls) before starting predbat (Ready to light the touch-paper) ie before removing   template: True
But I cannot find predbat_dashboard.yaml
It is not anywhere in the addons_config area.
Does it get created AFTER template: True is removed?

gcoan, 2024-06-15:
I've been updating the documentation tonight, making it clearer where to find the predbat_dashboard.yaml and how to merge that into your own dashboard

The draft version is here: https://github.com/gcoan/batpred/blob/main/docs/output-data.md

Have a look in the /addon_configs/46f69597_appdaemon-predbat folder but to answer your question, and having looked at the code, it looks like predbat aborts processing as soon as it detects Template is still set to true. In which case none of the Predbat outputs or controls are created including this file.

I'll need to look at changing the install instructions as it looks like they can't be precisely followed as written!

LeeFarm11, 2024-06-18:
I have managed to get dashboard up and running and now also predbat plan. I am not live yet. But now running into this issue with 'possible thread starvation'. I see this has been happening for couple of days despite updates and restarts - Any ideas please?

[image]

gcoan, 2024-06-18:
First thing to try is a restart of AppDaemon and see if that clears the problem.

If not, do you have 'threads' defined in your apps.yaml, if so, what is it set to ? 

https://springfall2008.github.io/batpred/apps-yaml/#threads

It should auto-determine the number of threads but maybe set it to say 2 or N for no threads.

gcoan, 2024-06-19:
> And I also got another error -
> Note: Can not find battery charge curve (settings missing), one of the required settings for soc_kw, battery_power and charge_rate are missing from apps.yaml
> 

The second error you can safely ignore. Predbat tries to calculate an accurate battery charge and discharge curve so it can more accurately model (in the plan) how the battery slows down charging/ discharging as it nears being full/empty.
To do this it uses the givtcp sensors mentioned in the error message to calculate the curve. If you are using the giv cloud link these sensors don't exist. 
It might be possible to use the cloud data to calculate the curves but as the cloud is only at 5 minute data granularity, I doubt it would be that accurate. I will change the code to stop it trying to auto calculate the curve for cloud connection.

On point 1, can you share the full logfile, does it do this repeatedly or was it a one off (which might be a comms issue talking to the inverter)

Cheers

LeeFarm11, 2024-06-19:
Hi. Thanks for your support. Seems like as soon as I turn off Read-Only it starts to have problems, and then eventually it picks up the threads issue. I am able to make changes On and Off to the Inverter directly via these switches (ge cloud). But seems Predbat might be struggling to use them.
[image]

[image]

Here is log file for today.
[predbat(1).log](https://github.com/user-attachments/files/15906438/predbat.1.log)

gcoan, 2024-06-19:
Looking at the logfile, the problem seems to be with the battery reserve.  Can you see what's in that entity?
I'm using givtcp not the ge cloud add-on and the entity is number.givtcp_<inverter serial>_battery_power_reserve so it might be something similar.

Predbat usually uses lower level controls such as battery reserve, battery mode, charge and discharge rate, the switches you highlighted are I believe 'macros' that change the underlying controls so these may not be what predbat uses.
Of course, not familiar with precisely what the GE cloud exposes...

From the logfile the issue is with the battery reserve, it's currently reported as 0.0% (which should never happen) and predbat tries to set it to the default 4% which crashes due to a data type mismatch.  I suspect the issue is with the GE cloud addon not correctly retrieving the battery reserve.

```
2024-06-19 21:08:34.645962 INFO pred_bat: Adjust reserve to default (as set_reserve_enable is true)
2024-06-19 21:08:34.647020 INFO pred_bat: Inverter 0 Current Reserve is 0.0 % and new target is 4.0 %
2024-06-19 21:08:34.647426 INFO pred_bat: Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
2024-06-19 21:08:34.648754 INFO pred_bat: Error: Traceback (most recent call last):
  File "/config/apps/predbat.py", line 11128, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/apps/predbat.py", line 9974, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 8873, in execute_plan
    inverter.adjust_reserve(0)
  File "/config/apps/inverter.py", line 1062, in adjust_reserve
    self.write_and_poll_value("reserve", self.base.get_arg("reserve", indirect=False, index=self.id), reserve)
  File "/config/apps/inverter.py", line 1272, in write_and_poll_value
    matched = abs(float(current_state) - new_value) <= fuzzy
                  ^^^^^^^^^^^^^^^^^^^^
TypeError: float() argument must be a string or a real number, not 'NoneType'

2024-06-19 21:08:34.661573 INFO pred_bat: Info: record_status Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
```

LeeFarm11, 2024-06-20:
These are the only entities with 'reserve'. 
[image]
Is it using 'GivTCP Control Battery Power Reserve' rather than anything from ge cloud?
[image]

gcoan, 2024-06-20:
Looking at the template apps.yaml and the one you posted earlier, I can see one bit that looks wrong and where the reserve gets picked up from:

```yaml
num_inverters: 1
  inverter_type:
   - 'GEC'
   - 'GEC'
```

You only have 1 inverter and have manually set geserial near the start.  I don't think 'GEC' should be duplicated as there's only 1 inverter, can you remove one line please.

```yaml
  reserve:
    - number.ge_inverter_{geserial}_battery_reserve_limit
    - number.ge_inverter_{geserial2}_battery_reserve_limit
```

Did you delete/comment out all the lines with geserial2 in them in apps.yaml? I think you should.

Then can you check for the existence/current value of number.ge_inverter_{geserial}_battery_reserve_limit
    
I might need to try installing the ge_cloud add-on myself to see more how it works.  The apps.yaml isn't (or shouldn't) be pointing to any givtcp components which are all prefixed givtcp_ so even though you have it installed, it shouldn't be an issue.

LeeFarm11, 2024-06-20:
Thanks.
Yes I did already have all those second inverter records commented out.
[image]

I do not have any entities starting with `number.ge_inverter_{geserial}_battery_reserve_limit`

I do have entities starting with 
`number.ge_inverter_{geserial}_ac_charge.......`
`number.ge_inverter_{geserial}_charge.......`
`number.ge_inverter_{geserial}_dc_discharge.......`
`number.ge_inverter_{geserial}_discharge.......`

but not

`number.ge_inverter_{geserial}_battery....`

The only GE cloud entities I have containing the word battery are these -

[image]

gcoan, 2024-06-20:
Thanks, its looking to me like a bug in the ge cloud integration https://github.com/springfall2008/ge_cloud/blob/main/README.md

I've just installed the GE Cloud integration with my Gen 1 hybrid inverters and it's created two sensors:

number.ge_inverter_XXX_battery_reserve_limit

I looked through the code but couldn't find anything obvious as to which sensors are created 
So I can only assume this is a problem with using GE Cloud with the 3 phase inverter behaving slightly differently 🤬

Probably worth looking at the system/log in Home Assistant to see if there is anything of interest, but looking at the code there isn't any debug logging options so its only going to be major fails that appear.

Need @springfall2008 to advise, sorry

closed 2025-01-01 by springfall2008