springfall2008/batpred #1211: Predbat dashboard isn't being written

gcoan, 2024-06-15:
**Describe the bug**
Originally raised on #1210, it caused me to look at my own /config/predbat_dashboard.yaml

The dashboard isn't being written to despite the logfile saying it is

**Expected behavior**
Dashboard written each predbat run (or even after every restart would be just as good)

**Predbat version**
8.0.0 but has been broken since 7.17.0?

**Environment details**
HAOS 2024.5.5
Gen 1 hybrid, 9.5 battery

**Screenshots**
Logfile says dashboard is being written to:
[image]

However filename clearly hasn't been updated for months:
[image]

And the file hasn't been written to for ages:
[image]

[predbat_dashboard.yaml.txt](https://github.com/user-attachments/files/15849137/predbat_dashboard.yaml.txt)

springfall2008, 2024-06-15:
I think you are looking in the wrong place it was moved to the addons_config area?

gcoan, 2024-06-15:
Yes thanks, I will delete my defunct one from /config and update the documentation to make it clearer where to find it.  I looked everywhere but there !

closed 2024-06-15 by gcoan