springfall2008/batpred #1214: SOC simulation is incorrect

jacekowski, 2024-06-16:
**Describe the bug**
Simulated SOC is incorrect when discharging but PV exceeds max inverter AC power.

From screenshot below, from 10:30 onwards there is more than 2.5kWh of PV available in each HH slot but SOC remains constant, while in reality inverter is going to produce at 5kW AC so 2.5kWh in each HH + inefficiencies, and everything above that will be charging the battery. Once battery hits 100%, PV is going to end up clipped at 5kW so there is good few kWh being wasted on days like today.

**Expected behavior**
When discharging, predbat should simulate battery charging if available PV exceeds inverter AC power. Then start discharge early to avoid battery hitting 100% and PV clipping.

**Predbat version**

7.22.5

**Environment details**

- sunsynk 5kW with 14.4kWh of batteries, 10kW of solar panels

**Screenshots**
[image]

springfall2008, 2024-06-16:
Can you check what the logfile shows for Inverter limit and Battery Max rate?

jacekowski, 2024-06-16:
`2024-06-16 15:35:18.801192: Found 1 inverters totals: min reserve 1.44 current reserve 1.44 soc_max 14.4 soc 13.25 charge rate 4.5 kW discharge rate 4.5 kW battery_rate_min 0.0 w ac limit 5.0 export limit 5.0 kW loss charge 3 % loss discharge 1 % inverter loss 5 %

2024-06-16 15:41:25.217869: Inverter 0 with soc_max 14.4 kWh nominal_capacity 14.4 kWh battery rate raw 4500.0 w charge rate 4.5 kW discharge rate 4.5 kW battery_rate_min 0.0 w ac limit 5.0 kW export limit 5.0 kW reserve 10.0 % current_reserve 10.0 %

`

springfall2008, 2024-06-22:
Potential fix on main for testing

springfall2008, 2024-06-23:
Closing, please re-open if not fixed

closed 2024-06-23 by springfall2008