springfall2008/batpred #1216: Prioritise Self Sufficiency option

Yuffi17, 2024-06-16:
Firstly thanks for sharing PredBat with us. - it is most impressive.

I am on the Octopus Agile import tariff, fixed export, with a hybrid Givenergy inverter and PV/battery system.
I would like to prioritise self-sufficiency more, so I import as little from the grid as possible. At the moment Predbat works fairly well but prioritises maximum profit for import/export overall and sometimes plans to import while holding charge (I understand why this behaviour is programmed). I'm happy to export from battery to take advantage of negative Agile pricing or a significant return (as defined by Metric Min Discharge), but I would prefer to use battery capacity over grid when not exporting. Currently I can use the GE Eco mode and manually set charging (which is a pain) or use Predbat and manually override marginal gains (specifically Hold Charge). There has to be a way to control or automate this properly.

I think this could be done with an option to disable "Holding charge" either as a switch or by setting a "Self Sufficiency" mode.  The aim would be for Predbat to ensure I have enough charge for the next day (as predicted currently) by utilising the lowest cost import Agile sessions when the predicted Solar will not be enough to charge the battery to cover this, while retaining the option to tune if/when I export using Metric Min Discharge.

A quick search of the tickets shows an older thread that requested support with something similar: the recommended solution was to use an external script to manually override "Holding Charge" and force Predbat to idle so the inverter eco mode is active (thus allowing the battery to discharge to load).  Depending on how embedded the need for holding charge is, I feel it would be better for Predbat to include a built in function to do this to ensure future compatibility and so all the settings are in one program.

Additional info: my request is based on the idea that I bought my PV system to reduce my monthly outgoings AND to reduce my overall load on the grid. The savings so far already look to be better than any interest I would have got on the system cost, and while interest rates may rise and fall a bit, energy prices are much more likely to go up and by a greater amount,. Therefore I would rather prioritise reduced import (self sufficiency) than maximising return.

springfall2008, 2024-06-16:
Have you thought about giving higher priority to carbon footprint, this maybe better than pure self sufficiency and is already supported?

Quick workaround for self sufficiency maybe to offset your export rate to make it lower than reality, e.g. 5p instead of 15p. 

I'll certainly self sufficiency as an option in future releases.

Yuffi17, 2024-06-16:
Thanks Trefor - much appreciated.

The carbon intensity graph on the Agile dashboard (https://agile.octopushome.net/dashboard) doesn't seem to coincide with the Agile pricing very well - I assume this is the same National Grid info - but I'll play with it in Read Only mode and see how it goes.

springfall2008, 2024-06-16:
https://github.com/springfall2008/batpred/pull/1217

springfall2008, 2024-06-16:
Version on main for testing with **input_number.metric_self_sufficiency** only in expert mode
Try a value of 10p

Yuffi17, 2024-06-17:
Thanks!

Up and running, with no Hold Charge sessions so far. 
If I read this correctly a setting of 10p essentially means Predbat won't plan to import unless the unit price is below 5p (15p-10p) or the battery needs charging to supply my expected use over the next 24 hours (in which case it will use the normal algorithms to pick the best slots for this).

mpartington, 2024-06-18:
I was interested in using this as a limit to stop hold charges or charging above a certain cost.
I initially entered 17p,.but it seems to have prevented charging at a much lower cost

[image]
[image]

Yuffi17, 2024-06-18:
I'm using 10p self sufficiency, and (at the moment) the plan intends to charge overnight when it doesn't need to - possibly  because the battery will drop to 12kWh charge by sunrise and my daily usage is about 13kWh? However there should be enough solar during the day to bring the battery back up to 100% before midday so this probably isn't necessary. 

In winter this would be sensible behaviour and, of course, the plan may change before it gets to the charge point. 
Still interesting though.

[image]

Yuffi17, 2024-06-18:
Yeah - posted too quickly obviously. Maybe I need to stop watching and just let Predbat do it's thing in peace...
(Or it reads these posts and responds accordingly...)

[image]

gcoan, 2024-06-18:
I do find the longer term predbat plan is not always accurate, as it gets nearer to execution time, especially the 4 hours beforehand it refines it and it can change

springfall2008, 2024-06-19:
I've made another fix to main that maybe worth looking at.

That said, keep in mind Predbat tracks your 10% curve as well so if you see a charge do check the what-if scenario for the best10/base10.

mpartington, 2024-06-20:
@springfall2008 , what is the logic for this feature? Is it as simple as the self sufficiency metric is a.cost.at which grid will never be used, or will be used if there is insufficient solar?

I'm still seeing freeze charges above the 10p limit I trialed. Unfortunately there are no rates below 15p to see if it would pick up or ignore a charge slot currently.

The update also seems to have added a lot of (I guess debug) variables above the table. Could these be moved to the bottom maybe or revert back to 8.0 format with just the key ones.

[image]

springfall2008, 2024-06-20:
The metric is just virtually make the import rates higher, so 15.24 is like 25.24 if you used 10p offset.

A freezecharge when the battery is 100% and the PV > load does nothing, so maybe a red herring?

mpartington, 2024-06-20:
If it's just an offset, then it does seem to be working correctly. Have now managed to 'cancel a charge', by setting it to a couple of pence.

Probably not a feature I will use, but great to see the options developed

closed 2024-06-22 by springfall2008