springfall2008/batpred #1218: predbat_dashboard.yaml cannot find file

HARJMI, 2024-06-16:
Sorry if asked before but cannot find predbat_dashboard.yaml.

Installed predbat standalone and also with app deamon. unisntalled and removed the files but cannot seem to locate the file.

Installed the add on version again
Log file attached
[predbat.log](https://github.com/user-attachments/files/15859846/predbat.log)

Is there anything obvious I am missing? Not in /addon_configs/6adb4f0d_predbat or the Root or HomeAssistant folder I can see

gcoan, 2024-06-16:
its in the predbat addon directory that you have been looking in, but its not created whilst template: true is still set in apps.yaml

here's an updated version of the documentation telling you how to find the file and how to use it to create your dashboard

https://github.com/gcoan/batpred/blob/main/docs/output-data.md

I'll need to correct the installation instructions about when you can use this file as it states the dashboard should be setup before you remove the template line and at the moment this is incorrect

HARJMI, 2024-06-17:
I have deleted and un-installed all versions and ran again and I still do not see the dashboard.yaml. I have followed the video and also checked the documentation. I have started the addon with and without the template and the file is not created.

Attached a fresh copy of log file if you can see anything that is wrong 

[predbat (1).log](https://github.com/user-attachments/files/15877793/predbat.1.log)

gcoan, 2024-06-17:
> I have deleted and un-installed all versions and ran again and I still do not see the dashboard.yaml. I have followed the video and also checked the documentation. I have started the addon with and without the template and the file is not created.

The key lines are at the bottom of your logfile:
```
2024-06-17 22:54:21.401768: Error: You still have a template configuration, please edit apps.yaml or restart AppDaemon if you just updated with HACS
2024-06-17 22:54:21.471931: Info: record_status Error: You still have a template configuration, please edit apps.yaml or restart AppDaemon if you just updated with HACS
```

As I said above, the installation instructions are slightly incorrect, the predbat_dashboard isn't created until you have configured apps.yaml for your setup and removed the 'template: true' line from apps.yaml

Follow the [installation instructions](https://springfall2008.github.io/batpred/install/#configuring-predbat), install GivTCP and the Octopus Integration (both which seem from the logfile to not be configured), and edit apps.yaml for your setup

Then when you get to the [ready to light the touch paper](https://springfall2008.github.io/batpred/install/#ready-to-light-the-touch-paper), only when you remove the template: True line from apps.yaml will the predbat_dashboard.yaml be created and you can THEN configure the [predbat output and configuration controls](https://springfall2008.github.io/batpred/install/#predbat-output-and-configuration-controls) - I will be updating the documentation to correct this

gcoan, 2024-06-18:
Looking at the logfile it looks like predbat is crashing because of a failure to load the historical house load data

Can you share a copy of apps.yaml please so I can see if there is a misconfiguration

Thanks

HARJMI, 2024-06-18:
Thank You I  changed the number of days from 7 to 1 and ran again. Both log and apps.yaml attached.

[predbat (3).log](https://github.com/user-attachments/files/15891994/predbat.3.log)

Added .log so I could upload hopefully removing will still allow it to open once reverted back. 

[appsyaml.log](https://github.com/user-attachments/files/15892041/appsyaml.log)

Thank You

gcoan, 2024-06-18:
At the bottom of the logfile you are still getting the same error message:

```
2024-06-18 20:57:15.604663: Warn: can not resolve load_today value sensor.givtcp_{geserial}_load_energy_today_kwh
2024-06-18 20:57:15.652810: Info: record_status Warn: can not resolve load_today value sensor.givtcp_{geserial}_load_energy_today_kwh
```

Which is saying that it can't load historical data from givtcp

But earlier in the log these are the key lines that indicate the direction of the problem:

```
2024-06-18 20:57:09.066759: Warn: Regular expression argument: geserial unable to match re:sensor.givtcp_(.+)_soc_kwh, now will disable
2024-06-18 20:57:09.069852: Warn: Regular expression argument: geserial2 unable to match re:sensor.givtcp2_(.+)_soc_kwh, now will disable
```

There's then messages about successfully finding your solcast integration and octopus energy integration, some warnings about not finding octopus intelligent go sensors which can be safely ignored unless you have an EV and OIG tariff.

So the problem is with finding your GivTCP setup.

In apps.yaml are the lines:

```yaml
# Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'
```

Which should set the variable geserial to your inverter serial number, and geserial2 to your second inverter serial number.  If you only have 1 inverter you can delete/comment out the geserial2 line.

When you setup GivTCP did you change the default prefix for GivTCP entities as shown below in the add-on configuration:
[image]

The default is 'givtcp', mine is changed to 'G', but normally you won't have changed this.  Can you check it please?

If you look in the givtcp add-on log, can you share what you see, is givtcp running and working OK?
If givtcp is running then it should create a sensor called givtcp_<your inverter serial number eg SD2353G832>_soc_kwh which is the current battery state of charge.  Its this that Predbat is failing to find - pointing to either givtcp is misconfigured or its not working properly

HARJMI, 2024-06-20:
Hi,
GivTcp seems to be running ok I am getting the battery and all the other info required on other cards. I cannot see givtcp__soc_kwh within /homeassistant/GivTCP or /addon_configs/6adb4f0d_predbat 

I can still not see the dashboard.yaml in /addon_configs/6adb4f0d_predbat.

I have checked my setup for GivTCP and the default above is GivTCP I also added my serial number. 

I have removed the history and the days and also added my serial number to the apps.yaml file and removed the second inverter.

[predbat (5).log](https://github.com/user-attachments/files/15912770/predbat.5.log)
[apps (3)yaml.log](https://github.com/user-attachments/files/15912777/apps.3.yaml.log)

If you can see anything let me know just cannot see where I am going wrong. I have checked MQTT and BROKER unisstaleld GivTCP and reinstalled.

gcoan, 2024-06-20:
> GivTcp seems to be running ok I am getting the battery and all the other info required on other cards. I cannot see givtcp__soc_kwh within /homeassistant/GivTCP or /addon_configs/6adb4f0d_predbat
> 

Thanks for sharing, the predbat logfile still shows the same error, it's not finding the serial number for your inverter from givtcp.

I think you misunderstood what I was saying, when givtcp is running it creates a series of entities for your inverter inside Home Assistant. These entities contain things like your current battery charge level, what power is coming from your solar panels, and there is a set of controls that predbat uses to control the inverter.

These entities can be seen in Home Assistant by navigating to System / Devices and Services / Entities and then you'll get a full list of the entities in your system. You should be able to filter on givtcp and see all the inverter entities which will be named something like givtcp_XXXX_YYYY where XXXX is your inverter serial number and YYYY is the specific control - e.g. soc_kwh which is the current state of charge for your battery in kWh.  Predbat is failing to find this entity which is why it's failing and not writing the dashboard.

These entities are all within Home Assistant, they don't appear as files in the file system.

Can you look to see if you can find an entity with the name ending 'soc_kwh' and see what that entity is set to in Home Assistant (navigate as above, filter on 'soc_kwh') and then click on it.

Should look like this:
[image]
[image]

Please check you have followed the setup fully for GivTCP, this video from speaktothegeek is very useful https://www.youtube.com/watch?v=ygD9KyciX54

I'm still thinking that GivTCP is either failing or not running or something. You need to install and start GivTCP add-on:
[image]

And installed and started Mosquitto add-on 
[image]

And configured the MQTT integration:
[image]

I can certainly add some more to the installation guide to cover these steps as they're in the video but not in the instructions.  Have a look at the video, I think that will identify the missing step

HARJMI, 2024-06-21:
Spot on Mr...All my GivTCP sensors had _2
soc_kwh_2 etc once I changed all the sensors in apps.yaml it is now working fine. 

To try and rectify I uninstalled GivTCP and also removed all the subfolders and GIVTCP folder unfortuanaltely they all came back as _2 unsure where it gets the data from but manually changing them all has got me up and running.

Thank You so much for your patience @gcoan I was going to give up

gcoan, 2024-06-21:
> Spot on Mr...All my GivTCP sensors had _2
> soc_kwh_2 etc once I changed all the sensors in apps.yaml it is now working fine.
>
> To try and rectify I uninstalled GivTCP and also removed all the [...]

🎉

Yey, got there in the end.  Glad we got it sorted.

To resolve the _2 issue, you probably need to stop GivTCP and MQTT, then navigate to System / Devices and Services / Devices, filter on MQTT, click on each device in turn, then three dots (left hand side) and delete the device.  Maybe start off with the devices that don't have _2 on the end, then when finished, restart Home Assistant and restart GivTCP and MQTT which should re-set everything up again.
But if it's working and you are not fussed about the entity names, then just leave it!

I will add some more to the install documentation as I realise now looking at it afresh that there are some gaps in the steps for getting givtcp working

Cheers, Geoffrey

gcoan, 2024-09-18:
Having needed to delete some stale MQTT sensors myself, I discovered that the above process doesn’t work, you have to delete the topics from MQTT first using MQTT Explorer.

Let me know if this is still an issue

Install instructions for MQTT and GivEnergy now included in #1465

closed 2024-09-18 by gcoan