springfall2008/batpred #1221: Mismatch between install instructions and code - when predbat dashboard is created

gcoan, 2024-06-18:
**Describe the bug**
The [installation instructions](https://springfall2008.github.io/batpred/install/#predbat-output-and-configuration-controls) are (in summary); 
1. install inverter control (eg GivTCP)
2. install predbat/appdaemon add-on
3. Solcast
4. configure apps.yaml
5. [predbat creates control entities in HA](https://springfall2008.github.io/batpred/install/#predbat-output-and-configuration-controls), so create your dashboard (use the predbat_dashboard.yaml)
6. [all ready](https://springfall2008.github.io/batpred/install/#ready-to-light-the-touch-paper)? remove template:true, set predbat mode to Monitor, or Charge/Discharge

Only as shown with feedback on a couple of recent new users #1210 #1218 this sequence doesn't actually work because the predbat_dashboard.yaml is not created until the first time Predbat runs when template: true is removed from apps.yaml

**Expected behavior**
I could change the documentation for a revised sequence of activity but I think it would be better if we changed the Predbat code.

The predbat entities in HA are created by load_user_config called as part of initialize but the Predbat dashboard is created by create_entity_list() which is only called from update_time_loop or run_time_loop both of which are called after the template check at line 11004 in predbat.py

Propose that when detecting a template config the predbat dashboard is still created so users can use it to build a dashboard from before having to turn Predbat 'on':

```yaml
        # Catch template configurations and exit
        if self.get_arg("template", False):
            self.log("Error: You still have a template configuration, please edit apps.yaml or restart AppDaemon if you just updated with HACS")
            self.record_status("Error: You still have a template configuration, please edit apps.yaml or restart AppDaemon if you just updated with HACS")

# create predbat dashboard
            try:
                self.create_entity_list()
            except Exception as e:
                self.log("Error: Exception raised {}".format(e))
                self.log("Error: " + traceback.format_exc())
                self.record_status("Error: Exception raised {}".format(e))
                raise e

            return
```

**Predbat version**
v8.0.0

**Environment details**
n/a

gcoan, 2024-06-19:
@springfall2008 I worked out how to test this now myself and it works fine, so unless you have a real objection to this change I'll include it in my next PR

springfall2008, 2024-06-19:
Seems like a fair change to make

gcoan, 2024-06-19:
Changed and tested, will include in next PR

closed 2024-06-19 by gcoan

gcoan, 2024-08-16:
Included in 8.3.3