springfall2008/batpred #1222: Predbat not recalculating charge limit

Sycamore967, 2024-06-19:
**Describe the bug**
I have 2 batteries and one went offline yesterday.  It's not unusual and has happened before. The home battery predication chart shows the reduced charge limit of 9.5KWh as a give away,  In the past when I get the 2nd battery back on line the chart sorts itself out and goes back to 19KWh.  In the current version of Predbat that hasn't happened, even leaving it over night.

**Expected behavior**
Home battery prediction, shows recalculated charge limit, once 2nd battery comes back on line and total capacity return to normal.  I have had this multiple times in the past and previous versions of PB recalculated the charge limit and showed the chart correctly once the total capacity increased again.

**Predbat version**

v8.0.0

**Environment details**

- GE Hybrid 3.6 and 2x 9.5KWh batteries
- Standard HAOS on Pi
- Battery dropped off line after charge cycle - took a bit longer yesterday to sync the battery voltages and get them both online together.  GE had to put one batteries back in to force charge mode for me. 

**Screenshots**
I've attached screen shots of the chart and SOC from PB and GivTCP

**Log file**
I can't give you this right now - I don't have access to HA to download but could get them after work.
[image]
[image]
[image]

springfall2008, 2024-06-19:
Did you try restarting the addon?

Sycamore967, 2024-06-19:
Hi. Yes I’ve restarted HA twice. It hasn’t helped. I’ve also noticed that the plan isn’t right either now. It’s holding 100% most of the time. If I can send you any logs etc just let me know.

Sycamore967, 2024-06-19:
mmmmm just after posting my last comment - I did a HAOS update and it rebooted again.  PB came back up and is now working fine again, chart is fine as well as the plan.  I can't explain why it didn't do that in the HA restarts yesterday.  I doubt the OS update  had anything to do with it.  But still, issue closed.

closed 2024-06-19 by Sycamore967