springfall2008/batpred #1223: FR - Add capability to pull data from the Hildebrand Glow IHD MQTT integration devices

thewookiewon, 2024-06-19:
**Is your feature request related to a problem? Please describe.**
Using the manual import/export capability in PredBat which is fine but rates change and people can forget to update Predbat

The Glow IHD+CAD is a device that connects to the Smart Meters and has built in capability to link into an existing MQTT broker and send status updates live every 10 seconds, this is very useful if your energy provider does not leverage API capabilities that match Octopus.

**Describe the solution you'd like**
Add capability to pull data from the Hildebrand Glow IHD MQTT integration devices using same automatic find sensors by name as the Octopus integration. 
https://github.com/megakid/ha_hildebrand_glow_ihd_mqtt

**Describe alternatives you've considered**
Using the manual import/export capability in PredBat which is fine but rates change and people can forget to update Predbat

**Additional context**
Currently using Eon-Next Drive V3 which is a 2-rate tariff import and a single track export rate

cyberkryten, 2024-06-25:
The Glow doesn't provide accurate billing data, at least not if you have rates that change during a day - I had a conversation with Jane about it after I'd sent in some debug about another issue as I'd included the Octopus Mini billing data and they asked how I got it.

There is a rate included in the data sent from the Glow but it is not accurate and nor is the standing charge.

Have you checked the debug output to see what is in the messages - if you do I suspect you'll find the same.

thewookiewon, 2024-06-25:
Capability is there though, it's the best option compared to users who aren't on Octopus.

cyberkryten, 2024-06-25:
The capability is not there, at least not if you have any sort of varying rate through the day.

Hildebrand read the data from the Smart Meter and the Smart Meter doesn't have accurate billing data as it can't cope with multiple rates in a day.  That doesn't matter as the actual billing in currency is done by your provider based on the updates of energy used at various times.

Have you checked the debug values of what you see coming from the Glow

Currently my actual rate is 27.5p and the Glow value reported just now is:

`"unitrate":0.07520`

So there's a number there but it is not accurate.

gcoan, 2024-06-25:
> The capability is not there, at least not if you have any sort of varying rate through the day.
>
> Hildebrand read the data from the Smart Meter and the Smart Meter doesn't have accurate billing data [...]

You raise a good point. The master for the billing process is the electricity supplier and I've seen discussion from other people on the GivEnergy forum where they've been highlighting the same, that the charge rate on the meter is inaccurate resulting in the in home display being wrong.

Rather than developing integration to the smart meter where the data can be wrong, would it be better to enable configuration of the import and export rates in apps.yaml to be Home Assistant entities?  Those entities could also be used in the energy dashboard so at least the data would be consistent and it'd be easier to change than having to edit a configuration file (apps.yaml).

e.g.

```yaml
- rates_import:
  start: "00:00:00"
  end: "07:00:00"
  rate: import_rate_overnight
  start: "07:00:00"
  end: "24:00:00"
  rate: import_rate_daytime
```

where import_rate_overnight and import_rate_daytime are entities in HA.

This would also give the option to substitute the above helper entities with entities from the Hildebrand Glow or any other similar integration that connects to the smart meter or your account details.