springfall2008/batpred #1224: Support for Myenergi Libbi hybrid inverter

MalcolmSpencer, 2024-06-19:
I would like to add support for the Mynergi Libbi hybrid inverter and battery.
I have used the SolarEdge config and it seems to be working well but I have I have a few questions about some parameters that I can't find answers to in the documentation. 

**Load today**
This doesn't exist in the inverter but I can create a helper to calculate it. Should I include the EV charger (Zappi) and hot water diverter (Eddi) in the load and then add them in the apps.yaml for EV charging and iSolar or exclude them from the load calculation?

**Charge rate/discharge rate**
This doesn't exist in the inverter (myenergi integration). The inverter will charge the battery and 4600w when on force charge or at the available PV power when charging from solar. The inverter will discharge the battery at 5000w minus the power coming from the PV as its hybrid. Should I just set these to 4600/5000 or create a helper to calculate the value?

**Charge Limit**
Is this the target SOC?

**Inverter Limit**
Is this import or export as I have different values for each (4600/5000)?

**Battery Rate max**
What is this and do I need to set it? the battery can charge and discharge at the full inverter rate 4600/5000.

gcoan, 2024-06-19:
Hello, thanks for your questions, I'll try to answer them, but do recommend looking at the documentation, in particular for apps.yaml https://springfall2008.github.io/batpred/apps-yaml/ which should guide you how to set Predbat for different inverters.

When you get to a working config please do share it so we can store it for other users, and also any updates/gotcha's for the Other Inverters part of the docs https://springfall2008.github.io/batpred/other-inverters/#

> **Load today** This doesn't exist in the inverter but I can create a helper to calculate it. Should I include the EV charger (Zappi) and hot water diverter (Eddi) in the load and then add them in the apps.yaml for EV charging and iSolar or exclude them from the load calculation?

Yes a helper is fine. There is a sample in https://springfall2008.github.io/batpred/apps-yaml/#data-from-home-assistant that you can tailor to your use.
Its probably best to exclude your EV and Eddi as these are usually variable things that don't happen every day and then predbat can predict your base house load.

> **Charge rate/discharge rate** This doesn't exist in the inverter (myenergi integration). The inverter will charge the battery and 4600w when on force charge or at the available PV power when charging from solar. The inverter will discharge the battery at 5000w minus the power coming from the PV as its hybrid. Should I just set these to 4600/5000 or create a helper to calculate the value? I appreciate that in hot or cold conditions it might not charge at these rates but I have no way of getting the actual rate.

Set them to the max values that the inverter will charge/discharge at, Predbat uses these to predict how quickly the battery will charge/discharge when grid charging/exporting
 
> **Charge Limit** Is this the target SOC?

This is the max SOC value recorded when the batteries are full (e.g. 100 for 100%)

> **Inverter Limit** Is this import or export as I have different values for each (4600/5000)?

This is the max AC limit from your inverter.  So 5000.

> **Battery Rate max** What is this and do I need to set it? the battery can charge and discharge at the full inverter rate 4600/5000.

This is the max output rate from the battery which may be lower than the inverter's limit, so hence the ability to set it.  There isn't the ability to set different charge and discharge rates so just set the higher.

MalcolmSpencer, 2024-06-25:
Hi @gcoan, thanks for your help. I have predbat working in read-only mode and the plans it produces look fairly sensible. The apps.yaml is attached. On the plan the thing I don't understand is why in the middle of the day with excess solar the plan is dumping power from the battery and then recharging it as shown on the picture below. This seems very inefficient, I would have thought that the battery would charge to 100%, then the isolar take the power and then it would spill to the grid until the peak time when the excess power is dumped. I can't see the point in charging and discharging the battery. When you first restart predbat it doesn't do this and then after it has run the model a few times it seems to come up with this dump and recharge behaviour.

[image]

[apps.yaml.txt](https://github.com/user-attachments/files/15965428/apps.yaml.txt)

gcoan, 2024-06-25:
Great to see you've made such progress.

The discharging then recharging, yes, I have seen similar behaviour in my own Predbat. Just wondering what version of predbat you are using, might be worth upgrading to the latest as some fixes were introduced in response to #1235.

If you are on a version less than 8.0.0 then you need to upgrade to 7.22.5 first, and then upgrade to v8+

To upgrade, select the version you want to upgrade to from the select.predbat_update control

Let me know if that improves things.

Can we also check what you have set your battery and inverter losses to, these should all be >0 https://springfall2008.github.io/batpred/customisation/#battery-loss-options

input_number.predbat_metric_battery_cycle can be set to 0 but if you set it to a value then predbat will allocate a 'cost' to using the battery which can dissuade it from charging and discharging as much.

But first try the latest version

MalcolmSpencer, 2024-06-25:
I am using the predbat addon V1.1.7 set to auto update and its using predbat 8.1.1 which is the latest version.
I haven't adjusted any of the battery or inverter losses and they are all >0

I think predbat_metric_battery_cycle is only in expert mode and I haven't ventured that far yet

[image]

[image]

MalcolmSpencer, 2024-06-25:
Just a thought but its almost like predbat is trying to keep the soc below the boost min soc to prevent iboost getting the power. Could this be correct? my iboost settings are 
[image]

gcoan, 2024-06-25:
> Just a thought but its almost like predbat is trying to keep the soc below the boost min soc to prevent iboost getting the power. Could this be correct?

I don't have an iboost so have never set it up in my predbat, but I don't know that Predbat is that clever, I thought it just turned iboost on when it predicted there would be excess solar.

You could test this though by turning off switch.predbat_iboost_enable and see how the plan changes?

If it keeps doing this then there must be something more fundamental happening, so can you share the logfile so @springfall2008 can look at it.

In the meantime you can use the controls select.predbat_manual_xxx to manually override Predbat's behaviour for a 30 minute time slot. So for example changing the slots where discharging is planned to Idle.

MalcolmSpencer, 2024-06-25:
@gcoan thanks for the suggestion. I think we have found the problem. When I turn on 'iboost on solar' predbat is trying to keep the soc below the 'iboost_min_soc' so that power is not diverted to iboost. screens shots of the plan with iboost on solar on and off. I appreciate that financially using iboost might not be the best thing to do but environmentally I want to use iboost in the summer rather than using gas. @springfall2008 shall I raise an issue for this?

iboost on solar on
[image]

iboost on solar off
[image]

gcoan, 2024-06-25:
Yes I suggest you do create a new issue for the iboost behavior and we'll keep this one for the libbi documentation which I'll do based on your experience 

Just a thought on the iboost, if you increase artificially the gas price it should favour using iboost over gas for the hot water

MalcolmSpencer, 2024-06-25:
Next observation.
I am on Octopus Flux which has 3 price bands and is therefore very stable.
Yesterday 24th the high and low thresholds were the same all day. Sunday (23rd) and today (25th) they seem to change every time predbat recalculates and I think this is shown in the noise on the calibration chart. Is this normal behaviour? 

attached is the predbat.log
[predbat.log](https://github.com/user-attachments/files/15970085/predbat.log)

[image]
[image]
[image]
[image]

gcoan, 2024-06-25:
> Yesterday 24th the high and low thresholds were the same all day. Sunday (23rd) and today (25th) they seem to change every time predbat recalculates and I think this is shown in the noise on the calibration chart. Is this normal behaviour?

This sounds like a defect.  There was I think a similar defect a while ago of the thresholds changing each time predbat runs, so have a look on github to see if its still open, or please raise a new ticket if it isn't.

Personally I don't really use these thresholds or the calibration chart, I just look at the plan and see if it's 'about right'.  I have expert mode turned on, because it gives more config options, but also because the predbat plan then shows import and export rates with losses which are more important (to me) to understand than the raw rates.  There's a long explanation in the FAQ's about how to understand and optimise the plan.

But the threshold and calibration charts and thresholds are part of predbat so should be accurate. As long as you have reasonable forecast window (24+ hours) they shouldn't wildly oscillate

MalcolmSpencer, 2024-06-25:
@gcoan Again, many thanks for your help

Turning to controls:

The Libbi via the HA Myenergi integration has 3 controls available via services

Mode
Stop - turn off the Inverter
Normal - charge from PV or grid when triggered, discharge to avoid grid use
Discharge - force discharge to grid

Set target battery SOC - %
enable grid charging - on, off

Easy one first to start discharge, I think this exists looking at other config files

  discharge_start_service:
   service: select.select_option
   entity_id: "select.myenergi_libbi_24042899_operating_mode"
   option: "Export"

SoC doesn't apply to discharge in Libbi and it keeps exporting at full rate until the operating mode is set to normal or the battery is empty. Does Predbat monitor the battery soc and then stop the export or would I need to automate that outside predbat?

If predbat stops the export when the target rate is reached then I need a config like the following but I can't see it any of the other configs?

  discharge_stop_service:
   service: select.select_option
   entity_id: "select.myenergi_libbi_24042899_operating_mode"
   option: "Normal"

If I need to automate then I would need the target soc for the discharge.

gcoan, 2024-06-25:
No problem, I see you created a new issue for the iboost and found the other issue that I thought might exist about the stability of the thresholds.

Answering your questions:

> Easy one first to start discharge, I think this exists looking at other config files
> 
> discharge_start_service: service: select.select_option entity_id: "select.myenergi_libbi_24042899_operating_mode" option: "Export"

Yes looks correct except do you mean "Discharge" (per the above list of modes)?

> SoC doesn't apply to discharge in Libbi and it keeps exporting at full rate until the operating mode is set to normal or the battery is empty. Does Predbat monitor the battery soc and then stop the export or would I need to automate that outside predbat?

Every time Predbat runs (every 5 minutes) it re-evaluates the plan based on load, solar, battery SoC, etc. When it decides it needs to be in Idle mode it will send the appropriate commands to the inverter to go into that mode.
So basically Predbat stops the discharge.

> If predbat stops the export when the target rate is reached then I need a config like the following but I can't see it any of the other configs?
> 
> discharge_stop_service: service: select.select_option entity_id: "select.myenergi_libbi_24042899_operating_mode" option: "Normal"

That's correct, its not described in the 'other inverters' doc but there is an example of using it in the Huawei inverter template.
I will correct the documentation for this as well!

MalcolmSpencer, 2024-06-25:
Sorry my mistake the modes are stopped, normal and export.

If predbat is monitoring the soc and turning charge/discharge on and off then that should make things relatively easy.
To start grid charging then mode = normal, enable grid charge and set target soc. I will manually set the target soc to 100% as predbat will monitor that then predbat can enable/disable change from grid to control the charge.

I think the yaml for this is 

  charge_start_service:
    service: select.select_option
    entity_id: "select.myenergi_libbi_24042899_charge_from_grid"
    option: "True"
  charge_stop_service:
    service: select.select_option
    entity_id: "select.myenergi_libbi_24042899_charge_from_grid"
    option: "False"
  discharge_start_service:
    service: select.select_option
    entity_id: "select.myenergi_libbi_24042899_operating_mode"
    option: "Export"
  discharge_stop_service:
    service: select.select_option
    entity_id: "select.myenergi_libbi_24042899_operating_mode"
    option: "Normal"

does that look correct?

gcoan, 2024-06-25:
Seems reasonable, fingers crossed 🤞

MalcolmSpencer, 2024-06-30:
I have been keeping an eye on the calibration chart and generally, the 1 hour looks fairly close to actual considering I am still in read-only mode.
However some days I see a lot of noise on the 8 and 12 hour predictions. There is no way the battery can charge and discharge this fast so any idea what could be causing these oscillations?
I have checked the load data and PV data and they both look clean and realistic 

[image]
[image]
[image]

MalcolmSpencer, 2024-07-01:
I dared to talk Predbat out of read-only mode today but it threw and error.

2024-07-01 15:23:17.568811: Inverter 0 Current Target SOC is 100 already at target
2024-07-01 15:23:17.568984: Inverter 0 Current Reserve is 0.41 % and new target is 4.0 %
2024-07-01 15:23:19.638258: Inverter 0 Wrote 4.0 to reserve, successfully now 4.0
2024-07-01 15:23:19.638481: Inverter 0 current discharge rate is 4800.0 and new target is 4400
2024-07-01 15:23:19.638605: Error: Exception raised 'int' object has no attribute 'split'
2024-07-01 15:23:19.640768: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 10713, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9625, in update_pred
    status, status_extra = self.execute_plan()
  File "/config/predbat.py", line 8104, in execute_plan
    self.reset_inverter()
  File "/config/predbat.py", line 8088, in reset_inverter
    inverter.adjust_discharge_rate(inverter.battery_rate_max_discharge * MINUTE_WATT)
  File "/config/inverter.py", line 1133, in adjust_discharge_rate
    self.write_and_poll_value(
  File "/config/inverter.py", line 1235, in write_and_poll_value
    domain, entity_name = entity_id.split(".")
AttributeError: 'int' object has no attribute 'split'

2024-07-01 15:23:19.681879: Info: record_status Error: Exception raised 'int' object has no attribute 'split'
2024-07-01 15:23:19.682074: Error: 'int' object has no attribute 'split'

Could this be because I have defined the charge and discharge rates as fixed numbers (integers) because the myenergi system doesn't have sensors for these values?

I have also not yet enabled charging but I don't think this is the problem because the error talks about discharge.
Perdbat.log attached.
[predbat(1).log](https://github.com/user-attachments/files/16056274/predbat.1.log)

gcoan, 2024-07-01:
> Could this be because I have defined the charge and discharge rates as fixed numbers (integers) because the myenergi system doesn't have sensors for these values?

You could define a helper entity and set it to the discharge rate, then use the helper name in apps.yaml to prove/disprove whether this is the cause

MalcolmSpencer, 2024-07-01:
i found that someone else ran into this problem but I'm not clear what the solution is https://github.com/springfall2008/batpred/issues/940#issuecomment-2068175327

gcoan, 2024-07-01:
Malcolm,

In the solution @nickgee31 found for his FoxESS, he did as I suggested, create two helper entities of type number and set these in apps.yaml:

```yaml
  charge_rate:
    - input_number.inverter_charge_rate
  discharge_rate:
    - input_number.inverter_discharge_rate
```

The alternative approach that @springfall2008 mentions is if you are creating an unsupported inverter type as described in the documentation https://springfall2008.github.io/batpred/other-inverters/#i-want-to-add-an-unsupported-inverter-to-predbat

You would copy the inverter type yaml into apps.yaml and configure it to your specific requirements of your inverter.
One of the options is to set the field 'output_charge_control' to 'current' 

and then you can control the charge/discharge using the

timed_charge_current/timed_discharge_current entities configured in apps.yaml

(this isn't very well documented, I admit, need to improve on this....)

What you have done at the moment is to copy and adapt the Solar Edge inverter definition (type: 'SE') which Predbat has a basic knowledge of how this inverter works, so when you give it something it doesn't expect, that's why it falls over.

In the Solar Edge template yaml file I can see it expects to find a sensor name configured for the charge and discharge rate, so when you put a fixed value in, that's how we get the problem....

This may be your only problem with reusing the SolarEdge config, and so doing what I suggested above could resolve your issue.  Or this might just be the start of a long series of problems where your Libbi doesn't behave in the same way as the Solar Edge - in which case you'll need to create a custom inverter config from scratch as above.
Ideally Predbat would be a bit more forgiving of numbers vs entities holding numbers, also that this would be documented a bit more fully.  Its on my to-do list to improve the documentation, so partly why I am assisting in debugging these issues to find what the doc gaps are

Cheers

MalcolmSpencer, 2024-07-03:
Hi Geoffrey

I took your advice and switched to using the non-standard template with control set to charge
It initially threw the same error but I eventually spotted the problem.
In the template in the documentation you have a couple of random " and once these are removed it seems to work OK so you might want to update the docs.

    has_ge_inverter_mode": False
    support_discharge_freeze": False

There also seems to be some other parameters in the code which aren't in the doc template i.e. "can_span_midnight"

I'll monitor it for a couple of days and send you the apps.yaml file.

Thanks again,

Malcolm.

gcoan, 2024-07-03:
That sounds brilliant progress, yes starting with your own custom inverter definition is probably going to be best. 

And I agree there's more to do on the documentation. Please do list all the gaps and errors you find and I'll try to resolve

MalcolmSpencer, 2024-07-04:
Looking at the code at https://github.com/springfall2008/batpred/blob/acf0f20e5666fcb3622a7de031c76aea3ffba44c/apps/predbat/inverter.py#L158 the template seems to be missing

current_dp - no idea what this is but default of 1 seems to work
has_timed_pause
has_idle_time
can_span_midnight

I'm not really sure what can span midnight does though is could guess is allow charge or discharge block to cross midnight. Is that correct?

The template also contains 'has_time_window' which I can't see in the code

MalcolmSpencer, 2024-12-01:
@MaximumFish Yes i have this working, i'll send over the YAML later once I have tidied it up. I am still on flux because the Agile rates so far this year haven't made it worth switching but I continue to monitor them.

MaximumFish, 2024-12-01:
That would be awesome, thank you. I've been struggling to make the numbers work out in Agile's favour on the comparison websites, but surely if the import and export slots can be super optimised through automation then it'll be cheaper, right?

MalcolmSpencer, 2024-12-15:
@gcoan Attached is the latest Apps.yaml to support the Myenergi Libbi inverter and battery. I have been using it for a while now and it seems to work OK. I have added as many comments as I can to assist others. The key things are:-

ragg987, 2025-09-24:
@gcoan I wonder if you can assist with this please. Sorry if these are basic questions; I have been reading the docs but there is a lot to absorb and it is only a few days since I started with predbat and Libbi.

I want to let predbat control the SOC of the Libbi, which the above template does not currently allow me to do. I've scanned the docs and I think the following would work.

Working config is:
```
  charge_start_service:
    - service: switch.turn_on
      entity_id: "switch.myenergi_libbi_charge_from_grid"
```
To which I could append
```
    - service: myenergi.myenergi_libbi_charge_target
      device_id: "5e500be27af46da065f0573ac2ce2599"
      soc: "{target_soc}"
```

I believe the above would work in that at the point of setting charge from grid predbat would also set the target SOC, which Libbi supports.

This leads me to the next questions.

Libbi reports current SOC as % but requires a Wh number to set the target. At present the inverter config has:
`soc_units: "%"`

Hence I need to convert the predbat target_soc to Wh. I would apply a formula with reference to another Libbi sensor which holds battery capacity. Something along the lines of:
`target_soc = target_soc * states('sensor.battery_capacity') | float * 1000`

How do I go about providing this calculation, can I add it to apps.yaml? And if so what is the syntax that apps.yaml is expecting? Jinja?

I had thought about creating the Wh target as a template sensor in HA, but I am unable to find where target_soc from predbat is - I cannot find something suitable in the predbat list of sensors exposed to HA.

Any pointers would be great - and thanks for your support so far.

ragg987, 2025-09-25:
One step forward.

So based on this old issue https://github.com/springfall2008/batpred/issues/2216#issuecomment-2799012158 I created an automation to set the target SOC on the Libbi.
```
alias: libbi_set_charge_target_soc_predbat
description: >-
  when predbat calls for force charge of libbi this will set the charge target
  based on predbat requirement
triggers:
  - trigger: state
    entity_id:
      - binary_sensor.predbat_charging
    from: "off"
    to: "on"
conditions: []
actions:
  - action: myenergi.myenergi_libbi_charge_target
    target:
      device_id: 5e500be27af46da065f0573ac2ce2599
    data:
      chargetarget: "{{ states('sensor.predbat_lib_0_charge_limit') | int * 92 }}"
mode: single

```
I also set the inverter parameter from false to true so that predbat does not call the libbi service to switch off force charge.
`has_target_soc: true`

Will test tonight and hopefully getting there...

chalfontchubby, 2026-08-28:
Libbi looks from the code to be supported - can this ticket be closed?

closed 2026-09-07 by chalfontchubby