springfall2008/batpred #1228: Environmentally Friendly Switch?

Grey-Sheep, 2024-06-19:
I am using Predbat with Octopus Agile Import and Octopus Fixed rate export.

When the cost of energy is above the value of export there is very little (or no) export from the battery to the grid (as expected).

One of my reasons for purchasing solar and battery was to help reduce the carbon footprint, and we know that this is highest between the hours of 1600 and 2000 most days.

When the cost of energy is above the value of export, my battery is  only providing electricity overnight and at the moment has lots of spare energy for the next day.

I would be happy to export an amount of energy each day at peak time, knowing that either it will be made up again the next day under solar, or that it could be replaced at some point with little cost difference.  It won't be for profit, so I don't see how the current system could be configured to do this?

I could use manual discharge, but I keep forgetting!  I don't think the "reduce carbon" option works this way?

Would an "Environmentally Friendly Switch" be possible that would export any calculated surplus energy during the peak period and replaced it in the most carbon neutral way be possible?

gcoan, 2024-06-19:
The easiest way to achieve this is with a rates_export_override in apps.yaml with a higher export rate than you actually will achieve, this will encourage Predbat to export in that period, and as long as there's still enough battery charge for overnight then it won't recharge from grid

See https://springfall2008.github.io/batpred/energy-rates/#manually-over-riding-energy-rates

springfall2008, 2024-06-20:
Have you look at the Carbon optimisation feature?

springfall2008, 2024-06-20:
input_number.predbat_carbon_metric (carbon enable) When Carbon footprint tracking is enabled (switch.predbat_carbon_enable) you can specify a cost per Kg of CO2 used to weight the selection of plans. Values of around 10-200 will give varying outcomes to trade off cost vs carbon footprint of your system.

Note: Carbon footprint tracking can only be enabled if apps.yaml is configured to point to the correct CO2 cost sensor

springfall2008, 2024-08-07:
Closing this for now as I think the metric carbon does the same thing already.

closed 2024-08-07 by springfall2008