springfall2008/batpred #1229: Predbat assumes the wrong config dir, and logs misleading messages as a result

gcoan, 2024-06-19:
**Describe the bug**
I'm running the original Predbat within AppDaemon install.

predbat save files, dashboard, config.json and appdaemon.yaml are thus in /addon_configs/a0d7b954_appdaemon

However the logile states that these are written to /config which is confusing for users:
```
2024-06-19 13:20:39.238349 INFO pred_bat: Saved current settings to /config/predbat_config.json
2024-06-19 13:20:39.445881 INFO pred_bat: Creating predbat dashboard at /config/predbat_dashboard.yaml
```

**Expected behavior**
Pathnames in the logfile match the actual path being written to

**Predbat version**
v8.0.0

**Environment details**
HAOS

**Diagnosis**
Predbat uses self.config_root throughout the code as the root directory that's prepended to filenames

self.config_root is determined by finding the first path _that exists_ in CONFIG_ROOTS (defined in config.py) which has the first entry as /config,  thus as long as this directory exists, config_root is set to /config

However AppDaemon and I assume the Predbat add-on map /config to the appropriate directory for the addon in /addon_configs, thus the writes are made eg to /config/predbat_dashboard.yaml but the file is actually written to addon_configs/a0d7b954_appdaemon/predbat_dashboard.yaml

I don't know how to work out what the actual directory that the add-on is running in otherwise I would change the code myself to correctly set self.config_root

springfall2008, 2024-06-20:
This is because /config is a link to the addon_config area inside the container. I don't know how Predbat could even know the true location?

gcoan, 2024-06-25:
@springfall2008 I've been looking through the HA documentation to try to see if there was a way of working out what the name/config directory is of the add-on that's running..

From https://developers.home-assistant.io/docs/api/supervisor/endpoints#addons

I think if you GET /addons/**self**/options/config you should be able to get the add-on's configuration. If we can find the slug name then we can construct the correct pathname for the logfile.
But this is getting to the limit of my comfort/desire to want to program in Python....

So what I have done at the moment is to check what self.config_root is, and if its /config (which is symbolic linked to /addon_config/<slug>), to change the **logged** message to be:

2024-06-25 20:18:28.811429 INFO pred_bat: Saved current settings to /addon_configs/<addon name>/predbat_config.json

I don't really like this, but unless we can work out the slug name I can't think of anything better and its an improvement (slight) on referring to the /config pathname which we know is wrong

Thoughts?

iainfogg, 2024-06-30:
Glad I found this ticket, just spent 30 mins trying to work out where my auto-generated dashboard YAML was, after repeatedly looking all over /config :)

gcoan, 2024-06-30:
> Glad I found this ticket, just spent 30 mins trying to work out where my auto-generated dashboard YAML was, after repeatedly looking all over `/config` :)

That's exactly what led me down this rabbit hole as well.

I think I will raise the 'how do I find what /config is sym linked to' question on the HA community forum as I really don't like the bodge outlined above, but it may be the best we can get to :-(

gcoan, 2024-06-30:
Created a [Home Assistant Community post](https://community.home-assistant.io/t/how-to-determine-actual-config-path-in-python/744800) to try to get advice on solving this problem

gcoan, 2024-07-05:
Worked out how to determine where /config is mounted to by getting the add-on slug name from Home Assistant and using that to construct the full hierarchical pathname that's now used in all the log messages

gcoan, 2024-07-05:
@iainfogg I have fixed this now so the logfile now determines what the add-on name it is running under is, and writes the proper hierarchical pathname in the logfile for the dashboard, config file, save files, etc
I have tested this on my HA under AppDaemon and Predbat add-on's and it all appears to work OK.
Would you be able to test it as well?  You just need to download all the .py files from my fork https://github.com/gcoan/batpred/tree/main/apps/predbat
Note this is predbat v8.0.0 so make sure you don't have auto-update turned on

iainfogg, 2024-07-06:
@gcoan just tried it, and it worked fine - thanks for sorting it!

I was running under the original AppDaemon setup.

SwiftRR, 2024-07-07:
2024-07-07 10:50
Predbat.log and predbat add-on log following 'downgrade' using py files from Geoffrey's branch back to v8.0.0.

[predbat-2.log](https://github.com/user-attachments/files/16118734/predbat-2.log)

[predbat add-on.txt](https://github.com/user-attachments/files/16118735/predbat.add-on.txt)

Rob

gcoan, 2024-08-16:
Fixed in release 8.3.3

closed 2024-08-16 by gcoan