springfall2008/batpred #1230: Don't discharge battery to car when using Zappi charger on IOG

sammort, 2024-06-20:
**Is your feature request related to a problem? Please describe.**
My setup is:
- GivEnergy All-In-One
- MyEnergi Zappi (device used for Intelligent Octopus Go)

I think this is a bit of an edge situation but I wanted to raise a feature request. When using the Zappi charger integration for IOG, it is required to be in 'Eco+' mode at all times. As a result, when the car is plugged in, any exported energy will be diverted to charge the car.

In the situation where the car is plugged in to the charger overnight, an IOG charging plan is created. If there is "excess" charge in the battery, predbat will plan to discharge the battery prior to the overnight cheap IOG rate. Predbat may even plan to charge and discharge the battery during the cheap overnight period as it is cost positive (7.5p import vs 15p export).

Because of the Eco+ mode, these exports will be diverted to the car.

**Describe the solution you'd like**
I'd like that if `switch.predbat_car_charging_from_battery==false`, `switch.predbat_octopus_intelligent_charging==true` and the IOG integration is detected as being the Zappi (this can be detected in the Octopus Energy HA integration), then predbat would not discharge whilst the car is plugged in (`car_charging_planned` is in a state that matches `car_charging_planned_response`). 

**Describe alternatives you've considered**
Rather than building the logic in to predbat based on the states that already exist (as suggested above), a new switch could be created, something like `switch.predbat_discharge_from_battery_when_car_plugged_in`, which would toggle whether discharging from battery is allowed when the car is detected to be planned charging.

**Additional context**
Attached is a predbat plan, currently my car is plugged in and planned to charge from 1am-5am.

[image]

The planned discharging in the slots from 11pm-1am would actually be discharged in to the car rather than exported to the grid because of the Zappi Eco+ mode

RobinCu, 2024-06-20:
I have an identical set up.  You can increase the export margin on the Zappi (in the settings on the actual device) to something higher than your inverters max output.

https://support.myenergi.com/hc/en-gb/articles/18468235308433-How-to-stop-menergi-products-diverting-exported-power

sammort, 2024-06-20:
> I have an identical set up. You can increase the export margin on the Zappi (in the settings on the actual device) to something higher than your inverters max output.
> 
> https://support.myenergi.com/hc/en-gb/articles/18468235308433-How-to-stop-menergi-products-diverting-exported-power

Ahh nice, thanks for the link, looks like that'll fix this scenario for me. Will close this request.

closed 2024-06-20 by sammort