springfall2008/batpred #1232: Install button not working

onetoomany, 2024-06-21:
Installed version v8.0.0
Latest version v8.1.0

From Home Assistant Settings, I am told that there is an updated version of Predbat available (along with a yellow 1). However, if I click on Install, nothing happens. I can, however, update from within HACS.

I can't see anything in the logs that relate to this.

Also, whenever I update, I have to restart AppDaemon - is there a way I can avoid this?

gcoan, 2024-06-21:
> Installed version v8.0.0
>
> Latest version v8.1.0
>
>
>
> From Home Assistant Settings, I am told that there is an updated version of Predbat available (along with a yellow 1). However, if I click on [...]

You could try using the select.predbat_update control to select the version to upgrade to, or turning on auto-update switch in Predbat.  Either of these will mean Predbat updates itself and doesn't need to restart AppDaemon.

If you update via HACS you will always get the 'template error' and have to restart AppDaemon, it's just the way it works, sorry.

matttheotter, 2024-06-22:
So it looks like it does install but there’s some error even after a restart causing it to not update status.

[image]

gcoan, 2024-06-22:
> So it looks like it does install but there’s some error even after a restart causing it to not update status.
> 
> 

Can you share what's in your logfile please.

What version were you upgrading from?  If it was an old version did you upgrade to 7.22.5 first?

matttheotter, 2024-06-22:
Upgrading from 8.0.0 to 8.1.0, manually switching back to 8.0.0 got rid of the error.

Will grab the logs.

springfall2008, 2024-06-22:
Potential fix on main for testing.

matttheotter, 2024-06-23:
Switched to main from 8.0.0 and got the same error, logfile looked the same too.

springfall2008, 2024-06-23:
Can you share the logfile?

PianSom, 2024-06-23:
Went 8.0 -> 8.1.1, got error. Restarted appdaemon (docker container), Error persisted. Went 8.1.1 -> main, error cleared.

matttheotter, 2024-06-23:
> Went 8.0 -> 8.1.1, got error. Restarted appdaemon (docker container), Error persisted. Went 8.1.1 -> main, error cleared.

Apologies, have been out.

That actually worked, I went 8.0.0 -> 8.1.0, error, restarted appdaemon, after the restart completed the error went, went up to 8.1.1 and all is good now.

closed 2025-01-01 by springfall2008