springfall2008/batpred #1233: Solaredge inverter & Tesla powerwall

sdrake001, 2024-06-21:
Hello
I have a solaredge inverter and Tesla powerwall for battery. Can/does predbat work with this combination.
I have installed solaredge modbus and am getting data but none of the sensors that are mentioned in prebat. I’m assuming that this is because solaredge can’t see the state of my battery?
Thanks for any help/support you can give.
Thanks

gcoan, 2024-10-09:
> Bit of a late response, but I have recently set up the HA predbat addon with my Tesla PW2/Gateway and SolarEdge PV system, now I have moved to an electricity tariff with a cheap overnight rate.

If you are able to share instructions of what integrations are required, your apps.yaml and any helpers/template entities you needed, I can add them to the documentation for other Tesla/SolarEdge users

Thanks

sdrake001, 2024-10-17:
Its interesting to hear that you have managed to get yours running - I have really struggled with the Solar edge and Tesla combination, I am not that experienced on Home Assistant so don't know half the time if its user error, probably is!
Would you be able to share your config file so that I can check to see if I have done mine properly?
Thanks

It is early days but seems to be working fine for me in Control Charge mode.

I use both of the Tesla integrations (Tesla Custom Integration & Tesla Powerwall) and the SolarEdge Modbus Multi integration.

I use the powerwall sensors for load, import, export and pv in the config. I don't use the SolarEdge sensors for predbat.
I use the Tesla backup_reserve setting to control charge and discharge from the grid; using a service call to set the reserve.

The only minor niggle that I haven't worked through yet is by using the backup_reserve grid charging is limited to 1.8kW, but PV charging can and does go higher.

I set the inverter_limit_charge to 1800 otherwise predbat thinks it can charge the PW2 quicker in my offpeak grid slot. This has the effect of the predbat simulation/prediction thinking PV charging is also limited to 1.8kW so predicts some export and a later time for the PW2 to be full.

Perhaps defining a second inverter may help but I haven't tried yet.

closed 2026-08-28 by chalfontchubby