springfall2008/batpred #1234: Error Since latest Update - ModuleNotFoundError: No module named 'download'

jswilkinson851, 2024-06-21:
I updated PredBat (among other components) for the first time in about two months. Previously my PredBat plan was working as were all of the sensors, etc. on the PredBat cards. However now none of it is working. Looking into the logs a bit more, it says the following:

024-06-21 12:17:26.074619 WARNING Error: ------------------------------------------------------------
2024-06-21 12:17:26.074978 WARNING Error: Unexpected error loading module: /config/apps/predbat.py:
2024-06-21 12:17:26.075203 WARNING Error: ------------------------------------------------------------
2024-06-21 12:17:26.086139 WARNING Error: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1000, in check_app_updates
    await utils.run_in_executor(self, self.read_app, mod["name"], mod["reload"])
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 783, in read_app
    self.modules[module_name] = importlib.import_module(module_name)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__[init__.py](https://l.facebook.com/l.php?u=http%3A%2F%2Finit__.py%2F%3Ffbclid%3DIwZXh0bgNhZW0CMTAAAR1n3lJx_u7LilEJezlvSU9_k1Oojbp4oR_krH-S2yCV3_3D75-qz1autBg_aem_ZmFrZWR1bW15MTZieXRlcw&h=AT2XMO6a-r5IXid8Z9rKM_0-YfHHUAhc6yb_9wr8y1tZ72T0mSgDwi6aCSdAscjfJj92NoFLCRW0lFLMBpnazLhh0JK13hUovMkRpOTIwcxKr_gXcC7lYCNefS5gXUNQ3g&__tn__=-UK-R&c[0]=AT2mnIm6FMsCG6dCJ3USZxdekO7WhxBb6nQtwVSDL8wuSlLWhze76A26QosZR0Le16JRFaYapu1c9qiSB8r1ymRXfHWmAJ6HS2HMbcv0z2KOFEC87BLIe-274QOjApLQKBnVa5Ez2ei9UR4BQedZhfQKKyZ8qp5S3krQZMsw14i9nNRqppQGD0EtrEREIrQ6_sv5cnfA8rYObc50Be7s9MxDQsalK3y7oySAM6At9g)", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/apps/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'

[image]

gcoan, 2024-06-21:
Did you upgrade to 7.22.5 first or jump straight to 8.0.0 /8.1.0

I think that might well be your problem.  7.22.5 splits predbat into separate modules and you have to do that before moving to v8

jswilkinson851, 2024-06-21:
It is likely that I did. How can I reverse the update? I have PredBat installed via the appdaemon-PredBat method.

mpartington, 2024-06-21:
Try installing the update via the HACs update method, that should fix it

jswilkinson851, 2024-06-21:
> Try installing the update via the HACs update method, that should fix it

Is that the way to do this if I have used the appdaemon-Predbat installation method? I only ask as I don't think I had installed predbat through HACS.

gcoan, 2024-06-21:
> Try installing the update via the HACs update method, that should fix it

Unfortunately this won't work as @jswilkinson851 is using the combined appdaemon-predbat install

Hopefully enough of predbat is still working that you can use the manual select.predbat_update to select the prior version, predbat would update itself and you might need to restart appdaemon-predbat to get it going again.

If that doesn't work then you can uninstall appdaemon-predbat add-on, re-install it, and once running, manually update to 7.22.5 using the select.predbat_update control

🤞

jswilkinson851, 2024-06-21:
> Unfortunately this won't work as @jswilkinson851 is using the combined appdaemon-predbat install
>
> Hopefully enough of predbat is still working that you can use the manual select.predbat_update to [...]

How do I go about trying your first option, to manually run select.predbat_update? Sorry to have to ask.

mpartington, 2024-06-21:
Ah yes, I haven't ever used the combined version.
If you have the entity:

select.predbat_update

In your dashboard, then you may be able to select a previous version
[image]

jswilkinson851, 2024-06-21:
> Ah yes, I haven't ever used the combined version. If you have the entity:
> 
> select.predbat_update
> 
> In your dashboard, then you may be able to select a previous version ![Screenshot_2024-06-21-14-10-49-636_io homeassistant companion android](https://private-user-images.githubusercontent.com/106986397/341773219-9b09c3ab-3533-4307-b629-c00d173e28a9.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg5NzU4NzgsIm5iZiI6MTcxODk3NTU3OCwicGF0aCI6Ii8xMDY5ODYzOTcvMzQxNzczMjE5LTliMDljM2FiLTM1MzMtNDMwNy1iNjI5LWMwMGQxNzNlMjhhOS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyMVQxMzEyNThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMjlhMDU1YzQxMmM2MDc5NGViZWQ1OWQ5MWE2NjM5YmIzYzYyZmIwOWJkNDMxMzEwNTU0YWQ0Y2Q3OTU2YjI0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.uKpYHi7UjppLdiJO3YM_dZcn5uD3ljSJwB1KwwYHwNU)

Unfortunately it looks like I may have to do a full re-install then as previously suggested, as I cannot see any of the entities on my PredBat dashboard (see screenshot above). I'll have a go at reinstalling this evening and get back to you all.

Thank you all for your help so far!

gcoan, 2024-06-21:
You shouldn't need to do a full reinstall if the entity isn't on your dashboard

You could just edit the dashboard and add it 😉

Or navigate to Settings / Devices and Services / Entities

Filter on update:

[image]

Click on the select.predbat_update:

[image]

And choose the version you want from the dropdown

jswilkinson851, 2024-06-21:
I'm not exactly sure what I did, but it is all working (all of your advice really did help, after doing a few reinstalls xD).

Thank you so much!

closed 2024-06-21 by jswilkinson851

stokebob, 2024-07-19:
I'm having the same issue as above. Running through the tread suggestions, I cannot get the select.predbat_update entity to show.

I've tried uninstalling/installing/restarting via the appdaemon-predbat info screen, but I'm stuck.

What can I try next? I have PredBat installed via the appdaemon-PredBat method.

Thanks

jswilkinson851, 2024-07-19:
> I'm having the same issue as above. Running through the tread suggestions, I cannot get the select.predbat_update entity to show.
>
> I've tried uninstalling/installing/restarting via the [...]

I can't remember what I had to do to get everything working again (I presume I had to rewind to an older version and then steadily reinstall updates again. What I would suggest is that there is a new method of installing PredBat independently of appdaemon now, which may be worth exploring. Check out the Facebook page for more information.

gcoan, 2024-07-19:
> I'm having the same issue as above. Running through the tread suggestions, I cannot get the select.predbat_update entity to show.
> 
> I've tried uninstalling/installing/restarting via the appdaemon-predbat info screen, but I'm stuck.

I'm not really sure from what you've written where you are up to and where its broken with what error message.

Can you share the logfile please.

Do you have the predbat.status and/or select.predbat_update entities in your entities list?

If you have completely uninstalled appdaemon-predbat and reinstalled it then you could well be back at the start of the installation process with a template apps.yaml file that you need to reconfigure (or restore from a prior backup), but Predbat will be showing this in predbat.status

stokebob, 2024-07-21:
I hadn't updated pedbat in some time, not sure what version I was updating from, but it would've been 2-3 months old. The update failed, and now I'm getting the same error message as the op. See below.

```
-----------------------------------------------------------
 Add-on: appdaemon-predbat
 Predbat pre-install in AppDaemon
-----------------------------------------------------------
 Add-on version: 1.0.11
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (amd64 / qemux86-64)
 Home Assistant Core: 2024.7.2
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[08:19:05] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-07-21 08:19:17.434273 WARNING Error: ------------------------------------------------------------
2024-07-21 08:19:17.435732 WARNING Error: Unexpected error loading module: /config/apps/predbat.py:
2024-07-21 08:19:17.436336 WARNING Error: ------------------------------------------------------------
2024-07-21 08:19:17.451930 WARNING Error: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1000, in check_app_updates
    await utils.run_in_executor(self, self.read_app, mod["name"], mod["reload"])
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 783, in read_app
    self.modules[module_name] = importlib.import_module(module_name)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/apps/predbat.py", line 37, in <module>
    from download import predbat_update_move, predbat_update_download, check_install
ModuleNotFoundError: No module named 'download'

2024-07-21 08:19:17.452613 WARNING Error: ------------------------------------------------------------
```

Entities wise, the only entity I am getting is ```update.appdaemon_predbat_update``` all others are missing.

I've tried uninstall, install, rebuild. I end up in the same place, stuck. Not sure what to do next.

gcoan, 2024-07-21:
> Entities wise, the only entity I am getting is update.appdaemon_predbat_update all others are missing.
> 
> I've tried uninstall, install, rebuild. I end up in the same place, stuck. Not sure what to do next

Same as the other person, if you are on an old version you need to upgrade to 7.22.5 before going to any version beyond version 8.  It looks like you are missing some of the bits of Predbat.

From the select.predbat_update can you select version 7.22.5,  then restart appdaemon-predbat,  this should give you a working version and you can then upgrade to a version beyond this.

if this doesn't work then you might be better changing to using the predbat add-on installation method (from appdaemon-predbat), upgrade instructions are in this install doc (which is due to be merged with the predbat docs):

https://github.com/gcoan/batpred/blob/main/docs/install.md

stokebob, 2024-07-21:
My ```select.predbat_update``` entity was not there, so rolling back the update was not possible.

In the end I went with your suggestion, removed appdeamon-predbat and installed the add-on predbat. Some tinkering later and I'm back up and running.

Thanks for all your help

gcoan, 2024-07-21:
> My ```select.predbat_update``` entity was not there, so rolling back the update was not possible.
> 
> 
> 
> In the end I went with your suggestion, removed appdeamon-predbat and installed the add-on predbat. Some tinkering later and I'm back up and running.
> 
> 
> 
> Thanks for all your help

Glad you got sorted