springfall2008/batpred #1237: Manual Force discharge leaves charge rate at current value ( possibly 0 )

reidjr2, 2024-06-22:
When using manual forced discharge the Charge rate remains at whatever value was set when the manual Discharge was triggered. 

**Expected behavior**
[Edge case]
I had expected the charge rate to reset to the default/max charge rate. The charge rate is important when Discharge is being used to avoid PV clipping. If the charge rate is  0 ( or any value <default), it will remain reduced and there is no clipping benifit. If predbat has previously set the charge rate to max through any other action this behaviour is masked, so the outcome is dependent on the starting condition.

**Predbat version**

8.1.0

To reproduce: Manually reduce charge rate to < max. OR allready be in a predbat controlled mode where charge rate is set to 0.
Apply Manual force discharge.

gcoan, 2024-06-22:
Could you add the logfile please covering the period of the forced discharge,  and some screen prints of the predbat status and the charge rate showing it unchanged when the discharge finishes.

Will make it easier to diagnose.

Also what inverter was this with?  A logfile from the inverter would be useful as well, it could be that predbat sent the charge rate change but it failed (eg comms issue).  It should normally be retried at the next 5 minute predbat run

springfall2008, 2024-06-22:
Potential fix on main for testing

reidjr2, 2024-06-22:
Yes, that was it. Charge rate now explicitly set on Manual forced discharge. Thank you :-)

closed 2024-06-23 by springfall2008