springfall2008/batpred #1238: solax x3 hybrid g4 apps.yaml problem

adamrhunt, 2024-06-22:
[image]
[image]
[image]
[image]
[image]
[appsyaml.pdf](https://github.com/user-attachments/files/15937021/appsyaml.pdf)
**Describe the bug**
A clear and concise description of what the bug is.

everything is working but the apps.yaml sx4 listed for my solax inverter some of sensor names were wrong ive changed them and all is working but its not controlling my charge and discharge times of my inverter that parts of my apps.yaml is wrong i believe.
i can write an automation to make my inverter go into force charge or force discharge. if needed, but im not sure how to trigger this from predbat. or unless you know a way to make the charge and discharge settings work on apps.yaml for my inverter. thanks

**Expected behavior**
A clear and concise description of what you expected to happen.

to make the apps.yaml match my inverter to control the charge and discharge times as per predbat charge and discharge schedule

**Predbat version**

xxxx v722.5

**Environment details**

- Inverter and battery setup solax x3 8kw 3 phase hybrid g4 inverter  3 solax batteries 
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

adamrhunt, 2024-06-22:
ive attached a list of my solax sensors and buttons available to use under its entities

adamrhunt, 2024-06-22:
[log.pdf](https://github.com/user-attachments/files/15937118/log.pdf)

springfall2008, 2024-08-06:
I'm a little confused as to why there is a HA automation for the button push inside your apps.yaml, did you put it in the wrong place as this would need to be setup in HA automations?

adamrhunt, 2024-08-06:
Yea sorry I was just trying to figure out ways of activating the charging as alot of the entities on the solax template were wrong for my inverter I had to change alot of them to match sensors listed on my modbus

adamrhunt, 2024-08-06:
The only way I've got it to currently work it's make an automation on home assistant that uses pred bat is charging and predbat is discharging sensors to switch  my inverter to manual then either force charge or manual then force discharge I can't figure out how to get that part of the apps.yaml right to control my inverter

adamrhunt, 2024-08-15:
[inverter apps.txt](https://github.com/user-attachments/files/16625471/inverter.apps.txt)
[pred bat charge  discharge automation.txt](https://github.com/user-attachments/files/16625474/pred.bat.charge.discharge.automation.txt)
[solax entities.txt](https://github.com/user-attachments/files/16625475/solax.entities.txt)
as requested here is the apps.yaml i have for my inverter so far and the list of solax entities available on my modbus and the automation i had to make on home assistant using the charge and discharge sensors on predbat as i cant get the charging and discharging bit to work on the apps.yaml for my inverter if you look at the app.yaml attached and my inverter entities can you work out whats wrong on my apps yaml as im stuck. thanks

closed 2026-08-28 by chalfontchubby