springfall2008/batpred #1249: Upgrading from v7.19.7 to latest PredBat

TheSargey, 2024-06-24:
Howdy folks,

Just a quick one, was hoping I could get some brief advice on the best route to upgrade from v7.19.7 to v8.1.1 - I note that I should upgrade to 7.22.5 first before going further however I see there's been some adjustment in the predbat-appdaemon configuration (which I'm currently using on my HAOS setup).

Is there a new FAQ around this or a closed topic as I've not been able to find it directly.

Thanks for help as always,

Sarge

gcoan, 2024-06-24:
Yes as per the title of the upgrades,  you must upgrade to 7.22.5 FIRST, let Predbat run and complete a full normal run,  then you can upgrade to any v8.x.x version.

You can use the select.predbat_update control or HACS if you are using the original AppDaemon installation method.

Don't turn on automatic updates (until you are on v8) as you need to upgrade to the interim version first

Probably should be put somewhere in the FAQ's or installation instructions, yes

TheSargey, 2024-06-24:
Hi Geoff,

That's great, thanks. I normally just update through the control so I'll go
with that.

Thanks again and all the best,

Sarge

TheSargey, 2024-06-24:
Sorry @gcoan - in my HAOS I have an update for appdaemon-predbat pending as well (to v1.0.11) should I update this as well before moving to V8?

Cheers,

Sarge

gcoan, 2024-06-24:
I'd suggest upgrading the AppDaemon predbat first as that contains a bundled version of predbat and may well take it to v8 in one go

TheSargey, 2024-06-24:
Thanks Geoff, upgraded to v8 no issues, then when to latest and now my status is "Changed to Error: Exception raised unsupported operand type(s) for +=: 'float' and 'NoneType'"  -  drat and blast!  Just having a look through now to see if there's something I've missed along the way.

TheSargey, 2024-06-24:
predbat.log has this error:

2024-06-24 23:30:00.052787 INFO pred_bat: --------------- PredBat - update at 2024-06-24 23:30:00+01:00 with clock skew 0 minutes, minutes now 1410
2024-06-24 23:30:00.055394 INFO pred_bat: Saved current settings to /config/predbat_config.json
2024-06-24 23:30:00.057672 INFO pred_bat: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 7.8 minutes
2024-06-24 23:30:00.059220 INFO pred_bat: Predbat /config/apps/predbat.py version v8.1.1 currently running, latest version is v8.1.1 latest beta v8.1.1
2024-06-24 23:30:00.060853 INFO pred_bat: Autoupdate: Currently up to date
2024-06-24 23:30:00.530494 INFO pred_bat: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2024-06-24 23:30:00.539938 INFO pred_bat: Predbat mode is set to Control charge & discharge
2024-06-24 23:30:04.831088 INFO pred_bat: Found 31680 load_today datapoints going back 10 days
2024-06-24 23:30:07.896572 INFO pred_bat: Car charging hold True threshold 6.0
2024-06-24 23:30:07.916431 INFO pred_bat: Current data so far today: load 30.7 kWh import 16.8 kWh export 0.0 kWh pv 15.4 kWh
2024-06-24 23:30:07.918455 INFO pred_bat: Fetching futurerate data from None
2024-06-24 23:30:07.997518 INFO pred_bat: Standing charge is set to 42.0105 p
2024-06-24 23:30:10.820086 INFO pred_bat: Rate min forward looking: now 13.31 at end of forecast 13.31
2024-06-24 23:30:10.831789 INFO pred_bat: Import rates min 13.31 max 37.0 average 19.83
2024-06-24 23:30:10.904721 INFO pred_bat: Export rates min 0 max 0 average 0.0
2024-06-24 23:30:10.907822 INFO pred_bat: Rate thresholds (for charge/discharge) are import 36.5p (0.0) export 13.31p (0.0)
2024-06-24 23:30:10.912999 INFO pred_bat: High export rate found rates in range 99 to -99
2024-06-24 23:30:10.926309 INFO pred_bat: Low Import rate found rates in range 13.31 to 34.22
2024-06-24 23:30:11.320375 INFO pred_bat: Todays energy import 16.8 kWh export 0.0 kWh cost 328.34 p import 328.34 p export 0.0 p carbon 0.0 kg
2024-06-24 23:30:11.343595 INFO pred_bat: Using solcast from inside HA
2024-06-24 23:30:11.350799 INFO pred_bat: PV Data for pv_forecast_today total 16.34 kWh
2024-06-24 23:30:11.385042 INFO pred_bat: PV Data for pv_forecast_tomorrow total 18.04 kWh
2024-06-24 23:30:11.419766 INFO pred_bat: PV Data for pv_forecast_d3 total 15.97 kWh
2024-06-24 23:30:11.450708 INFO pred_bat: PV Data for pv_forecast_d4 total 19.1 kWh
2024-06-24 23:30:11.510195 INFO pred_bat: PV Forecast for today is 16.34 (12.46 10% 20.8 90%) kWh and left today is 0.0 (0.0 10% 0.0 90%) kWh
2024-06-24 23:30:11.625202 INFO pred_bat: PV Forecast for day tomorrow is 18.04 (13.61 10% 21.81 90%) kWh
2024-06-24 23:30:11.862279 INFO pred_bat: PV Forecast for day d2 is 15.97 (10.01 10% 22.55 90%) kWh
2024-06-24 23:30:12.027379 INFO pred_bat: PV Forecast for day d3 is 19.1 (5.99 10% 24.88 90%) kWh
2024-06-24 23:30:12.408218 INFO pred_bat: Historical data totals for days [7, 14, 21] are [29.2, 29.46, 29.46] - min 29.2
2024-06-24 23:30:12.423656 INFO pred_bat: Model filter enabled - Discarding day 7 as it is the lowest of the 3 datapoints
2024-06-24 23:30:12.429310 INFO pred_bat: Historical days now [14, 21] weight [1, 1]
2024-06-24 23:30:12.531369 INFO pred_bat: Today's load divergence 111.64 % in-day adjustment 111.05 % damping 0.95x, Predicted so far 21.62 kWh with 0.0 kWh car excluded and 2.14 kWh import ignored and 0 forecast extra, Actual so far 24.53 kWh with 1.05 kWh car excluded and 5.09 kWh import ignored
2024-06-24 23:30:12.875399 INFO pred_bat: Inverter 0: Type GE GivEnergy
2024-06-24 23:30:12.877494 INFO pred_bat: Inverter 0 using Rest API http://homeassistant.local:6345
2024-06-24 23:30:13.955118 INFO pred_bat: Inverter 0 does not have timed pause support enabled
2024-06-24 23:30:13.961146 INFO pred_bat: Invertor time 2024-06-24 23:29:50+01:00 AppDaemon time 2024-06-24 23:30:13.955538+01:00 difference -0.4 minutes
2024-06-24 23:30:13.967566 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-06-24 23:30:13.972487 INFO pred_bat: Inverter 0 with soc_max 10.44 kWh nominal_capacity 10.44 kWh battery rate raw 2600 w charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 3.6 kW export limit 3.6 kW reserve 4.0 % current_reserve 4.0 %
2024-06-24 23:30:15.580882 INFO pred_bat: Inverter 0 SOC: 0.52 kW 5 % Current charge rate 2600.0 w Current discharge rate 2600.0 w Current power 1552.0 w Current voltage 52.0
2024-06-24 23:30:15.597365 INFO pred_bat: Inverter 0 scheduled charge enable is False
2024-06-24 23:30:15.601604 INFO pred_bat: Inverter 0 charge windows currently []
2024-06-24 23:30:15.606859 INFO pred_bat: Inverter 0 Charge settings: timed charged is disabled, power 2.6 kW
2024-06-24 23:30:15.609867 INFO pred_bat: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
2024-06-24 23:30:15.618471 INFO pred_bat: Inverter 0 sensor.predbat_GE_0_scheduled_discharge_enable set to False
2024-06-24 23:30:15.622309 INFO pred_bat: Inverter 0 scheduled discharge enable is False
2024-06-24 23:30:15.625781 INFO pred_bat: Inverter 0 discharge windows currently [{'start': 0, 'end': 0, 'average': 0}, {'start': 1440, 'end': 1440, 'average': 0}]
2024-06-24 23:30:15.628683 INFO pred_bat: Found 1 inverters totals: min reserve 0.42 current reserve 0.42 soc_max 10.44 soc 0.52 charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 3.6 export limit 3.6 kW loss charge 3 % loss discharge 3  [... cut ...]
  File "/config/apps/predbat.py", line 6383, in sort_window_by_price_combined
    average += self.metric_self_sufficiency
TypeError: unsupported operand type(s) for +=: 'float' and 'NoneType'

2024-06-24 23:30:16.942438 INFO pred_bat: Info: record_status Error: Exception raised unsupported operand type(s) for +=: 'float' and 'NoneType'

From what I can see, it looks like  "INFO pred_bat:     BAT:" is reporting random "g<be-,    g<be-,    g<be-,    g<be-" text rather than numbers?  Is that' what's causing the exception?

gcoan, 2024-06-24:
I had this when I upgraded from v8.0.0 to 8.1.1.  Reverting back and the error stopped.

Tried again to upgrade, same error.  Then I restarted AppDaemon and the error cleared.

Try restarting your appdaemon-predbat add-on

TheSargey, 2024-06-24:
Simple as that, it fixed it.

I'll have to remember to give appdaemon-predbat a nudge first before flooding the forum!  

Thanks as always sir, much appreciated.

closed 2024-06-24 by TheSargey