springfall2008/batpred #1250: Solar Clipping on Givenergy Hybrid Inverter

robyholmes, 2024-06-24:
**Describe the bug**
Predbat isn't trying to stop hybrid inverters from clipping by discharging the battery before the peak of the day. So for example a 5kw inverter with the battery at 100%, and solar over forecast over the 5kw is not being discharged in the morning to avoid it clipping.

**Expected behavior**
Planned discharging in the morning when the solar forecast is less than the inverter capacity, creating room in the battery so DC to DC charging.

**Predbat version**

V8.1.1 & Main on 24/06/2024
Predbat Add-on
None AppDaemon

**Environment details**

Givenergy Hybrid 5kw Gen 3
2x 9.5kwh Batteries
7.2kwh Peak Solar Panels
HAOS 2024.6.4

**Screenshots**
Taken from Facebook Post (24/06/2024)
V8.1.2
[image]

**Log file**
To follow soon, as I missed downloading it today before the peak today. Tomorrow doesn't look to be over the peak so might be later this week.

gcoan, 2024-06-24:
There are a few other issues already raised related to inverter clipping and Predbat's management of it, #886 and #1206 as examples.

At the moment Predbat doesn't yet have actively try to reduce clipping, but this is something Trefor is working on.

There are different strategies such as force discharging the battery as you said, or letting the battery trickle charge at a lower rate so it doesn't fill up too early in the day.

To prevent duplication of issues can you see if you can add your experience/feedback to one of the existing issues please

robyholmes, 2024-06-24:
I understand, however as Trefor was talking to me and others about this issue on facebook, trying different settings etc. I thought putting it on Github as an issue was the best way of tracking it, rather than facebook replies.

gcoan, 2024-06-24:
Agreed it's a hot topic that others have been raising for some time.  I was just suggesting that you add to an existing issue rather than duplicate the conversation about how to implement clipping control over multiple tickets so you join an existing conversation rather than starting a new one.

Have a look at the other issues though

robyholmes, 2024-06-26:
So far today, things are looking food. It looks to be trying to avoid clipping by putting some planned discharges in.
[image]

closed 2025-01-01 by springfall2008