springfall2008/batpred #1251: Accomodation of GivTCP data glitches

mpartington, 2024-06-25:
**Describe the bug**
This is not Predbat driven, but perhaps could be accommodated as part off the data filtering.  I've recently noticed that if I restart HA, I momentarily get spikes or drops on various parameters.  This impacts Predbats calculations, for example it over counts export, which then overestimates the savings. 

I suspect the same happens with load, but I have a custom sensor set for that

**Expected behavior**
For GivTCP to pass good data (have reported on Beta chat), however failing that is it possible for Predbat to ignore the glitches?

**Predbat version**
8.1.2

GivTCP beta 2.4.324

**Environment details**

- Inverter and battery setup
- Standard HAOS installer 

**Screenshots**

[image]

[image]

**Log file**
Not sure it will capture anything else, but will.provide if needed

mpartington, 2024-06-28:
Had to restart again today, and the spike added 4kWh into Predbats export, which spuriously knocked 60p off my accumulated cost for the day. 

[image]

gcoan, 2024-06-28:
I raised a potentially linked issue a couple of days ago #1253, the predbat import and export figures are not correctly being calculated. I'm not seeing spikes but the values were not the same as the givtcp sensors

springfall2008, 2025-01-01:
Should have been implemented now

closed 2025-01-01 by springfall2008