springfall2008/batpred #1252: Predbat blocking iSolar

MalcolmSpencer, 2024-06-25:
When I turn on 'iboost on solar' predbat is trying to keep the battery soc below the 'iboost_min_soc' by discharging the battery so that power is not diverted to iboost. screens shots of the plan with iboost on solar on and off below.

I appreciate that financially using iboost might not be the best thing to do but environmentally I want to use iboost in the summer rather than using gas.

This might be because of the rates and thresholds
low threshold = 14.6p
High threshold = 24.2p
import = 25.3p
export = 15.7p

**Expected behavior**
Predbat should not export power until iboost is full (boost max energy)

**Predbat version**
predbat addon 1.1.7
predbat 8.1.1

**Environment details**
Core 2024.6.4
Supervisor 2024.06.0
Operating System 12.4
Frontend 20240610.1

- Inverter and battery setup
Myenergi Libbi
Myenergi Eddi
Myenergi Zappi

**Screenshots**
iboost on solar power off
[image]

iboost on solar power on
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/15974983/predbat.log)

springfall2008, 2024-07-03:
If you change input_number.iboost_value_scaling to something higher e.g. 1.0 then you shouldn't get this as much.

springfall2008, 2024-08-06:
Did you manage to confirm this?

closed 2025-01-01 by springfall2008