springfall2008/batpred #1253: h0 sensor values are incorrect

gcoan, 2024-06-25:
**Describe the bug**
Recently #1204 was fixed to correct the value of predbat_load_energy_h0

I think there may be a problem with other h0 sensors as well:

predbat_import_energy_h0 (how I spotted this issue), value 0.8kWh, but the givtcp import today kwh is 1.5kWh:
[image]

export is closer, export_energy_h0 is 61.6kWh but givtcp value is 61.9kWh:
[image]

pv_energy_h0 is also close, 61.5kWh vs the helper sum that adds my two GE inverter sensors together is 61.3kWh [possibly just a rounding issue]
[image]

Couple of other things, none of these h0 sensors are mentioned in the output data documentation, only a passing reference to soc_kwh_h0 being the scaled battery kWh in apps.yaml.
I can add them if you can confirm that they are supposed to be the measured inverter import, export, PV, etc.

And I think there might be something wrong with the way the sensors are defined in predbat as the HA history graph shows them separate from the givtcp sensors, so there must be a mismatch of type, uom or something??

**Expected behavior**
Predbat sensors all described in the documentation and provide same results as equivalent inverter sensors.

**Predbat version**
8.1.2
HAOS
2x Gen 1 hybrid inverters

springfall2008, 2024-06-30:
One thing on my potential todo list is a breaking change to rename all predbat.* entities to sensor.* which will likely fix some of the issues in HA.

gcoan, 2024-06-30:
I'm working on a PR to the documentation so will add the h0 entities in.  If/when you rename them then they're there.

gcoan, 2024-10-23:
@springfall2008 I have been updating the documentation and have included all the h0 sensors in the output-data page.

Checking back at the sensors, the original issue I had with h0 sensors not matching the source sensors in apps.yaml is largely resolved with PV, Import and Export all matching, BUT load_energy_h0 is consistently lower than the source sensor.

Top graph is the input sensors in apps.yaml, bottom graph the Predbat h0 sensors:
[image]

I wonder if the problem is the loading and filling in of perceived gaps of the load_energy sensor? 
I always get warnings about gaps in the load energy, vis:
```
2024-10-23 16:35:00.718903: Found 12961 load_today datapoints going back 9 days
2024-10-23 16:35:01.144247: Car charging hold True threshold 6.0
2024-10-23 16:35:01.144795: Current data so far today: load 9.68 kWh import 4.9 kWh export 9.4 kWh pv 19.1 kWh
...
2024-10-23 16:35:01.757731: Historical data totals for days [2, 3, 4, 5, 6, 7, 8] are [18.54, 39.14, 26.09, 14.32, 17.41, 16.28, 26.04] - min 14.32
2024-10-23 16:35:01.757784: Model filter enabled - Discarding day 5 as it is the lowest of the 7 datapoints
2024-10-23 16:35:01.757898: Warn: Historical day 2 has 10 minutes of gap in the data, filled from 18.54 kWh to make new average 18.67 kWh (percent 99%)
2024-10-23 16:35:01.764633: Warn: Historical day 6 has 10 minutes of gap in the data, filled from 17.41 kWh to make new average 17.53 kWh (percent 99%)
2024-10-23 16:35:01.768757: Warn: Historical day 7 has 10 minutes of gap in the data, filled from 16.28 kWh to make new average 16.39 kWh (percent 99%)
2024-10-23 16:35:01.778926: Warn: Historical day 8 has 5 minutes of gap in the data, filled from 26.04 kWh to make new average 26.13 kWh (percent 100%)
2024-10-23 16:35:01.782176: Historical days now [2, 3, 4, 6, 7, 8] weight [1, 1, 1, 1, 1, 1]
```

but as my load_today sensor is a time based trigger template that fires every 5 minutes to calculate the house load, this is impossible. If I look at the sensor I don't see any data gaps at all.

Wonder if the problem is the data load in predbat somehow?

Now using the predbat addon which was supposed to be more reliable on loading data.  I know @swiftrr also always gets warnings about missing data on load_energy

SwiftRR, 2024-10-23:
These are my graphs:
[image]
I experiment with predai and I use either predai or historical data by commenting and uncommenting the load forecast lines in apps.yaml.
I like to see the effect on the in-day chart.

Sods law that I changed to predai part way through today!

**Notes**
- I have always experienced warnings for gaps in historical day data. I can get rid of the warnings by increasing the load_filter_threshold from the default 30 min to 60 min. I don't know if this is wise but it is nice to removed the warnings.

- When I run predai, the warnings disappear.

I will change back to historical now and run with it for all of tomorrow.
After a day of historical data, I will generate further graphs and post here.

Rob

springfall2008, 2025-01-01:
As this ticket is old I'm going to close it, please re-open if the issue is there on latest.

gcoan, 2025-01-01:
I've reopened this because I believe the load_energy_h0 sensor is still wrong and under-reads when compared to the input load_energy_today sensor that predbat is configured to use in apps.yaml.

Looking at today and yesterday for example:
[image]

Top graph is h0 sensor, bottom is load sensor in apps.yaml.

Yesterday h0 peak value is 52.58kWh, load sensor is 48kWh
Today (power cut from 1am to 7am so hence flat line) but h0 peak right now is 50.92kWh, load sensor is 42.9kWh

springfall2008, 2025-01-04:
I assume you mean over read? Are you sure its not as you have load_scaling set to 1.1?

gcoan, 2025-01-04:
Good spot, load scaling is definitely part of the difference, I hadn't thought of that and will add something to the documentation to highlight that

But, I don't think its the whole answer.  h0 seems to over-read despite load scaling

I have load scaling currently set to 2.0 as its cold and the heat pump is going into overdrive (its giving stupid figures now into Predbat so I'll change it back, but it does help illustrate the differences).

If I compare the h0 load sensor (top) and the house load (middle), h0 is only updating every half hour rather than every 10 minute predbat recalculate, but at no times is it twice the house load

eg 19:00 h0 206, house load 101.5
19:30 h0 209.12, 20:00 213.82, 20:30 215.72 but house load remained flat at 101.6

[image]

gcoan, 2025-01-04:
logfile: 
[predbat.log](https://github.com/user-attachments/files/18308291/predbat.log)

springfall2008, 2025-01-05:
Did you account for in-day adjustment?

Not sure about the 30 minute update, it maybe due to chunks that history is calculated in but I need to check.

gcoan, 2025-01-06:
I think I know what the discrepancy is, I'll check and confirm thanks

gcoan, 2025-01-06:
> If I compare the h0 load sensor (top) and the house load (middle), h0 is only updating every half hour rather than every 10 minute predbat recalculate, but at no times is it twice the house load
>
> eg [...]

Overnight I realised what was probably the cause of the discrepancy; that the house load that Predbat was using is not my custom HA house load sensor but the inverter house load taken from the GE Cloud (this is a temporary thing due to me accidentally purging some history I shouldn't have done, been using the GE Cloud).

As I have two inverters that share the load the inverter house load data isn't quite correct.

Checking the data on the GE portal:
19:00 103
19:30 104.4
20:00 106.8
20:30 107.8

Which pretty much match the values given by the h0 sensor.

I'll update the documentation and I think we're OK now.  thanks

closed 2025-01-06 by gcoan