springfall2008/batpred #1254: Can't upgrade past v8.0.0

Sycamore967, 2024-06-26:
**Describe the bug**
Upgrading creates error - Unsupported operand - see screenshots

**Expected behavior**
Can upgrade to latest version

**Predbat version**

8.0.0

**Environment details**

I've upgraded to 7.22.5 first and went to v8.0.0 without any issues.
If it helps my install is old and was an AppDaemon install - Predbat doesn't appear in Addons
After trying to update, downgrading to v8.0.0 fixes the issue again

**Screenshots**
A couple attached
[image]

[image]

**Log file**
No unless you need it.

gcoan, 2024-06-26:
I had the same issue as have others, restart AppDaemon when you install the upgraded version and this fixes this float error.

Predbat add on isn't in the addon store, you need to manually add the repository. If you want to upgrade to it from AppDaemon, upgrade instructions are in my fork of the documentation https://github.com/gcoan/batpred/blob/main/docs/install.md

Sycamore967, 2024-06-26:
Hello - brill thanks for the update.  It helped me sort the issue and I've now migrated to the Predbat Addon.