springfall2008/batpred #1261: Deduct iboost power from home load

MalcolmSpencer, 2024-06-28:
**Is your feature request related to a problem? Please describe.**
Looking at the code it seems that predbat doesn't subtract the iboost power from today's load in the same way that it does for car charging load. This leads to an over estimate of the house load consumption unless an external helper is created to deduct the iboost load from the inverter home load.

**Describe the solution you'd like**
when iboost is enabled and an energy sensor defined then Predbat should remove the iboost load from the home load in the same way that is does for car charging load

**Describe alternatives you've considered**
create an external helper to remove the iboost power from the inverter how power.
if this is the preferred solution then update the documentation to make it clear.

**Additional context**
Add any other context or screenshots about the feature request here.

gcoan, 2024-06-28:
I was thinking about this issue. 

There should be consistency as much as possible in Predbat. The problem is that at the moment there isn't a sensor for historical iboost load so Predbat isn't aware of it to subtract it from the house load. 

But what you can do is use car_charging_energy in apps.yaml and point it to a sensor that records the iboost energy and that should do what you want https://springfall2008.github.io/batpred/apps-yaml/

Although it's called car energy you can use it for any subtractive load as mentioned in the documentation

MalcolmSpencer, 2024-06-28:
In apps.yaml iboost is configured with a sensor to record the daily load consumed. it just doesn't seem to be included in the load calc.

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  # Set to automatch to Myenergi Eddi
  iboost_energy_today: 're:(sensor.myenergi_eddi_[0-9a-z]+_energy_consumed_session)'
  iboost_energy_scaling: 1.0

gcoan, 2024-06-28:
Ah yes, that would be perfect.  It sounds like a bug rather than a feature request as it does say that setting this sensor'makes the predictions more accurate', and if it's not included in the load calc then it's clearly not!

Typo on kWh as well.

I'm working on a PR for Predbat so if you have already identified where in the code its missing the subtraction, can you let me know and I can include it in my next set of changes

MalcolmSpencer, 2024-06-28:
The deduction of the car charge load from the raw load is at https://github.com/springfall2008/batpred/blob/main/apps/predbat/predbat.py#L1431

so I guess the deduction of iboost would be similar

gcoan, 2024-06-28:
Thanks I'll have a look at it

springfall2008, 2024-06-30:
You are right it doesn't subtract iboost today from the load, it probably should do. It's not a bug as such but more of an omission.

closed 2025-01-01 by springfall2008