springfall2008/batpred #1265: Predbat stop

KeithB, 2024-06-28:
**Describe the bug**
Predbat stops it's standard cycles of refreshing the plan after 8 hours uptime. Restart of Predbat required to fix.

Root cause seems to be a DNS lookup failure - interestingly, what is also a repeat behaviour is the cause of the failure. Pi-Hole adds a temporary block because the container has done 1,000 DNS lookups in the space of 60 second. Pi-Hole reconfigured to 5,000 to see if this helps as a workaround.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Predbat version**

8.1.1

**Environment details**
- GivEnergy Gen 3 Hybrid 5 kw + 9.5KwH battery
- HA running in docker.
- Predbat running in separate docker container.

**Screenshots**

**Log file**
Following error is seen then nothing but a repeat of the refresh config entities log.
```
2024-06-28 17:30:47.278810: Info: Refresh config entities due to their age of 480.0 minutes
2024-06-28 17:30:47.352283: Refreshing Predbat configuration
2024-06-28 17:30:48.578933: Error: HTTPSConnectionPool(host='homeassistant.myhouse.io', port=443): Max retries exceeded with url: /api/states/switch.predbat_set_discharge_freeze_only (Caused by NameResolutionError("<urllib3.connection.HTTPSConnectionobject at 0x7f6efcfaa930>: Failed to resolve 'homeassistant.myhouse.io' ([Errno -3] Temporary failure in name resolution)"))
2024-06-28 17:30:48.579061: Running task: <bound method PredBat.run_time_loop_balance of <predbat.PredBat object at 0x7f6efde20c50>>
2024-06-28 17:31:02.585546: Running task: <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7f6efde20c50>>
2024-06-28 17:31:02.585817: Info: Refresh config entities due to their age of 480.0 minutes
2024-06-28 17:31:15.591817: Running task: <bound method PredBat.run_time_loop_balance of <predbat.PredBat object at 0x7f6efde20c50>>
2024-06-28 17:31:17.593313: Running task: <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7f6efde20c50>>
2024-06-28 17:31:17.593618: Info: Refresh config entities due to their age of 480.0 minutes
```

springfall2008, 2024-07-03:
So you have configured ha_url as homeassistant.myhouse.io? I suggest just changing that to the IP address to avoid the DNS?

gcoan, 2024-07-03:
Added this tip to the documentation

KeithB, 2024-07-05:
> So you have configured ha_url as homeassistant.myhouse.io? I suggest just changing that to the IP address to avoid the DNS?

Correct, ha_url is (and always has been since install I think) a domain name rather than IP address. Doesn't affect this scenario but I have nginx sat in front of most things to simplify https connections when the majority of self-host software defaults to http. In this case, nginx deals with the forwarding to HA on port 8123.

I agree there are several ways to solve it from a DNS perspective (use IP making sure it's static, adding an entry to /etc/hosts, changing PiHole/Adguard abuse block thresholds etc.) but I still see it more as a workaround. 

The things that came to mind were 
1. What was Predbat doing (why always the 8 hour mark and what on made it trigger > 1,000 calls in 60 seconds) 
2. Could the error handling be improved when the https call fails?

closed 2025-01-01 by springfall2008